Koenkk/zigbee2mqtt #13756: Cannaot switch to dev branche

Dylantje, 2022-08-29:
### What happened?

When i try to follow the wiki to change to dev branche.
I get a error:
`pi@odroidn2:/opt/zigbee2mqtt$ sudo systemctl stop zigbee2mqtt
pi@odroidn2:/opt/zigbee2mqtt$ cd /opt/zigbee2mqtt
pi@odroidn2:/opt/zigbee2mqtt$ git fetch
pi@odroidn2:/opt/zigbee2mqtt$ git checkout dev
error: pathspec 'dev' did not match any file(s) known to git
`
master:
`
pi@odroidn2:/opt/zigbee2mqtt$ git checkout master
M       data/configuration.yaml
Already on 'master'
Your branch is up to date with 'origin/master'.
`

Update the master no problem.
Is there something i can do?
Or other options?

### What did you expect to happen?

update to dev

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

    20201026

### Adapter

    zStack3x0

### Debug log

Zigbee2MQTT versie
    [1.27.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.27.0) commit: [a9b8808](https://github.com/Koenkk/zigbee2mqtt/commit/a9b8808)

Coördinator type
    zStack3x0

Coordinator revisie
    20201026

Coordinator IEEE adres
    0x00124b0021cc386f

Versie frontend
    0.6.107

Koenkk, 2022-08-30:
Try with:
```bash
git remote add origin https://github.com/Koenkk/zigbee2mqtt.git
git fetch origin
git checkout dev
```

closed 2022-10-08 by github-actions[bot]

Dylantje, 2022-11-06:
Give it a try...
Only no luck @ all:
`
pi@odroidn2:~$ sudo systemctl stop zigbee2mqtt
pi@odroidn2:~$ cd /opt/zigbee2mqtt
pi@odroidn2:/opt/zigbee2mqtt$ cp -R data data-backup
pi@odroidn2:/opt/zigbee2mqtt$ git remote add origin https://github.com/Koenkk/zi                          gbee2mqtt.git
error: remote origin already exists.
pi@odroidn2:/opt/zigbee2mqtt$ git fetch origin
pi@odroidn2:/opt/zigbee2mqtt$ git checkout dev
error: pathspec 'dev' did not match any file(s) known to git
pi@odroidn2:/opt/zigbee2mqtt$
`