Koenkk/zigbee2mqtt #13758: Router Stick Marked Offline

jerrm, 2022-08-29:
### What happened?

Zigbee2MQTT 1.27  
Home Assitant Add-On 
Sonoff ZBDongle-P (TI CC2652P) x 4
Coordinator FW: 20220103
Router FW: 20220125
"Simple" Availability enabled

TL;DR - If your new router stick is going offline, try rebooting the Zigbee2MQTT host (restarting the Z2M container may be enough, but I haven't tested).

I don't know how well known this is, but I didn't see anything directly on point after some googling.  

Also wonder if it should be under [the zstack github](https://github.com/Koenkk/Z-Stack-firmware), but the solution seemed to be on the zigbee2mqtt side of things.  @koenkk, let me know if I should move this. 

Ideally the report gives @Koenkk some hints to fix (and apologies if already addressed and my google skills failed me).

Worst case, my experience is preserved for anyone googling with the same issues.

I have had three Sonoff ZBDongle-P (cc2652p) sticks since start of the year (1 coordinator + 2 routers), and just added a fourth.  Overall very reliable, but the router sticks had some odd behaviors.

The first router stick I flashed never had any issues.  Apparently worked great and stayed online continuously (but now wonder if I had rebooted early on at some point).

The second stick would be marked offline, failing to respond to pings sent by z2m.  After some googling, I set the reporting interval and moved on.  Been that way for months now.

I just added a third router stick and faced the same issue with it being marked offline after 10 minutes, failing to respond to pings sent by z2m:

```
Warning 2022-08-29 16:15:54 Failed to ping 'Router 3' (attempt 1/2, Read 0x00124b0024c146fd/242 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4477 - 242 - 54 - 0 - 1 after 10000ms))
Warning 2022-08-29 16:16:21 Failed to ping 'Router 3' (attempt 2/2, Read 0x00124b0024c146fd/242 genBasic(["zclVersion"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 4477 - 242 - 55 - 0 - 1 after 10000ms))
Debug 2022-08-29 16:16:24 Active device 'Router 3' was last seen '10.67' minutes ago.
Info 2022-08-29 16:16:24 MQTT publish: topic 'zigbee2mqtt/Router 3/availability', payload 'offline'
```

The odd thing is the second stick now responds to pings.  It's probably been that way for months, I had not looked at it in quite a while, assuming the reporting interval was what was keeping it active:
```
Debug 2022-08-29 16:17:05 Received Zigbee message from 'Router 2', type 'attributeReport', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 8 with groupID 0
Info 2022-08-29 16:17:05 MQTT publish: topic 'zigbee2mqtt/Router 2', payload '{"last_seen":"2022-08-29T16:33:00-04:00","linkquality":76}'
```

Trying to see what changed in for "Router 2" since I installed it months ago I did some more testing:

* I first tried the reconfigure option on the new "Router 3" stick - Same Ping Results.
* I then tried rebooting the new "Router 3" stick (unplug/re-plug) - Same Ping Results.
* I then did a "Reboot host" from HA Settings->System->Hardware - ***SUCCESSFUL PING!***
```
Debug 2022-08-29 16:33:05 Received Zigbee message from 'Router Sonoff 1', type 'attributeReport', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 8 with groupID 0
Info 2022-08-29 16:33:05 MQTT publish: topic 'zigbee2mqtt/Router Sonoff 1', payload '{"last_seen":"2022-08-29T16:33:00-04:00","linkquality":134}'
Debug 2022-08-29 16:37:30 Received Zigbee message from 'Router 3', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 8 with groupID 0
Info 2022-08-29 16:37:30 MQTT publish: topic 'zigbee2mqtt/Router 3', payload '{"last_seen":"2022-08-29T16:37:25-04:00","linkquality":109}'
Debug 2022-08-29 16:37:30Succesfully pinged 'Router 3' (attempt 1/2)
```

### Zigbee2MQTT version

1.27

### Adapter firmware version

20220103

### Adapter

Sonoff ZBDongle-P (TI CC2652P)

jerrm, 2022-09-08:
Did you try restarting the container or rebooting the pc?

If not please try the container first, then reboot if no change and report back.

jerrm, 2022-10-09:
Just a note before this ages out.

I've subsequently have tested two more sticks and found restarting z2m resolves the offline timeout issues.  Before restaring pings would fail and the stick would be marked offline.  After restarting z2m everything works as it should.  This is all **without** changing reporting intervals on the stick. 

At this point I have 5 sonoff sticks as routers on two different HA instances without issue.

The restart also seems to have resolved the same issue posted at https://github.com/Koenkk/zigbee2mqtt/issues/14312#issuecomment-1267808805.

closed 2022-11-18 by github-actions[bot]

harvindhillon, 2024-09-02:
This issue is still there. On the latest release and routers based on TI chip constantly goes offline. Sometimes restarting fixes it, sometimes it doesnt and I am forced to repair which can take sometime as interview always fails.