Koenkk/zigbee2mqtt #13760: FP1 no reading

Jens-Wymeersch, 2022-08-30:
### What happened?

I've paired 2 FP1 devices successfully but I'm unable to get a reading out of them. No presence detection, no temperature, etc...

[image]

The buttons won't work.

I have a fairly large zigbee network of total of 178 of which Router: 94 and End devices: 84

### What did you expect to happen?

That I get a reading like my other devices

### How to reproduce it (minimal and precise)

I have 2 identical brand new devices where this has occurred 

### Zigbee2MQTT version

 1.27.0 

### Adapter firmware version

20220219

### Adapter

ZZH!

### Debug log

When I look at the state of the device 

```
{
    "approach_distance": "medium",
    "linkquality": 32,
    "monitoring_mode": "undirected",
    "motion_sensitivity": "low",
    "presence": null
}
```

When I want to change the sensitivity I get the following error message :

`2022-08-30 09:58:20Publish 'set' 'motion_sensitivity' to '0x54ef44100053de97' failed: 'Error: Write 0x54ef44100053de97/1 aqaraOpple({"268":{"value":3,"type":32}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 17876 - 1 - 29 - 64704 - 4 after 10000ms)'`

Jens-Wymeersch, 2022-08-30:
This is the log when pairing 

`info  2022-08-30 12:12:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Aqara presence detector FP1 (regions not supported for now)","friendly_name":"0x54ef44100053de97","model":"RTCZCGQ11LM","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'`

Here is the log related to the payload

`info  2022-08-30 12:12:20: MQTT publish: topic 'zigbee2mqtt/0x54ef44100053de97', payload '{"approach_distance":null,"device_temperature":null,"linkquality":47,"monitoring_mode":null,"motion_sensitivity":"low","power_outage_count":null,"presence":null,"presence_event":null,"update":{"state":null},"update_available":null}'`

Jens-Wymeersch, 2022-08-31:
@Koenkk any ideas what I can do to fix this ?

sorryusernameisalreadytaken, 2022-09-02:
Are able to do a ota? Can you tell us your FW on the FP1?

Also good to know is your zigbee channel you are working with.

Jens-Wymeersch, 2022-09-02:
@sorryusernameisalreadytaken i'm able to find the device in the OTA tab but like the rest of my aqara devices, I am unable to update the firmware. The firmware is 20210121
The zigbee channel is 20

sorryusernameisalreadytaken, 2022-09-02:
`This sensor does not work on Zigbee channel 21-24`
Source: https://www.zigbee2mqtt.io/devices/RTCZCGQ11LM.html#xiaomi-rtczcgq11lm

I know it can be a lot of work to set your zigbee channel to, say, 15 or 25, but I don't think you can get around it.

The "20210121" seems to be somewhat unfavorable on the part of the manufacturer. As far as I have read, the first version is the build number, which is also used for other firmware updates. Trigger an ota requests. In the logs should then be a file version.

Jens-Wymeersch, 2022-09-03:
Hello, so I have 2 that work perfectly. And the last 2 I have problems with. So I know that they work. Related to the OTA, I am unable to make it work for anything else then HUE and IKEA. Any advice is welcome

Jens-Wymeersch, 2022-09-03:
In another similar environment, I got them paired very quickly and found the data I was looking for. Maybe there is a restriction related to the number of devices I can run on ZZH!

sorryusernameisalreadytaken, 2022-09-04:
How many devices do you have in your network and how many of them are zigbee routers?
What is your ZZH device, this stick here: https://electrolama.com/projects/zig-a-zig-ah/

Please check your firmware like described here:
https://community.home-assistant.io/t/new-aqara-human-presence-sensor-fp1/373471/186?u=jamieme

And can you tell me what happened if you remove the 2-good once from your z-network and try to re-add the 2 faulty once?

Jens-Wymeersch, 2022-09-04:
I have more than 170 zigbee devices in my house - about 90 are routers.

sorryusernameisalreadytaken, 2022-09-04:
> I have more than 170 zigbee devices in my house - about 90 are routers.

and the others questions....? :D

Jens-Wymeersch, 2022-09-04:
It looks like the fireware is indicated as 1 (fileVersion":1) in the one's that don't work and "fileVersion":54 in the ones that work.

@sorryusernameisalreadytaken I quickly responded as I didn't have time till now.
On the last question "what happened if you remove the 2-good once from your z-network and try to re-add the 2 faulty once?"
- They stay faulty - that said please note that if I setup a clean Z2M - I didn't have any trouble
- Afterwards, i re-paired the good ones without problems

closed 2022-10-12 by github-actions[bot]