Koenkk/zigbee2mqtt #13766: Woox modelID: 'TS0215A'

ijs1550, 2022-08-30:
### Link

https://wooxhome.com/products-c10/security-c6/woox-r7054-smart-remote-control-p53

### Database entry

{"id":23,"type":"EndDevice","ieeeAddr":"0xa4c1383557974167","nwkAddr":4287,"manufId":4417,"manufName":"_TZ3000_0zrccfgx","powerSource":"Battery","modelId":"TS0215A","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1025,"inClusterList":[1,3,1280,0],"outClusterList":[1281,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b002298a4f3","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1661863699443,"defaultSendRequestWhen":"immediate"}

### Comments

{ fingerprint: [{modelID: 'TS0215A', manufacturerName: '_TZ3000_p6ju8myv'}, {modelID: 'TS0215A', manufacturerName: '_TZ3000_fsiepnrh'}], model: 'TS0215A_remote', vendor: 'TuYa', description: 'Security remote control', fromZigbee:[fz.command_arm,fz.command_emergency,fz.battery], exposes: [e.battery(), e.action(['disarm', 'arm_day_zones', 'arm_night_zones', 'arm_all_zones', 'exit_delay', 'emergency'])], toZigbee: [], whiteLabel: [{vendor: 'Woox', model: 'R7054'}, {vendor: 'Nedis', model: 'ZBRC10WT'}], meta: {configureKey: 1}, configure: async (device, coordinatorEndpoint, logger) => { const endpoint = device.getEndpoint(1); await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genTime', 'genBasic', 'ssIasAce', 'ssIasZone']); }, 
Is no longer in the woox.js herdsman-convertes

Koenkk, 2022-08-30:
It is here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/54cd3f92d683a06d2b5ce1884ef72345799129a6/devices/tuya.js#L1224

ijs1550, 2022-08-30:
On Tweakers I found this: { fingerprint: [{modelID: 'TS0215A', manufacturerName: '_TZ3000_p6ju8myv'}, {modelID: 'TS0215A', manufacturerName: '_TZ3000_fsiepnrh'}], model: 'TS0215A_remote', vendor: 'TuYa', description: 'Security remote control', fromZigbee:[fz.command_arm,fz.command_emergency,fz.battery], exposes: [e.battery(), e.action(['disarm', 'arm_day_zones', 'arm_night_zones', 'arm_all_zones', 'exit_delay', 'emergency'])], toZigbee: [], whiteLabel: [{vendor: 'Woox', model: 'R7054'}, {vendor: 'Nedis', model: 'ZBRC10WT'}], meta: {configureKey: 1}, configure: async (device, coordinatorEndpoint, logger) => { const endpoint = device.getEndpoint(1); await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genTime', 'genBasic', 'ssIasAce', 'ssIasZone']); }, But this is no longer in the Zigbee-Herdsman converters

Koenkk, 2022-08-30:
> But this is no longer in the Zigbee-Herdsman converters

I don't understand you, but it is there: https://github.com/Koenkk/zigbee-herdsman-converters/blob/54cd3f92d683a06d2b5ce1884ef72345799129a6/devices/tuya.js#L1224

ijs1550, 2022-09-01:
works

closed 2022-09-01 by ijs1550