Koenkk/zigbee2mqtt #13767: [New device support]: Hiking DDS238-2 Zigbee

kirillec, 2022-08-30:
### Link

https://aliexpress.ru/item/1005002409588154.html?spm=a2g2w.orderdetails.0.0.13cd4aa6REhO1V&sku_id=12000020560541066&_ga=2.200703146.569687716.1661863146-1629532030.1622346183

### Database entry

{"id":34,"type":"Router","ieeeAddr":"0x84b4dbfffeb31339","nwkAddr":44869,"manufId":4098,"manufName":"_TZE200_bkkmqmyo","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�+�*i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_bkkmqmyo","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1661865626295,"defaultSendRequestWhen":"immediate"}

### Comments

I bought a Hiking DDS238-2 Zigbee energy meter (Ver.IVAP is indicated on the device)
The Ver.IVAP device I received is not supported by Zigbee2mqtt latest available version 1.27.0-1.
I found a converter on the net, but the device did not work with it either ((
This is very strange, because this device is already considered an add-on, apparently the manufacturer has changed something in the circuit (
Please help!

Log:
Info: 2022-08-30 20:29:03MQTT publish: topic 'zigbee2mqtt/0x84b4dbfffeb31339', payload '{"last_seen":"2022-08-30T13:29:06.208Z","linkquality":0}'
Warning: 2022-08-30 20:29:04Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_bkkmqmyo'
Warning: 2022-08-30 20:29:04Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_bkkmqmyo'
        },
    ],
    model: 'TS0601_din',
    vendor: 'TuYa',
    description: 'Zigbee smart energy meter DDS238-2',
    fromZigbee: [fz.tuya_dinrail_switch],
    toZigbee: [tz.tuya_switch_state],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
    },
    exposes: [e.switch().setAccess('state', ea.STATE_SET), e.voltage(), e.power(), e.current(), e.energy()],
};

module.exports = definition;
```

kirillec, 2022-08-31:
Found some information online.
It seems to be like a new upgraded version of the device.

The manufacturer claims new function:
1. Positive and negative phase power + historical power
2. Power email export function
3. PV difference power display
4. Electricity bill display
5. Carbon emissions settings
6. Note that there is no temperature display, leakage protection, unbalance protection function, external power failure, the meter will not be switched off for protection
7. Manual button switch control
8. The function of comparing the power with the power of the total meter
9. Power curve display every 15 minutes for 24 hours of the day 1
10. Overvoltage and overcurrent protection settings
11. Operation records and failsafe records
12. Password protected settings

[image]
[image]

kirillec, 2022-09-01:
I'm trying to do something on my own, but since there is not enough knowledge in this topic, it turns out bad(
From what I managed to do, I changed the converter to this state:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_byzdayie'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_fsb6zw01'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_ewxhg6o9'},
        {modelID: 'TS0601', manufacturerName: '_TZE200_bkkmqmyo'}],           
    model: 'TS0601_din',
    vendor: 'TuYa',
    description: 'Zigbee smart energy meter DDS238-2 Zigbee',
    fromZigbee: [
        fz.tuya_dinrail_switch,
        fz.ignore_basic_report,
        fz.tuya_data_point_dump,
    ],

    toZigbee: [
        tz.tuya_switch_state,
        tz.tuya_data_point_test,
    ],
    onEvent: tuya.onEventSetTime,
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [e.switch().setAccess('state', ea.STATE_SET), e.voltage(), e.power(), e.current(), e.energy()],
};
module.exports = definition;
```
I'm sorry if any of this looks stupid)
In response to this, the z2m log issued the following:
```
Zigbee2MQTT:warn  2022-09-01 20:40:53: zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #16 with data {"dp":16,"datatype":1,"data":{"type":"Buffer","data":[1]}}
Zigbee2MQTT:info  2022-09-01 20:40:53: zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint #16 from 0x84b4dbfffeb31339 with raw data '{"dp":16,"datatype":1,"data":{"type":"Buffer","data":[1]}}': type='commandDataReport', datatype='bool', value='true', known DP# usage: ["moesHeatingSetpoint","tvHeatingSetpoint","haozeeMaxTemp","AM02Border","x5hSetTemp","connecteHeatingSetpoint","alectoSilence"]
Zigbee2MQTT:info  2022-09-01 20:40:53: MQTT publish: topic 'zigbee2mqtt/0x84b4dbfffeb31339', payload '{"current":null,"energy":null,"last_seen":"2022-09-01T13:40:53.814Z","linkquality":0,"power":null,"state":"ON","voltage":null}'
Error: ENOENT: no such file or directory, open 'data/tuya.dump.txt'
```
After that, z2m goes into reboot and so on in a circle((
Where can i find this file "data/tuya.dump.txt"

Knowledgeable people tell me what to do next to bring the matter to the end

gericho, 2022-09-20:
Same boat here.

vladlaw, 2022-09-21:
I have a similar problem, version ver.IVAP
Please add new version support
_TZE200_bkkmqmyo

`{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0,0,19,135],\"type\":\"Buffer\"},\"datatype\":2,\"dp\":105}],\"seq\":48128}' from endpoint 1 with groupID 0"}
{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0,5,0,100,0,5,0,30,0,60,0,0,0,0,0,0,0],\"type\":\"Buffer\"},\"datatype\":0,\"dp\":18}],\"seq\":47872}' from endpoint 1 with groupID 0"}
{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0,0,0,0,0,0,0,0,0,0,0,100,0,1,14,0,0,170,0,0,0,0,0,10,0,0,0,0,0,0],\"type\":\"Buffer\"},\"datatype\":0,\"dp\":17}],\"seq\":47616}' from endpoint 1 with groupID 0"}
{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0],\"type\":\"Buffer\"},\"datatype\":5,\"dp\":9}],\"seq\":47360}' from endpoint 1 with groupID 0"}
{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0,0,1,182],\"type\":\"Buffer\"},\"datatype\":2,\"dp\":111}],\"seq\":47104}' from endpoint 1 with groupID 0"}
{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0,0,0,50],\"type\":\"Buffer\"},\"datatype\":2,\"dp\":110}],\"seq\":46848}' from endpoint 1 with groupID 0"}
{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0,0,0,62],\"type\":\"Buffer\"},\"datatype\":2,\"dp\":109}],\"seq\":46592}' from endpoint 1 with groupID 0"}
{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0,0,0,36],\"type\":\"Buffer\"},\"datatype\":2,\"dp\":103}],\"seq\":46336}' from endpoint 1 with groupID 0"}
{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0,0,0,0],\"type\":\"Buffer\"},\"datatype\":2,\"dp\":102}],\"seq\":46080}' from endpoint 1 with groupID 0"}
{"level":"debug","message":"Received Zigbee message from '0x84b4dbfffeb8a11e', type 'commandDataReport', cluster 'manuSpecificTuya', data '{\"dpValues\":[{\"data\":{\"data\":[0,0,0,147],\"type\":\"Buffer\"},\"datatype\":2,\"dp\":101}],\"seq\":45824}' from endpoint 1 with groupID 0"}`
[image]
[image]

gericho, 2022-09-22:
exactly same errors here, the non IVAP version works perfectly, I contacted the vendor, they said they "improved" the firmware....

gericho, 2022-09-24:
updated to last EDGE but still no luck

Koenkk, 2022-09-25:
- Can someone provide me the data/database.db entry of this device?
- Can someone figure out all the datapoint values of this device? To do this follow: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#instructions

gericho, 2022-09-25:
Thank you so much for your effort! This is the database entry

`{"id":33,"type":"Router","ieeeAddr":"0x84b4dbfffeb5f542","nwkAddr":3373,"manufId":4098,"manufName":"_TZE200_bkkmqmyo","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"   *i   *iʷ *i˷ *i̷ *i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_bkkmqmyo","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9f1e4","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1664092687707,"defaultSendRequestWhen":"immediate"}
`

just a sec, I'll send you the datapoints as per your guide

gericho, 2022-09-25:
Please find below the datapoints, please note home assistant with EDGE shows altready 3 values that are : switch.powermeter (always on, not switchable), sensor.powermeter_energy (correct), sensor.powermeter_power (correct)
PLEASE NOTE, OLD MODEL of this meter HAVE "Electric Current (A)" value published, that is very important IMHO, this one seems to not have it but the app shows it, How is it possible? (see screenshot below) Thank you!

105 	frequency (translated from chinese) - value 49.99Hz
101 	total active power (translated from chinese) - value 123.70kWh
6	Phase A - value is garbage
102   Reverse active power (translated from chinese) - value is 0.00kWh
103 	total active power (translated from chinese) - value is 0.19KW
9	Fault - value is 0
109	total reactive power (translated from chinese) - value is 14.24kvarh
110   total reactive power (translated from chinese) - value is 0.03kvar
111	total power factor (translated from chinese) - value is 0.82
16	Switch - value ON  (please note this shows already in in home assisstant with EDGE, but it doesn't work properly, always active, cannot be turned off)
17	Alarm set1 - value seems garbage "AAAAAAAAAAAAAABkAAEOAACqAAAAAAAKAAAAAAAA"
18	Alarm set2 - value seems garbage "AAUAZAAFAB4APAAAAAAAAAA="
1	Total forward energy - value is 123.79kW-h

[image]

kirillec, 2022-09-29:
Thank you! Tonight after work I will definitely check)

> Could you check if the device works with the following external converter: https://gist.github.com/Koenkk/34e60391471f8a6333176594dd80445d
> . . .
> I did not add all the datapoints yet, if you want you can add them yourself, just add the datapoint to list on line 119 and add an exposes on line 117

gericho, 2022-09-29:
`[10:32:34] INFO: Prepare Zigbee2MQTT start...
[10:32:41] INFO: Handing over control to Zigbee2MQTT Core ...
/app/node_modules/zigbee-herdsman-converters/index.js:23
    zigbeeModel = zigbeeModel ? zigbeeModel.toLowerCase() : null;
                                            ^
TypeError: zigbeeModel.toLowerCase is not a function
    at addToLookup (/app/node_modules/zigbee-herdsman-converters/index.js:23:45)
    at Object.addDefinition [as addDeviceDefinition] (/app/node_modules/zigbee-herdsman-converters/index.js:110:13)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:15:17)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:108:18)`

Th3M1k3y, 2022-10-01:
I tried to add my DDS238-2 with the _TZE200_bkkmqmyo as manufacturer to Z2M 1.28.0, it got added and said supported, but still gave me the NOT RECOGNIZED messages.

I then tried to add the external converter and it got most of it working, the only problem seems to be the voltage being shown as 0V and getting `Exception while calling fromZigbee converter: dpEntry[2].from is not a function}` in log.

Koenkk, 2022-10-04:
This device is now properly supported.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-04 by Koenkk

mihaimdinca, 2023-01-10:
Hello, I have the same device and after a day or so seems to be working well. The problem is I cannot find a way to set the overvoltage and overcurrent alarms. I understand that they are not yet supported in z2m but does anyone know of another way to set them (using the button maybe). I do not own a Tuya ZigBee Hub. I bought the device to use it with z2m. Thank you in advance.

mihaimdinca, 2023-01-10:
Data on the DataPoints labeled as "garage" earlier looks similar two what was identified in this post https://github.com/jasonacox/tinytuya/issues/55

DominikWrobel, 2023-02-11:
> Thank you! Tonight after work I will definitely check)

Hey, I have the same problem with a WiFi meter using local tuya, how can I change the script to work with a wifi device?

crisc2000, 2024-02-15:
Is the problem with this device fixed (I mean the IVAP version).

Strange that the link on AliExpress
https://www.aliexpress.com/item/1005002409588154.html
do not show IVAP on the pictures.

I'm searching for a zigbee bidirectional energy-meter