Koenkk/zigbee2mqtt #13772: [New device support]: Lidl Silvercrest TS011F

radaoggy, 2022-08-30:
### Link

https://www.lidl.cz/p/silvercrest-zigbee-3-0-smart-home-inteligentni-zasuvka/p100355084

### Database entry

{"id":31,"type":"Router","ieeeAddr":"0xa4c138f46e448944","nwkAddr":45281,"manufId":4417,"manufName":"_TZ3000_ynmowqk2","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","zclVersion":3}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":0,"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1661888036837,"defaultSendRequestWhen":"immediate"}

### Comments

Hi, I bought SilverCrest smart plug model TS011F ([https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html](https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html)), but it shows as "Not supported", because it has different manufacture ID (**_TZ3000_ynmowqk2**). Maybe new batch? I don't know... Is it possible to make this plug work? Thank you.

closed 2022-08-31 by Koenkk

radaoggy, 2022-08-31:
Thanks! Device is now shows as "**Supported**". I can turn it on and off. Anyway this plug supports energy monitor, but it's not there. When I tried add external converter, it works well. I found this configuration below:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
{
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_ynmowqk2'}],
        model: 'TS011F_plug',
        description: 'Smart plug (with power monitoring)',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement']);
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {
                acVoltageMultiplier: 1, acVoltageDivisor: 1, acCurrentMultiplier: 1, acCurrentDivisor: 1000, acPowerMultiplier: 1,
                acPowerDivisor: 1,
            });
        },
        // This device doesn't support reporting correctly.
        // https://github.com/Koenkk/zigbee-herdsman-converters/pull/1270
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage')],
        onEvent: (type, data, device, options) => {
            const endpoint = device.getEndpoint(1);
            if (type === 'stop') {
                clearInterval(globalStore.getValue(device, 'interval'));
                globalStore.clearValue(device, 'interval');
            } else if (!globalStore.hasValue(device, 'interval')) {
                const seconds = options && options.measurement_poll_interval ? options.measurement_poll_interval : 60;
                if (seconds === -1) return;
                const interval = setInterval(async () => {
                    try {
                        await endpoint.read('haElectricalMeasurement', ['rmsVoltage', 'rmsCurrent', 'activePower']);
                    } catch (error) {/* Do nothing*/}
                }, seconds*1000);
                globalStore.putValue(device, 'interval', interval);
            }
        },
    }
];
```

It's not originally for Lidl SilverCrest, but works well. Maybe some adjustments is needed. I'm noobie in this and just try some solution found on google.

jeremynieuwpoort, 2023-11-02:
Hello,
Many this is the wrong place to ask something about this product but I tried everything. 

Connecting this product with mqtt works fine, but it creates multiple entities (L1) (L2) (L3). When exporting to HomeKit it will not show as a power strip but as tree separate switches.  

Can someone help please?