Koenkk/zigbee2mqtt #13777: TuYa [TV02]  - TS0601 _TZE200_hue3yfsn - error "No converter available" when setting schedule or holiday_start_stop

roobbb, 2022-08-31:
### What happened?

Device comes with white label "EARU TV02" 

Publishing values for "schedule" or "holiday_start_stop" as described on the device's page, always gives an error: 
`No converter available for 'undefined' ...`

I am not able to program the device.

Device page:
schedule: https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#schedule-composite
holiday_start_stop: https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#holiday-mode-date-composite

Am I doing something wrong?

### What did you expect to happen?

Setting values for the device's schedule or holiday_start_stop works and no error message appears.

### How to reproduce it (minimal and precise)

publish something like that:
`zigbee2mqtt/myThermostat/set { "undefined": {"holiday_start_stop": "start --> 2022 - 09 - 01 01 : 01 stop --> 2022 - 09 - 10 01 : 01"} }`

or 
`zigbee2mqtt/myThermostat/set {"undefined":{"schedule_monday":"6h:0m 20°C, 12h:0m 20°C, 14h:0m 20°C, 17h:0m 20°C, 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C","schedule_tuesday":"6h:0m 20°C, 12h:0m 20°C, 14h:0m 20°C, 17h:0m 20°C, 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C","schedule_wednesday":"6h:0m 20°C, 12h:0m 20°C, 14h:0m 20°C, 17h:0m 20°C, 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C","schedule_thursday":"6h:0m 20°C, 12h:0m 20°C, 14h:0m 20°C, 17h:0m 20°C, 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C","schedule_friday":"6h:0m 20°C, 12h:0m 20°C, 14h:0m 20°C, 17h:0m 20°C, 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C","schedule_saturday":"6h:0m 20°C, 12h:0m 20°C, 14h:0m 20°C, 17h:0m 20°C, 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C","schedule_sunday":"6h:0m 20°C, 12h:0m 20°C, 14h:0m 20°C, 17h:0m 20°C, 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C , 24h:0m 20°C"}}`

It is actually not clear to me how set such values the right way. I just fail setting {"undefined": {}}

### Zigbee2MQTT version

1.27.0 commit: a9b8808

### Adapter firmware version

20210218

### Adapter

ZigStar LAN Gateway

### Debug log

```
debug 2022-08-31 12:21:47Received MQTT message on 'zigbee2mqtt/myThermostat/set' with data '{ "undefined": {"holiday_start_stop": "set"} }'
error 2022-08-31 12:21:47No converter available for 'undefined' ({"holiday_start_stop":"set"})
```

apostolos-martzios, 2022-10-01:
I have exactly the same issue

roobbb, 2022-10-05:
@vladi1234 pulled a request for holiday_start_stop here ( https://github.com/Koenkk/zigbee-herdsman-converters/issues/3142 )
I am not sure when it will be rolled out. But many thanks to him 😀

Meanwhile I found out the get statements even do not work as stated on TV02's documentation site. 
`To read send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"current_heating_setpoint": ""}.`
`To read send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"preset": ""}.`

When I publish these ones the Log tells me:
`No converter available for 'get' 'current_heating_setpoint' ()`
`No converter available for 'get' 'preset' ()`

Maybe the TS0601 _TZE200_hue3yfsn works a bit different as the other TV02 models? I'm not sure what is missing.

Kind regards
roobbb

roobbb, 2022-10-06:
Hello vladi1234.

Thank you for letting me know. Actually I tried both (and several strange things more I've found for other thermostats 😁 ). 
Now I have running the latest version and tried again. Yes, setting "holiday_start_stop" works fine now 😀

The thing with schedule isn't that easy, right?

Do you have any idea how to perform a get for "current_heating_setpoint" or  "preset"?

Thank you very much and kind regards
roobbb

vladi1234, 2022-10-06:
Hello @roobbb,

it works the same way, only with `get`.

`zigbee2mqtt/myThermostat/get {"current_heating_setpoint": ""}`

Everything is well described here.
https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee

roobbb, 2022-10-06:
unfortunately it throws me:
`debug 2022-10-06 11:24:31 Received MQTT message on 'zigbee2mqtt/Earu_Thermostat01/get' with data '{"current_heating_setpoint": ""`
`error 2022-10-06 11:24:31 No converter available for 'get' 'current_heating_setpoint' ()`

> Everything is well described here. https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee

Of course it is. And I already read it carefully. But some of those information do not work with my EARU Thermostat. Maybe it does some things a bit different than the other devices reported under TV02. I don't know.

I only have problems passing these statements as described in the documentation:
`Climate : zigbee2mqtt/FRIENDLY_NAME/get {"preset": ""}`
`Climate : zigbee2mqtt/FRIENDLY_NAME/get {"current_heating_setpoint": ""}`
`Schedule (composite) : zigbee2mqtt/FRIENDLY_NAME/set {"undefined": {}}`

All other statements work fine.

Kind regards
roobbb

vladi1234, 2022-10-06:
Everything is well described here.
https://www.zigbee2mqtt.io/devices/TV02-Zigbee.html#tuya-tv02-zigbee
Schedule (composite) : zigbee2mqtt/FRIENDLY_NAME/set {"undefined": {}}
https://github.com/Koenkk/zigbee-herdsman-converters/issues/3142#issuecomment-1026078736

Yes, it's true `'get'` for `{"preset": ""}` and `{"current_heating_setpoint":""}` does not work.
But I do not understand why you need this?
Thermostat itself supplies these `{"preset": ""}` and `{"current_heating_setpoint":""}`.
With `'set'`, Thermostat also provides these values.

roobbb, 2022-10-10:
Well, I just went through the official documentation you mentioned above, configured my smart home controller based on the information found and fell over those "get" entries.
You are right, it is not that important having get-abilities. The data come in periodically. I guess "get" could be useful when someone needs to pull data outside the regular push-cycle.

Because "get" is not possible, I think it would be helpful when this information could even be placed on device's page to avoid confusion.
What do you think?

Best regards
roobbb

roobbb, 2022-10-10:
Thank you very much. I really like your work 😀

closed 2022-11-18 by github-actions[bot]

lukima61, 2023-03-12:
Hello,
has the problem with _Schedule_ been solved in the meantime? I have the same error message and would appreciate the solution description. 
I found out that _working_day_ is not set to _mon_sun_, _mon_fri+sat+sun_ or _separate_, but to 1, 2 or 3. All other values are ignored and _working_day_ is set to 0.

yoshibz, 2023-11-12:
Hello,
has the problem with Schedule been solved in the meantime? I have the same error message and would appreciate the solution description for TS0601_thermostat.