Koenkk/zigbee2mqtt #13787: Ugly OTA message: No image available for imageType '5634'

inrepublica, 2022-09-01:
### What happened?

When asking for updates, an ugly error appears. I don't know if this is expected, but seems more like a no controlled error. The device is a TS011F_plug_3.
[https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html#tuya-ts011f_plug_3](url)

### How to reproduce it (minimal and precise)

Simply go to OTA page and press to ckeck for updates.

### Zigbee2MQTT version

1.27.1-1

### Adapter firmware version

20210708

### Adapter

zStack3x0

### Debug log

```
error Failed to check if update available for 'interrupteur_chauffage_lit_cecilia' (No image available for imageType '5634')
error Failed to check if update available for 'interrupteur_chauffage_lit_micka' (No image available for imageType '5634')
```

Koenkk, 2022-09-05:
This means we don't have an image available for this plug. To retrieve it follow: https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html

inrepublica, 2022-09-07:
Ah ok, i don't have Xiaomi or tuya gateway. Thanks for your work 👍

closed 2022-09-07 by inrepublica