Koenkk/zigbee2mqtt #13788: 1.27.1 wont work properly

b-reich, 2022-09-01:
### What happened?

Zigbee2mqtt doenst start properly.

No error in the log.
Frontend doenst work.
No Device avaible in mqtt.

Fix: Revertiung back to 1.27.0

_Originally posted by @b-reich in https://github.com/Koenkk/zigbee2mqtt/issues/13382#issuecomment-1233890344_

### How to reproduce it (minimal and precise)

Running the latest (1.27.1) Docker Container

### Zigbee2MQTT version

1.27.1

### Adapter firmware version

20211217

### Adapter

Slaesh CC2652RB 

### Debug log

debug 2022-09-01 09:46:16: Loaded state from file /app/data/state.json
info  2022-09-01 09:46:16: Logging to console and directory: '/app/data/log/2022-09-01.09-46-16' filename: log.txt
debug 2022-09-01 09:46:16: Removing old log directory '/app/data/log/2022-08-04.23-22-24'
info  2022-09-01 09:46:16: Starting Zigbee2MQTT version 1.27.1 (commit #d381f2d)
info  2022-09-01 09:46:16: Starting zigbee-herdsman (0.14.51)
debug 2022-09-01 09:46:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyUSB0"}}'

b-reich, 2022-09-01:
More info

docker-compose
```yaml
  version: '3'
  services:
    zigbee2mqtt:
      container_name: zigbee2mqtt
      image: koenkk/zigbee2mqtt:1.27.0
      volumes:
        - ./data:/app/data
        - /run/udev:/run/udev:ro
      devices:
        - /dev/ttyUSB0:/dev/ttyUSB0
      restart: always
      network_mode: host
      privileged: true
      environment:
        - TZ=Europe/Berlin
```

config:
```yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://share.lan
  keepalive: 60
  reject_unauthorized: true
  version: 5
  include_device_information: true
serial:
  port: /dev/ttyUSB0
devices:
  <reduced>
groups:
 <reduced>
advanced:
  report: true
  pan_id: 6755
  last_seen: ISO_8601_local
  log_level: debug
  ikea_ota_use_test_url: false
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
frontend:
  port: 8080
  host: 0.0.0.0
availability: true
```

ggottwald, 2022-09-01:
For me it can't connect to the local mqtt server.
I found this in the logs
```
Zigbee2MQTT:error 2022-09-01 10:23:06: Not connected to MQTT server!
Zigbee2MQTT:error 2022-09-01 10:23:06: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
Zigbee2MQTT:info  2022-09-01 10:23:06: Disconnecting from MQTT server
/opt/zigbee2mqtt/lib/mqtt.ts:126
        this.client.end();
                    ^
TypeError: Cannot read properties of undefined (reading 'end')
    at MQTT.disconnect (/opt/zigbee2mqtt/lib/mqtt.ts:126:21)
    at Controller.stop (/opt/zigbee2mqtt/lib/controller.ts:197:9)
    at stop (/opt/zigbee2mqtt/index.js:113:5)
    at process.handleQuit (/opt/zigbee2mqtt/index.js:119:9)
```

capu-bia, 2022-09-01:
Same problem here without docker...
same log output too

Koenkk, 2022-09-01:
Can you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

capu-bia, 2022-09-01:
zigbee-herdsman:adapter Path '/dev/ttyUSB0' is valid for 'EZSPAdapter' +0ms

 zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:ezsp:uart Serialport opened +20ms
  zigbee-herdsman:adapter:ezsp:uart Uart reseting +2ms
  zigbee-herdsman:adapter:ezsp:uart --> Write reset +1ms
  zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e] +0ms

Dylantje, 2022-09-01:
Here also.. running on "normal" linux installation..
`
sep 01 10:58:53 odroidn2 npm[7559]:     at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
sep 01 10:58:53 odroidn2 npm[7559]:     at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:198:14)
sep 01 10:58:53 odroidn2 npm[7559]:     at /opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:242:12
sep 01 10:58:53 odroidn2 systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
sep 01 10:58:53 odroidn2 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
sep 01 10:58:53 odroidn2 systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 29.
sep 01 10:58:53 odroidn2 systemd[1]: Stopped zigbee2mqtt.
sep 01 10:58:53 odroidn2 systemd[1]: Started zigbee2mqtt.
sep 01 10:58:54 odroidn2 npm[7717]: > zigbee2mqtt@1.27.1 start
sep 01 10:58:54 odroidn2 npm[7717]: > node index.js
sep 01 10:58:57 odroidn2 npm[7738]: Zigbee2MQTT:info  2022-09-01 10:58:57: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-09-01.10-58-57' filename: log.txt
sep 01 10:58:57 odroidn2 npm[7738]: Zigbee2MQTT:info  2022-09-01 10:58:57: Starting Zigbee2MQTT version 1.27.1 (commit #d381f2d)
sep 01 10:58:57 odroidn2 npm[7738]: Zigbee2MQTT:info  2022-09-01 10:58:57: Starting zigbee-herdsman (0.14.51)
sep 01 10:58:58 odroidn2 npm[7738]: Zigbee2MQTT:error 2022-09-01 10:58:58: Error while starting zigbee-herdsman
sep 01 10:58:58 odroidn2 npm[7738]: Zigbee2MQTT:error 2022-09-01 10:58:58: Failed to start zigbee
sep 01 10:58:58 odroidn2 npm[7738]: Zigbee2MQTT:error 2022-09-01 10:58:58: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
sep 01 10:58:58 odroidn2 npm[7738]: Zigbee2MQTT:error 2022-09-01 10:58:58: Exiting...
sep 01 10:58:58 odroidn2 npm[7738]: Zigbee2MQTT:error 2022-09-01 10:58:58: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB.CC2531-01'
sep 01 10:58:58 odroidn2 npm[7738]:     at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
sep 01 10:58:58 odroidn2 npm[7738]:     at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:198:14)
sep 01 10:58:58 odroidn2 npm[7738]:     at /opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:242:12
sep 01 10:58:58 odroidn2 systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
sep 01 10:58:58 odroidn2 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
sep 01 10:58:58 odroidn2 systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 30.
sep 01 10:58:58 odroidn2 systemd[1]: Stopped zigbee2mqtt.
sep 01 10:58:58 odroidn2 systemd[1]: Started zigbee2mqtt.
sep 01 10:58:59 odroidn2 npm[7895]: > zigbee2mqtt@1.27.1 start
sep 01 10:58:59 odroidn2 npm[7895]: > node index.js
`

Did change serial code like the wiki says. Only no luck @ all.

Koenkk, 2022-09-01:
@capu-bia  @kirovilya can you check https://github.com/Koenkk/zigbee2mqtt/issues/13788#issuecomment-1233974599 , maybe something broke in the EZSP integration?

@Dylantje  ` No such file or directory, cannot open /dev/ttyUSB.CC2531-01'`  -> wrong path in your `configuration.yaml` (or your CC2531 adapter crashed. Replug it and make sure the path is correct: https://www.zigbee2mqtt.io/guide/installation/01_linux.html#determine-location-of-the-adapter-and-checking-user-permissions

b-reich, 2022-09-01:
```
Zigbee2MQTT:debug 2022-09-01 11:03:54: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2022-09-01 11:03:54: Logging to console and directory: '/app/data/log/2022-09-01.11-03-54' filename: log.txt
Zigbee2MQTT:debug 2022-09-01 11:03:54: Removing old log directory '/app/data/log/2022-08-08.14-18-17'
Zigbee2MQTT:info  2022-09-01 11:03:54: Starting Zigbee2MQTT version 1.27.1 (commit #d381f2d)
Zigbee2MQTT:info  2022-09-01 11:03:54: Starting zigbee-herdsman (0.14.51)
Zigbee2MQTT:debug 2022-09-01 11:03:54: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyUSB0"}}'
2022-09-01T09:03:55.200Z zigbee-herdsman:adapter Path '/dev/ttyUSB0' is valid for 'EZSPAdapter'
2022-09-01T09:03:55.203Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6755,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"disableLED":false,"concurrent":null,"delay":null}}'
2022-09-01T09:03:55.206Z zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2022-09-01T09:03:55.218Z zigbee-herdsman:adapter:ezsp:uart Serialport opened
2022-09-01T09:03:55.219Z zigbee-herdsman:adapter:ezsp:uart Uart reseting
2022-09-01T09:03:55.219Z zigbee-herdsman:adapter:ezsp:uart --> Write reset
2022-09-01T09:03:55.221Z zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e]
```

Koenkk, 2022-09-01:
@b-reich also using EZSP so it looks there is some problem there.

@taimaiduc also using EZSP?

Dylantje, 2022-09-01:
Set back my old settings

My debug:
` 
pi@odroidn2:/opt/zigbee2mqtt$ npm start

> zigbee2mqtt@1.27.1 start
> node index.js

Zigbee2MQTT:debug 2022-09-01 11:09:02: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-09-01 11:09:02: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-09-01.11-09-02' filename: log.txt
Zigbee2MQTT:debug 2022-09-01 11:09:02: Removing old log directory '/opt/zigbee2mqtt/data/log/2022-09-01.11-05-19'
Zigbee2MQTT:info  2022-09-01 11:09:02: Starting Zigbee2MQTT version 1.27.1 (commit #d381f2d)
Zigbee2MQTT:info  2022-09-01 11:09:02: Starting zigbee-herdsman (0.14.51)
Zigbee2MQTT:debug 2022-09-01 11:09:02: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB2"}}'
Zigbee2MQTT:error 2022-09-01 11:09:03: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-01 11:09:03: Failed to start zigbee
Zigbee2MQTT:error 2022-09-01 11:09:03: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-01 11:09:03: Exiting...
Zigbee2MQTT:error 2022-09-01 11:09:03: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/uart.ts:90:28)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:242:12
pi@odroidn2:/opt/zigbee2mqtt$
.
`

b-reich, 2022-09-01:
@Koenkk can I change something? I run the stick from Slaesh (CC2652R). 1.27.0 was working for me.

liambrugaro, 2022-09-01:
Hello @Koenkk 

here are my debug logs.
i use a Slaesh CC2652RB stick

Zigbee2MQTT:debug 2022-09-01 11:10:28: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2022-09-01 11:10:28: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-09-01.11-10-28' filename: log.txt
Zigbee2MQTT:debug 2022-09-01 11:10:28: Removing old log directory '/opt/zigbee2mqtt/data/log/2022-09-01.10-05-48'
Zigbee2MQTT:info  2022-09-01 11:10:28: Starting Zigbee2MQTT version 1.27.1 (commit #d381f2da)
Zigbee2MQTT:info  2022-09-01 11:10:28: Starting zigbee-herdsman (0.14.51)
Zigbee2MQTT:debug 2022-09-01 11:10:28: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":true},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":49649},"serialPort":{"path":"/dev/ttyUSB0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyUSB0' is valid for 'EZSPAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[61,218,195,151,190,217,29,113,248,53,113,126,127,46,69,71],"panID":49649,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"/dev/ttyUSB0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"disableLED":true,"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:ezsp:uart Opening SerialPort with /dev/ttyUSB0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:ezsp:uart Serialport opened +15ms
  zigbee-herdsman:adapter:ezsp:uart Uart reseting +0ms
  zigbee-herdsman:adapter:ezsp:uart --> Write reset +0ms
  zigbee-herdsman:adapter:ezsp:uart --> [1ac038bc7e] +0ms

Koenkk, 2022-09-01:
@b-reich in your z2m configuration.yaml can you force the adapter to zstack?

```
serial:
  port: /dev/ttyACM0
  adapter: zstack # <--- add this
```

liambrugaro, 2022-09-01:
> @b-reich in your z2m configuration.yaml can you force the adapter to zstack?
> 
> ```
> serial:
>   port: /dev/ttyACM0
>   adapter: zstack # <--- add this
> ```

thanks, it works for me :)

capu-bia, 2022-09-01:
> @b-reich in your z2m configuration.yaml can you force the adapter to zstack?
> 
> ```
> serial:
>   port: /dev/ttyACM0
>   adapter: zstack # <--- add this
> ```

Working for me too!

b-reich, 2022-09-01:
> @b-reich in your z2m configuration.yaml can you force the adapter to zstack?
> 
> ```
> serial:
>   port: /dev/ttyACM0
>   adapter: zstack # <--- add this
> ```

this works for me

Dylantje, 2022-09-01:
What to do for this adapter?:
`
pi@odroidn2:~$ ls -l /dev/serial/by-id
total 0
lrwxrwxrwx 1 root root 13 sep  1 11:06 usb-0658_0200-if00 -> ../../ttyACM0
lrwxrwxrwx 1 root root 13 sep  1 11:06 usb-FTDI_FT232R_USB_UART_AI05V71S-if00-port0 -> ../../ttyUSB0
lrwxrwxrwx 1 root root 13 sep  1 11:06 usb-RFXCOM_RFXtrx433_A1XSS661-if00-port0 -> ../../ttyUSB1
lrwxrwxrwx 1 root root 13 sep  1 11:06 usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_21_CC_38_6F-if00-port0 -> ../../ttyUSB2
`

b-reich, 2022-09-01:
@Dylantje a code block would be much more readable :)

liambrugaro, 2022-09-01:
> What to do for this adapter?: `pi@odroidn2:~$ ls -l /dev/serial/by-id total 0 lrwxrwxrwx 1 root root 13 sep 1 11:06 usb-0658_0200-if00 -> ../../ttyACM0 lrwxrwxrwx 1 root root 13 sep 1 11:06 [...]

you're using the same adapter than us. try adding "adapter: zstack"

Koenkk, 2022-09-01:
1.27.2 incoming

Dylantje, 2022-09-01:
What is the best option to do??
Need to add, adapter?
Or wait to 1.7.2 or both :-)
 
Love YOU dutchman koen :-) What a perfect system you build THANKS!!!!

Koenkk, 2022-09-01:
1.27.2 should be there in a few mins, please check if it fixes the issue (working **without** `adapter: zstack` in the `configuraiton.yaml`)

Dylantje, 2022-09-01:
Works fine [ until now] did ad adapter and update

HERO

Thanks

b-reich, 2022-09-01:
@Koenkk Thanks for the real quick fix!
1.27.2 just works fine.

I close this issue.

closed 2022-09-01 by b-reich

capu-bia, 2022-09-01:
@Koenkk 1.27.2 is working fine
thanks for the very fast response!!

Koenkk, 2022-09-01:
Thanks for reporting & enjoy 😄

ali-84, 2024-05-26:
Hi, I can't start it at all, what should I do?

[15:08:18] INFO: Preparing to start...
[15:08:18] INFO: Socat not enabled
[15:08:21] INFO: Starting Zigbee2MQTT...
[2024-05-26 15:08:28] info: 	z2m: Logging to console, file (filename: log.log)
[2024-05-26 15:08:28] info: 	z2m: Starting Zigbee2MQTT version 1.37.1 (commit #unknown)
[2024-05-26 15:08:28] info: 	z2m: Starting zigbee-herdsman (0.46.6)
[2024-05-26 15:08:28] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-05-26 15:08:28] info: 	zh:zstack:znp: Serialport opened
[2024-05-26 15:08:28] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-05-26 15:08:30] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-05-26 15:08:48] error: 	z2m: Error while starting zigbee-herdsman
[2024-05-26 15:08:48] error: 	z2m: Failed to start zigbee
[2024-05-26 15:08:48] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-05-26 15:08:48] error: 	z2m: Exiting...
[2024-05-26 15:08:48] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:101:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:124:29)
    at Zigbee.start (/app/lib/zigbee.ts:62:27)
    at Controller.start (/app/lib/controller.ts:109:27)
    at start (/app/index.js:107:5)