Koenkk/zigbee2mqtt #13799: Develco Frient ZHEMI101 Energy Meter - Power Rate Adjust

Hoovercraft, 2022-09-01:
### What happened?

Installed a Develco Frient ZHEMI101 Energy Meter on my iskra mt175. Pairing with Z2M was no problem. 

### What did you expect to happen?

Changed the pulse rate to 10.000, as it is mentioned on the MT175. 

Imho the power rate must also be changed, but i could not find a possibility to do this.

I use about 5.200 KwH per year. Means 5200 / 365 /24 = ~ 0.6 KwH per hour.

But the power measurement shows me about 3 - 6 KwH each hour.

[image]

### How to reproduce it (minimal and precise)

unpaired device, paired again, result is allways the same

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

Slae.sh CC2652RB

rickgardner077, 2022-09-01:
Could you please clarify what you mean by "power rate"? I have the same device with a MT175 as well and would like to compare my settings

Hoovercraft, 2022-09-01:
First, sorry for my bad english, german speaker here.

I mean the power consumption, on the screenshot it shows 3200 watt. My MT175 shows only 320 watt.

So i think the used power must be devided by 10 as it is also done with the pulse ratio (changed it from 1000 to 10000).

rickgardner077, 2022-09-01:
Don't worry - I am German speaker as well

The power consumption is a result of the pulse ratio. I might happen that you get a false reading if you first paired the device with Home Assistant (that is what you are using right?) and adjusted the pulse ratio afterwards. It happened to me too. 

I resolved the issue by deleting the ZHEMI entirely and re-pairing it. Then I set the pulse ratio to 10000 first before I added it to Home Assistant. This way I get the correct readings.

If you like, check my post at https://rick.gr/2022/08/stromverbrauch-in-echtzeit-aufzeichnen-mit-home-assistant-frient-und-iskra-zaehler-mt175/ (in German though)

Hoovercraft, 2022-09-01:
hmm, i had found your post but i thought it couldn't help me.

Got one question: are you using an external Z2M? I use the addon in HA (via repository).

I just unpaired the device. Paired it again, changed pulse ratio to 10000 and choose 'energy', checked status in Z2M, attached the led to the MT175 and again i get ~4000 watt :-(

Hoovercraft, 2022-09-01:
no need to answer my question, just found the answer in your post ;-)

Okay, lets start over. 

I will remove the frient from Z2M. Do a reset, activate search in Z2M, when frient is found i change pulse ratio to 10.000 and interface to 'electricity'. Wait a minute and check status and cross fingers for a good result. 

Is this okay?

Hoovercraft, 2022-09-01:
still no luck _:-(

attached is the log from Z
[image]
2M

rickgardner077, 2022-09-01:
Another thought: Did you unpair the device in Z2M AND reset it to factory settings?

Yes, I use the same add-on that you do

rickgardner077, 2022-09-01:
Sollte das gleiche sein
[image]

Hoovercraft, 2022-09-01:
habe gerade einen Austausch beantragt. Mal sehen, ob der nächste mehr das macht was er soll. 
Dir auf jeden Fall erst mal vielen Dank !

closed 2022-09-03 by Hoovercraft