Koenkk/zigbee2mqtt #13800: Zigbee smart dimmer reported "Received unexpected Tuya DataPoint"

madmax1970, 2022-09-01:
### Link

https://smartguys.com.au/makegood-zigbee-dimmer-smart-light-switch.html

### Database entry

0x588e81fffed0d59d (0xA21F)

### Comments

Zigbee Model                  :     TS0601
Zigbee Manufacturer      :     _TZE200_1agwnems

On/Off - working
Dimmer - not working
brightness - reverts back to 0
min_brightness - reverts back to 0
max_brightness - reverts back to 0

Any help on how to fix this please.

State
{
    "brightness": -3,
    "level": 0,
    "linkquality": 255,
    "state": "ON"
}

Warning (plenty of this)
Zigbee2MQTT:warn  2022-09-01 20:32:00: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #14 from 0x588e81fffed0d59d with raw data '{"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum', value='0', known DP# usage: ["runningState","moesSbattery","wooxBattery","haozeeFaultAlarm","nousTempAlarm","moesSwitchPowerOnBehavior","alectoBatteryState"]
Zigbee2MQTT:warn  2022-09-01 20:32:00: zigbee-herdsman-converters:tuya_dimmer: Received Tuya DataPoint #14 with unexpected datatype from 0x588e81fffed0d59d with raw data '{"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum' (instead of 'undefined'), value='0', known DP# usage: ["runningState","moesSbattery","wooxBattery","haozeeFaultAlarm","nousTempAlarm","moesSwitchPowerOnBehavior","alectoBatteryState"]

shanelord01, 2022-09-02:
This is the same issue I've reported here https://github.com/Koenkk/zigbee2mqtt/issues/13697

Glad I'm not alone - hoping @Koenkk can help fix it :)

benflux, 2022-09-02:
Same here with the devices from this seller, hoping they are the same? @shanelord01 @madmax1970 

https://www.ozsmartthings.com.au/products/zigbee-dimmer-switch-white

shanelord01, 2022-09-02:
> Same here with the devices from this seller, hoping they are the same? @shanelord01 @madmax1970
> 
> https://www.ozsmartthings.com.au/products/zigbee-dimmer-switch-white

Yes these are the ones I use.

benflux, 2022-09-04:
Shouldn’t we be submitting a whole new device here, I mean the picture is currently a square device and these are rectangular?

shanelord01, 2022-09-04:
I don’t believe so. They are the same underlying technology/product just in different form factors.

benflux, 2022-09-10:
@shanelord01 i know this isn’t the right place to ask this but I can’t PM you. Do your dimmers always turn on at their brightest and then ramp down to the last remembered level when using the physical button? Surely there must be a setting to turn on to the last remembered level directly?

shanelord01, 2022-09-10:
> @shanelord01 i know this isn’t the right place to ask this but I can’t PM you. Do your dimmers always turn on at their brightest and then ramp down to the last remembered level when using the physical button? Surely there must be a setting to turn on to the last remembered level directly?

Yes mine do that too. Hoping it’s related to the Z2M issues.

Koenkk, 2022-09-12:
Let's first try to fix the brightness part, could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    tuya_dimmer_level: {
        key: ['brightness_min', 'min_brightness', 'max_brightness', 'brightness', 'brightness_percent', 'level'],
        convertSet: async (entity, key, value, meta) => {
            // upscale to 1000
            let newValue;
            let dp = tuya.dataPoints.dimmerLevel;
            if (['_TZE200_3p5ydos3', '_TZE200_9i9dt8is', '_TZE200_dfxkcots', '_TZE200_w4cryh2i', '_TZE200_1agwnems'].includes(meta.device.manufacturerName)) {
                dp = tuya.dataPoints.eardaDimmerLevel;
            }
            if (key === 'brightness_min') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                    dp = tuya.dataPoints.dimmerLevel;
                } else {
                    throw new Error('Dimmer brightness_min is out of range 0..100');
                }
            } else if (key === 'min_brightness') {
                if (value >= 1 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 1, 255, 0, 1000);
                    dp = tuya.dataPoints.dimmerMinLevel;
                } else {
                    throw new Error('Dimmer min_brightness is out of range 1..255');
                }
            } else if (key === 'max_brightness') {
                if (value >= 1 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 1, 255, 0, 1000);
                    dp = tuya.dataPoints.dimmerMaxLevel;
                } else {
                    throw new Error('Dimmer min_brightness is out of range 1..255');
                }
            } else if (key === 'level') {
                if (value >= 0 && value <= 1000) {
                    newValue = Math.round(Number(value));
                } else {
                    throw new Error('Dimmer level is out of range 0..1000');
                }
            } else if (key === 'brightness_percent') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness_percent is out of range 0..100');
                }
            } else { // brightness
                if (value >= 0 && value <= 254) {
                    newValue = utils.mapNumberRange(value, 0, 254, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness is out of range 0..254');
                }
            }
            // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
            await tuya.sendDataPointValue(entity, dp, newValue, 'dataRequest', 1);
        },
    }
}

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_1agwnems'},
    ],
    model: 'TS0601_dimmer',
    vendor: 'TuYa',
    description: 'Zigbee smart dimmer',
    fromZigbee: [fz.tuya_dimmer, fz.ignore_basic_report],
    toZigbee: [tz.tuya_dimmer_state, tzLocal.tuya_dimmer_level],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
    exposes: [e.light_brightness().withMinBrightness().withMaxBrightness().setAccess(
        'state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).setAccess(
        'min_brightness', ea.STATE_SET).setAccess('max_brightness', ea.STATE_SET)],
    whiteLabel: [
        {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
        {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
        {vendor: 'Mercator Ikuü', model: 'SSWD01'},
        {vendor: 'Moes', model: 'ZS-USD'},
    ],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

shanelord01, 2022-09-12:
The converter is definately loaded as I changed description: 'Zigbee smart dimmer', to description: 'Zigbee smart dimmer ext', and it is showing in the device in Z2M.

shanelord01, 2022-09-12:
```
Debug 2022-09-12 21:21:34Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}],"seq":26883}' from endpoint 1 with groupID null
Warning 2022-09-12 21:21:34zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #6 from 0x9035eafffecc7967 with raw data '{"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}': type='commandDataResponse', datatype='value', value='0', known DP# usage: ["silvercrestSetEffect","haozeeRunningState","evanellBattery","tshpsSelfTest"]
Warning 2022-09-12 21:21:34zigbee-herdsman-converters:tuya_dimmer: Received Tuya DataPoint #6 with invalid value 0 for brightness from 0x9035eafffecc7967 which is outside the expected range from 10 to 1000
Info 2022-09-12 21:21:34MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":-3,"level":0,"linkquality":255,"state":"OFF"}'
Debug 2022-09-12 21:21:34Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":27139}' from endpoint 1 with groupID null
Warning 2022-09-12 21:21:34zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #14 from 0x9035eafffecc7967 with raw data '{"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum', value='0', known DP# usage: ["runningState","moesSbattery","wooxBattery","haozeeFaultAlarm","nousTempAlarm","moesSwitchPowerOnBehavior","alectoBatteryState"]
Warning 2022-09-12 21:21:34zigbee-herdsman-converters:tuya_dimmer: Received Tuya DataPoint #14 with unexpected datatype from 0x9035eafffecc7967 with raw data '{"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum' (instead of 'undefined'), value='0', known DP# usage: ["runningState","moesSbattery","wooxBattery","haozeeFaultAlarm","nousTempAlarm","moesSwitchPowerOnBehavior","alectoBatteryState"]
Debug 2022-09-12 21:21:42Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":27139}' from endpoint 1 with groupID null
Info 2022-09-12 21:21:42MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":-3,"level":0,"linkquality":255,"state":"OFF"}'
Debug 2022-09-12 21:21:42Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,107],"type":"Buffer"},"datatype":2,"dp":2}],"seq":27395}' from endpoint 1 with groupID null
Warning 2022-09-12 21:21:42zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #2 from 0x9035eafffecc7967 with raw data '{"dp":2,"datatype":2,"data":{"type":"Buffer","data":[0,0,3,107]}}': type='commandDataResponse', datatype='value', value='875', known DP# usage: ["heatingSetpoint","coverPosition","eardaDimmerLevel","moesHold","moesSheatingSetpoint","silvercrestChangeMode","tuyaSabCO2","tuyaSahkMP25","tuyaSabCO","moes105DimmerLevel1","trsSensitivity","trsfSensitivity","tvMode","haozeeHeatingSetpoint","nousHumidity","tthHumidity","tIlluminanceLux","evanellMode","AM02PercentControl","x5hMode","connecteMode","tshpscSensitivity","alectoSmokeValue"]
Info 2022-09-12 21:21:42MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":222,"level":875,"linkquality":255,"state":"OFF"}'
Debug 2022-09-12 21:21:43Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":27651}' from endpoint 1 with groupID null
Warning 2022-09-12 21:21:43zigbee-herdsman-converters:tuya_dimmer: Received Tuya DataPoint #3 with invalid value 1 for brightness from 0x9035eafffecc7967 which is outside the expected range from 10 to 1000
Info 2022-09-12 21:21:43MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":-2,"level":1,"linkquality":255,"state":"OFF"}'
Debug 2022-09-12 21:21:43Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":27907}' from endpoint 1 with groupID null
Warning 2022-09-12 21:21:43zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #4 from 0x9035eafffecc7967 with raw data '{"dp":4,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum', value='0', known DP# usage: ["mode","moesSboostHeating","silvercrestSetColorTemp","haozeeBoostHeating","nousBattery","tthBattery","wlsBatteryPercentage","evanellHeatingSetpoint","AM02Mode","tshpsMaximumRange","lmsBattery"]
Warning 2022-09-12 21:21:43zigbee-herdsman-converters:tuya_dimmer: Received Tuya DataPoint #4 with unexpected datatype from 0x9035eafffecc7967 with raw data '{"dp":4,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum' (instead of 'undefined'), value='0', known DP# usage: ["mode","moesSboostHeating","silvercrestSetColorTemp","haozeeBoostHeating","nousBattery","tthBattery","wlsBatteryPercentage","evanellHeatingSetpoint","AM02Mode","tshpsMaximumRange","lmsBattery"]
[... cut ...] moesSwitchPowerOnBehavior","alectoBatteryState"]
Warning 2022-09-12 21:22:10zigbee-herdsman-converters:tuya_dimmer: Received Tuya DataPoint #14 with unexpected datatype from 0x9035eafffecc7967 with raw data '{"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum' (instead of 'undefined'), value='0', known DP# usage: ["runningState","moesSbattery","wooxBattery","haozeeFaultAlarm","nousTempAlarm","moesSwitchPowerOnBehavior","alectoBatteryState"]
```

Koenkk, 2022-09-13:
I made an error, updated https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1243346097

shanelord01, 2022-09-13:
No luck... brightness won't change anything (in fact it jumps up a down about 3 steps over and over again). On/Off works. Lots of red errors on join and usage though.

```
info  2022-09-13 18:10:56: Device '0x9035eafffecc7967' joined
info  2022-09-13 18:10:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffecc7967","ieee_address":"0x9035eafffecc7967"},"type":"device_joined"}'
info  2022-09-13 18:10:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x9035eafffecc7967"},"type":"device_connected"}'
info  2022-09-13 18:10:56: Starting interview of '0x9035eafffecc7967'
info  2022-09-13 18:10:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffecc7967","ieee_address":"0x9035eafffecc7967","status":"started"},"type":"device_interview"}'
info  2022-09-13 18:10:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x9035eafffecc7967"},"type":"pairing"}'
debug 2022-09-13 18:10:57: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_1agwnems","modelId":"TS0601"}' from endpoint 1 with groupID null
debug 2022-09-13 18:10:57: Skipping message, definition is undefined and still interviewing
debug 2022-09-13 18:10:57: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
debug 2022-09-13 18:10:57: Skipping message, definition is undefined and still interviewing
debug 2022-09-13 18:10:57: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
debug 2022-09-13 18:10:57: Skipping message, definition is undefined and still interviewing
debug 2022-09-13 18:10:57: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"appVersion":68}' from endpoint 1 with groupID null
debug 2022-09-13 18:10:57: Skipping message, definition is undefined and still interviewing
debug 2022-09-13 18:10:57: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID null
debug 2022-09-13 18:10:57: Skipping message, definition is undefined and still interviewing
debug 2022-09-13 18:10:57: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
debug 2022-09-13 18:10:57: Skipping message, definition is undefined and still interviewing
debug 2022-09-13 18:10:58: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID null
debug 2022-09-13 18:10:58: Skipping message, definition is undefined and still interviewing
debug 2022-09-13 18:10:58: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID null
debug 2022-09-13 18:10:58: Skipping message, definition is undefined and still interviewing
info  2022-09-13 18:10:58: Successfully interviewed '0x9035eafffecc7967', device has successfully been paired
info  2022-09-13 18:10:58: Device '0x9035eafffecc7967' is supported, identified as: TuYa Zigbee smart dimmer (TS0601_dimmer)
info  2022-09-13 18:10:58: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee smart dimmer","exposes":[{"features":[{"access":3,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":3,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":3,"description":"Minimum light brightness","name":"min_brightness","property":"min_brightness","type":"numeric","value_max":255,"value_min":1},{"access":3,"description":"Maximum light brightness","name":"max_brightness","property":"max_brightness","type":"numeric","value_max":255,"value_min":1}],"type":"light"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601_dimmer","options":[],"supports_ota":false,"vendor":"TuYa"},"friendly_name":"0x9035eafffecc7967","ieee_address":"0x9035eafffecc7967","status":"successful","supported":true},"type":"device_interview"}'
info  2022-09-13 18:10:58: Configuring '0x9035eafffecc7967'
info  2022-09-13 18:10:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee smart dimmer","friendly_name":"0x9035eafffecc7967","model":"TS0601_dimmer","supported":true,"vendor":"TuYa"},"type":"pairing"}'
info  2022-09-13 18:10:58: MQTT publish: topic 'homeassistant/light/0x9035eafffecc7967/light/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/0x9035eafffecc7967/set","device":{"identifiers":["zigbee2mqtt_0x9035eafffecc7967"],"manufacturer":"TuYa","model":"Zigbee smart dimmer (TS0601_dimmer)","name":"0x9035eafffecc7967"},"json_attributes_topic":"zigbee2mqtt/0x9035eafffecc7967","name":"0x9035eafffecc7967","schema":"json","s [... cut ...] moesSwitchPowerOnBehavior","alectoBatteryState"]
warn  2022-09-13 18:11:34: zigbee-herdsman-converters:tuya_dimmer: Received Tuya DataPoint #14 with unexpected datatype from 0x9035eafffecc7967 with raw data '{"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum' (instead of 'undefined'), value='0', known DP# usage: ["runningState","moesSbattery","wooxBattery","haozeeFaultAlarm","nousTempAlarm","moesSwitchPowerOnBehavior","alectoBatteryState"]
```

Koenkk, 2022-09-13:
So it seems this device uses a different data point, to find out you need a TuYa bridge and follow: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

benflux, 2022-09-13:
I'm happy to go buy a bridge if nobody else has one...

shanelord01, 2022-09-13:
I just bought one. Will be here in 2 days.

shanelord01, 2022-09-15:
> So it seems this device uses a different data point, to find out you need a TuYa bridge and follow: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

Here you go!

```
========================================
# Switch1
#

{startRowId: "", pageNo: 1, pageSize: 10, code: "1", startTime: 1663043163748, endTime: 1663215963749,…}
code
: 
"1"
deviceId
: 
"bf086bbf425e7ff5cdppvp"
endTime
: 
1663215963749
pageNo
: 
1
pageSize
: 
10
pageStartRow
: 
""
projectCode
: 
"p1624331325472a3e7qk"
region
: 
"EU"
sourceId
: 
"eu1598151012762yUZE1"
sourceType
: 
"4"
startRowId
: 
""
startTime
: 
1663043163748

========================================
#Brightness1
#

{startRowId: "", pageNo: 1, pageSize: 10, code: "2", startTime: 1663043224334, endTime: 1663216024334,…}
code
: 
"2"
deviceId
: 
"bf086bbf425e7ff5cdppvp"
endTime
: 
1663216024334
pageNo
: 
1
pageSize
: 
10
pageStartRow
: 
""
projectCode
: 
"p1624331325472a3e7qk"
region
: 
"EU"
sourceId
: 
"eu1598151012762yUZE1"
sourceType
: 
"4"
startRowId
: 
""
startTime
: 
1663043224334

========================================
# Minimum brightness1
#

{startRowId: "", pageNo: 1, pageSize: 10, code: "3", startTime: 1663043299929, endTime: 1663216099929,…}
code
: 
"3"
deviceId
: 
"bf086bbf425e7ff5cdppvp"
endTime
: 
1663216099929
pageNo
: 
1
pageSize
: 
10
pageStartRow
: 
""
projectCode
: 
"p1624331325472a3e7qk"
region
: 
"EU"
sourceId
: 
"eu1598151012762yUZE1"
sourceType
: 
"4"
startRowId
: 
""
startTime
: 
1663043299929

========================================
# Type of light source1
#

{startRowId: "", pageNo: 1, pageSize: 10, code: "4", startTime: 1663043325401, endTime: 1663216125402,…}
code
: 
"4"
deviceId
: 
"bf086bbf425e7ff5cdppvp"
endTime
: 
1663216125402
pageNo
: 
1
pageSize
: 
10
pageStartRow
: 
""
projectCode
: 
"p1624331325472a3e7qk"
region
: 
"EU"
sourceId
: 
"eu1598151012762yUZE1"
sourceType
: 
"4"
startRowId
: 
""
startTime
: 
1663043325401

========================================
# Maximum brightness1
#

{startRowId: "", pageNo: 1, pageSize: 10, code: "5", startTime: 1663043346205, endTime: 1663216146206,…}
code
: 
"5"
deviceId
: 
"bf086bbf425e7ff5cdppvp"
endTime
: 
1663216146206
pageNo
: 
1
pageSize
: 
10
pageStartRow
: 
""
projectCode
: 
"p1624331325472a3e7qk"
region
: 
"EU"
sourceId
: 
"eu1598151012762yUZE1"
sourceType
: 
"4"
startRowId
: 
""
startTime
: 
1663043346205

========================================
# Countdown1
#

{startRowId: "", pageNo: 1, pageSize: 10, code: "6", startTime: 1663043364581, endTime: 1663216164581,…}
code
: 
"6"
deviceId
: 
"bf086bbf425e7ff5cdppvp"
endTime
: 
1663216164581
pageNo
: 
1
pageSize
: 
10
pageStartRow
: 
""
projectCode
: 
"p1624331325472a3e7qk"
region
: 
"EU"
sourceId
: 
"eu1598151012762yUZE1"
sourceType
: 
"4"
startRowId
: 
""
startTime
: 
1663043364581

========================================
# Power-on Status Setting
#

{startRowId: "", pageNo: 1, pageSize: 10, code: "14", startTime: 1663043388195, endTime: 1663216188196,…}
code
: 
"14"
deviceId
: 
"bf086bbf425e7ff5cdppvp"
endTime
: 
1663216188196
pageNo
: 
1
pageSize
: 
10
pageStartRow
: 
""
projectCode
: 
"p1624331325472a3e7qk"
region
: 
"EU"
sourceId
: 
"eu1598151012762yUZE1"
sourceType
: 
"4"
startRowId
: 
""
startTime
: 
1663043388195

========================================

```

shanelord01, 2022-09-15:
Here is some extra info that may or may not help:

[image]
[image]

Koenkk, 2022-09-15:
Thanks, can you try with this external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const tzLocal = {
    tuya_dimmer_level: {
        key: ['brightness_min', 'min_brightness', 'max_brightness', 'brightness', 'brightness_percent', 'level'],
        convertSet: async (entity, key, value, meta) => {
            // upscale to 1000
            let newValue;
            let dp = tuya.dataPoints.dimmerLevel;
            if (['_TZE200_3p5ydos3', '_TZE200_9i9dt8is', '_TZE200_dfxkcots', '_TZE200_w4cryh2i', '_TZE200_1agwnems'].includes(meta.device.manufacturerName)) {
                dp = tuya.dataPoints.eardaDimmerLevel;
            }
            if (key === 'brightness_min') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                    dp = tuya.dataPoints.dimmerLevel;
                } else {
                    throw new Error('Dimmer brightness_min is out of range 0..100');
                }
            } else if (key === 'min_brightness') {
                if (value >= 1 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 1, 255, 0, 1000);
                    dp = tuya.dataPoints.dimmerMinLevel;
                } else {
                    throw new Error('Dimmer min_brightness is out of range 1..255');
                }
            } else if (key === 'max_brightness') {
                if (value >= 1 && value <= 255) {
                    newValue = utils.mapNumberRange(value, 1, 255, 0, 1000);
                    dp = tuya.dataPoints.dimmerMaxLevel;
                } else {
                    throw new Error('Dimmer min_brightness is out of range 1..255');
                }
            } else if (key === 'level') {
                if (value >= 0 && value <= 1000) {
                    newValue = Math.round(Number(value));
                } else {
                    throw new Error('Dimmer level is out of range 0..1000');
                }
            } else if (key === 'brightness_percent') {
                if (value >= 0 && value <= 100) {
                    newValue = utils.mapNumberRange(value, 0, 100, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness_percent is out of range 0..100');
                }
            } else { // brightness
                if (value >= 0 && value <= 254) {
                    newValue = utils.mapNumberRange(value, 0, 254, 0, 1000);
                } else {
                    throw new Error('Dimmer brightness is out of range 0..254');
                }
            }
            // Always use same transid as tuya_dimmer_state (https://github.com/Koenkk/zigbee2mqtt/issues/6366)
            await tuya.sendDataPointValue(entity, dp, newValue, 'dataRequest', 1);
        },
    }
}

[... 49 lines cut ...]
    model: 'TS0601_dimmer',
    vendor: 'TuYa',
    description: 'Zigbee smart dimmer',
    fromZigbee: [fzLocal.tuya_dimmer, fz.ignore_basic_report],
    toZigbee: [tz.tuya_dimmer_state, tzLocal.tuya_dimmer_level],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
    exposes: [e.light_brightness().withMinBrightness().withMaxBrightness().setAccess(
        'state', ea.STATE_SET).setAccess('brightness', ea.STATE_SET).setAccess(
        'min_brightness', ea.STATE_SET).setAccess('max_brightness', ea.STATE_SET)],
    whiteLabel: [
        {vendor: 'Larkkey', model: 'ZSTY-SM-1DMZG-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAA-EU'},
        {vendor: 'Earda', model: 'EDM-1ZAB-EU'},
        {vendor: 'Earda', model: 'EDM-1ZBA-EU'},
        {vendor: 'Mercator Ikuü', model: 'SSWD01'},
        {vendor: 'Moes', model: 'ZS-USD'},
    ],
};

module.exports = definition;

```

shanelord01, 2022-09-15:
Seems like more green popups than red, but still errors and no dimming.

```
info  2022-09-15 19:17:09: Zigbee: allowing new devices to join.
info  2022-09-15 19:17:09: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"m6rbn-1"}'
debug 2022-09-15 19:17:10: Received Zigbee message from 'Study Motion Sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":33}' from endpoint 1 with groupID null
info  2022-09-15 19:17:10: MQTT publish: topic 'zigbee2mqtt/Study Motion Sensor', payload '{"battery":80,"battery_low":false,"linkquality":184,"occupancy":true,"tamper":false,"temperature":23.37,"voltage":2900}'
debug 2022-09-15 19:17:12: Received Zigbee message from 'Front Lounge Blinds 1', type 'raw', cluster '64528', data '{"data":[13,99,18,9,128,187,0,0,0,0,2,0,1,0,0,0,61,0,0,180,65,0,255,82,1,61,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:12: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,9,128,187,0,0,0,0,2,0,1,0,0,0,61,0,0,180,65,0,255,82,1,61,0],"type":"Buffer"}'
debug 2022-09-15 19:17:16: Received Zigbee message from 'Back Lounge Blinds 4', type 'raw', cluster '64528', data '{"data":[13,99,18,90,128,80,0,0,0,0,2,0,0,0,157,55,65,0,0,182,65,0,255,103,1,61,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:16: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,90,128,80,0,0,0,0,2,0,0,0,157,55,65,0,0,182,65,0,255,103,1,61,0],"type":"Buffer"}'
debug 2022-09-15 19:17:17: Received Zigbee message from 'Sunroom Downlight', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":269,"manufacturerCode":4478}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:17: Device 'Sunroom Downlight' requested OTA
debug 2022-09-15 19:17:24: Received Zigbee message from 'Back Lounge Blinds 2', type 'raw', cluster '64528', data '{"data":[13,99,18,80,128,78,202,226,128,254,2,0,0,0,226,160,64,0,0,183,65,0,255,135,1,62,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:24: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,80,128,78,202,226,128,254,2,0,0,0,226,160,64,0,0,183,65,0,255,135,1,62,0],"type":"Buffer"}'
debug 2022-09-15 19:17:30: Received Zigbee message from 'Back Lounge Blinds 1', type 'raw', cluster '64528', data '{"data":[13,99,18,115,128,81,0,0,0,0,2,0,0,0,244,29,64,0,0,182,65,0,255,169,1,61,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:30: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,115,128,81,0,0,0,0,2,0,0,0,244,29,64,0,0,182,65,0,255,169,1,61,0],"type":"Buffer"}'
info  2022-09-15 19:17:30: Device '0x9035eafffecc7967' joined
info  2022-09-15 19:17:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffecc7967","ieee_address":"0x9035eafffecc7967"},"type":"device_joined"}'
info  2022-09-15 19:17:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x9035eafffecc7967"},"type":"device_connected"}'
info  2022-09-15 19:17:31: Starting interview of '0x9035eafffecc7967'
info  2022-09-15 19:17:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffecc7967","ieee_address":"0x9035eafffecc7967","status":"started"},"type":"device_interview"}'
info  2022-09-15 19:17:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x9035eafffecc7967"},"type":"pairing"}'
debug 2022-09-15 19:17:31: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_1agwnems","modelId":"TS0601"}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:31: Skipping message, definition is undefined and still interviewing
debug 2022-09-15 19:17:31: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:31: Skipping message, definition is undefined and still interviewing
debug 2022-09-15 19:17:32: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:32: Skipping message, definition is undefined and still interviewing
debug 2022-09-15 19:17:32: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"appVersion":68}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:32: Skipping message, definition is undefined and still interviewing
debug 2022-09-15 19:17:32: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:32: Skipping message, definition is undefined and still interviewing
debug 2022-09-15 19:17:32: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID null
debug 2022-09-15 19:17:32: Skipping message, definition is undefined and still interviewing
[... cut ...]
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:527:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:503:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:714:18)
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at DeconzAdapter.emit (node:events:527:28)
```

Koenkk, 2022-09-16:
Updated https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1247738975

shanelord01, 2022-09-16:
Lot's less red this time...

```
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,166],"type":"Buffer"},"datatype":2,"dp":2}],"seq":20173}' from endpoint 1 with groupID null
info  2022-09-16 20:13:51: MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":42,"linkquality":255,"state":"OFF"}'
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,166],"type":"Buffer"},"datatype":2,"dp":2}],"seq":20173}' from endpoint 1 with groupID null
info  2022-09-16 20:13:51: MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":42,"linkquality":255,"state":"OFF"}'
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,166],"type":"Buffer"},"datatype":2,"dp":2}],"seq":20173}' from endpoint 1 with groupID null
info  2022-09-16 20:13:51: MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":42,"linkquality":255,"state":"OFF"}'
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":20429}' from endpoint 1 with groupID null
info  2022-09-16 20:13:51: MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":42,"linkquality":255,"min_brightness":1,"state":"OFF"}'
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":20429}' from endpoint 1 with groupID null
info  2022-09-16 20:13:51: MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":42,"linkquality":255,"min_brightness":1,"state":"OFF"}'
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":20685}' from endpoint 1 with groupID null
warn  2022-09-16 20:13:51: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #4 from 0x9035eafffecc7967 with raw data '{"dp":4,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum', value='0', known DP# usage: ["mode","moesSboostHeating","silvercrestSetColorTemp","haozeeBoostHeating","nousBattery","tthBattery","wlsBatteryPercentage","evanellHeatingSetpoint","AM02Mode","tshpsMaximumRange","lmsBattery"]
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":20941}' from endpoint 1 with groupID null
info  2022-09-16 20:13:51: MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":42,"linkquality":255,"max_brightness":7,"min_brightness":1,"state":"OFF"}'
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":20941}' from endpoint 1 with groupID null
info  2022-09-16 20:13:51: MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":42,"linkquality":255,"max_brightness":7,"min_brightness":1,"state":"OFF"}'
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}],"seq":21197}' from endpoint 1 with groupID null
warn  2022-09-16 20:13:51: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #6 from 0x9035eafffecc7967 with raw data '{"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}': type='commandDataResponse', datatype='value', value='0', known DP# usage: ["silvercrestSetEffect","haozeeRunningState","evanellBattery","tshpsSelfTest"]
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}],"seq":21197}' from endpoint 1 with groupID null
warn  2022-09-16 20:13:51: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #6 from 0x9035eafffecc7967 with raw data '{"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}': type='commandDataResponse', datatype='value', value='0', known DP# usage: ["silvercrestSetEffect","haozeeRunningState","evanellBattery","tshpsSelfTest"]
debug 2022-09-16 20:13:51: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":21453}' from endpoint 1 with groupID null
warn  2022-09-16 20:13:51: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #14 from 0x9035eafffecc7967 with raw data '{"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='e [... cut ...]
debug 2022-09-16 20:14:32: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":29133}' from endpoint 1 with groupID null
warn  2022-09-16 20:14:32: zigbee-herdsman-converters:tuya_dimmer: Received unexpected Tuya DataPoint #14 from 0x9035eafffecc7967 with raw data '{"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}': type='commandDataResponse', datatype='enum', value='0', known DP# usage: ["runningState","moesSbattery","wooxBattery","haozeeFaultAlarm","nousTempAlarm","moesSwitchPowerOnBehavior","alectoBatteryState"]
```

shanelord01, 2022-09-17:
Removed from Z2M and re-joined it. Then tried to change brightness and max min brightness settings:

```
info  2022-09-17 12:46:59: Zigbee: allowing new devices to join.
info  2022-09-17 12:46:59: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"ky7az-1"}'
debug 2022-09-17 12:47:00: Received Zigbee message from 'Study Motion Sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":33}' from endpoint 1 with groupID null
info  2022-09-17 12:47:00: MQTT publish: topic 'zigbee2mqtt/Study Motion Sensor', payload '{"battery":100,"battery_low":false,"linkquality":255,"occupancy":true,"tamper":false,"temperature":23.46,"voltage":2900}'
debug 2022-09-17 12:47:11: Received Zigbee message from 'Front Lounge Blinds 2', type 'raw', cluster '64528', data '{"data":[13,99,18,73,128,187,0,0,0,0,2,0,0,32,72,173,66,0,0,185,65,1,255,96,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-17 12:47:11: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,73,128,187,0,0,0,0,2,0,0,32,72,173,66,0,0,185,65,1,255,96,0,0,0],"type":"Buffer"}'
debug 2022-09-17 12:47:13: Received Zigbee message from 'Back Lounge Blinds 1', type 'raw', cluster '64528', data '{"data":[13,99,18,105,128,79,0,0,0,0,2,0,0,128,12,175,65,0,0,176,65,1,255,150,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-17 12:47:13: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,105,128,79,0,0,0,0,2,0,0,128,12,175,65,0,0,176,65,1,255,150,0,0,0],"type":"Buffer"}'
debug 2022-09-17 12:47:14: Received Zigbee message from 'Pool Filter Smart Plug', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,24572]}' from endpoint 1 with groupID null
info  2022-09-17 12:47:14: MQTT publish: topic 'zigbee2mqtt/Pool Filter Smart Plug', payload '{"energy":245.72,"linkquality":255,"power":825,"state":"ON"}'
debug 2022-09-17 12:47:17: Received Zigbee message from 'Front Lounge Blinds 1', type 'raw', cluster '64528', data '{"data":[13,99,18,5,128,186,0,0,0,0,2,0,1,8,55,191,67,0,0,185,65,1,255,70,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-17 12:47:17: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,5,128,186,0,0,0,0,2,0,1,8,55,191,67,0,0,185,65,1,255,70,0,0,0],"type":"Buffer"}'
debug 2022-09-17 12:47:19: Received Zigbee message from 'Back Lounge Blinds 2', type 'raw', cluster '64528', data '{"data":[13,99,18,69,128,76,202,226,128,254,2,0,0,240,118,17,67,0,0,179,65,1,255,112,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-17 12:47:19: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,69,128,76,202,226,128,254,2,0,0,240,118,17,67,0,0,179,65,1,255,112,0,0,0],"type":"Buffer"}'
debug 2022-09-17 12:47:21: Received Zigbee message from 'Study Motion Sensor', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID null
info  2022-09-17 12:47:21: MQTT publish: topic 'zigbee2mqtt/Study Motion Sensor', payload '{"battery":100,"battery_low":false,"linkquality":255,"occupancy":false,"tamper":false,"temperature":23.46,"voltage":2900}'
info  2022-09-17 12:47:21: Device '0x9035eafffecc7967' joined
info  2022-09-17 12:47:22: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffecc7967","ieee_address":"0x9035eafffecc7967"},"type":"device_joined"}'
info  2022-09-17 12:47:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x9035eafffecc7967"},"type":"device_connected"}'
info  2022-09-17 12:47:22: Starting interview of '0x9035eafffecc7967'
info  2022-09-17 12:47:22: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x9035eafffecc7967","ieee_address":"0x9035eafffecc7967","status":"started"},"type":"device_interview"}'
info  2022-09-17 12:47:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x9035eafffecc7967"},"type":"pairing"}'
debug 2022-09-17 12:47:22: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_1agwnems","modelId":"TS0601"}' from endpoint 1 with groupID null
debug 2022-09-17 12:47:22: Skipping message, definition is undefined and still interviewing
debug 2022-09-17 12:47:23: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID null
debug 2022-09-17 12:47:23: Skipping message, definition is undefined and still interviewing
debug 2022-09-17 12:47:23: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID null
debug 2022-09-17 12:47:23: Skipping message, definition is undefined and still interviewing
debug 2022-09-17 12:47:23: Received Zigbee message from '0x9035eafffecc7967', type 'readResponse', cluster 'genBasic', data '{"appVersion":68}' from endpoint 1 with groupID null
debug 2022-09-17 12:47:23: Skipping message, definition is undefined and still interviewing
[... cut ...]
info  2022-09-17 12:49:15: MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":254,"linkquality":255,"max_brightness":7,"min_brightness":1,"state":"OFF"}'
debug 2022-09-17 12:49:15: Received Zigbee message from '0x9035eafffecc7967', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1793}' from endpoint 1 with groupID null
info  2022-09-17 12:49:15: MQTT publish: topic 'zigbee2mqtt/0x9035eafffecc7967', payload '{"brightness":254,"linkquality":255,"max_brightness":7,"min_brightness":1,"state":"OFF"}'

```

Koenkk, 2022-09-19:
Does de dimming work now?

shanelord01, 2022-09-19:
> Does the dimming work now?

No.

shanelord01, 2022-09-20:
> Does the dimming work now?

Update: Yes it does.

Within Z2M: Brightness works fine. Setting min-brightness and max-brightness don't work.
Within HA entity created by Z2M: Jumps to full brightness then moves to brightness requested every time you make a brightness adjustment.

Koenkk, 2022-09-22:
@shanelord01 can you check if with [this](https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96) converter the brightness and state works correctly?

shanelord01, 2022-09-23:
> @shanelord01 can you check if with [this](https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96) converter the brightness and state works correctly?

I get this error when I try and change state in Z2M (off to on):

```
debug 2022-09-23 10:36:25: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Back Lounge Dimmer Light","transaction":"tgzsy-1"}'
info  2022-09-23 10:36:25: Configuring 'Back Lounge Dimmer Light'
info  2022-09-23 10:36:25: Successfully configured 'Back Lounge Dimmer Light'
info  2022-09-23 10:36:25: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Back Lounge Dimmer Light"},"status":"ok","transaction":"tgzsy-1"}'
debug 2022-09-23 10:36:25: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4462}' from endpoint 1 with groupID null
debug 2022-09-23 10:36:25: No converter available for 'TS0601_dimmer' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4462}'
debug 2022-09-23 10:36:25: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4462}' from endpoint 1 with groupID null
debug 2022-09-23 10:36:25: No converter available for 'TS0601_dimmer' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4462}'
debug 2022-09-23 10:36:26: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4462}' from endpoint 1 with groupID null
debug 2022-09-23 10:36:26: No converter available for 'TS0601_dimmer' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4462}'
debug 2022-09-23 10:36:26: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":4718}' from endpoint 1 with groupID null
debug 2022-09-23 10:36:26: No converter available for 'TS0601_dimmer' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":4718}'
debug 2022-09-23 10:36:26: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":4974}' from endpoint 1 with groupID null
debug 2022-09-23 10:36:26: No converter available for 'TS0601_dimmer' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":4974}'
debug 2022-09-23 10:36:26: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":5230}' from endpoint 1 with groupID null
debug 2022-09-23 10:36:26: No converter available for 'TS0601_dimmer' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":5230}'
debug 2022-09-23 10:36:26: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":5486}' from endpoint 1 with groupID null
debug 2022-09-23 10:36:26: No converter available for 'TS0601_dimmer' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":5486}'
debug 2022-09-23 10:36:26: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":5486}' from endpoint 1 with groupID null
debug 2022-09-23 10:36:26: No converter available for 'TS0601_dimmer' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":5486}'
debug 2022-09-23 10:36:26: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}],"seq":5742}' from endpoint 1 with groupID null
debug 2022-09-23 10:36:26: No converter available for 'TS0601_dimmer' with cluster 'manuSpecificTuya' and type 'commandDataResponse' and data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}],"seq":5742}'
debug 2022-09-23 10:36:26: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":5998}' from endpoint 1 with groupID null [... cut ...]
    at /app/lib/extension/publish.ts:209:60
    at Array.find (<anonymous>)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:209:42)
    at EventEmitter.emit (node:events:539:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:140:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:94:27)
    at WebSocket.emit (node:events:527:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1178:20)
    at Receiver.emit (node:events:527:28)
```

Koenkk, 2022-09-23:
Made a mistake, updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96 please try agian.

shanelord01, 2022-09-24:
> Made a mistake, updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96 please try agian.

Now getting these errors:

```
debug 2022-09-24 11:18:10: Received Zigbee message from 'Back Lounge Dimmer Light', type 'attributeReport', cluster 'genBasic', data '{"65506":31,"65508":0,"appVersion":68}' from endpoint 1 with groupID null
debug 2022-09-24 11:18:12: Received MQTT message on 'zigbee2mqtt/bridge/request/device/configure' with data '{"id":"Dining Room Dimmer Light","transaction":"7a2dn-1"}'
info  2022-09-24 11:18:12: Configuring 'Dining Room Dimmer Light'
info  2022-09-24 11:18:12: Successfully configured 'Dining Room Dimmer Light'
info  2022-09-24 11:18:13: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Dining Room Dimmer Light"},"status":"ok","transaction":"7a2dn-1"}'
debug 2022-09-24 11:18:15: Received Zigbee message from 'Front Lounge Blinds 2', type 'raw', cluster '64528', data '{"data":[13,99,18,68,128,183,0,0,0,0,2,0,0,128,41,224,65,0,0,196,65,1,255,177,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-24 11:18:15: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,68,128,183,0,0,0,0,2,0,0,128,41,224,65,0,0,196,65,1,255,177,0,0,0],"type":"Buffer"}'
debug 2022-09-24 11:18:17: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":29696}' from endpoint 1 with groupID null
info  2022-09-24 11:18:17: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"linkquality":255,"max_brightness":7,"min_brightness":1,"state":"OFF"}'
debug 2022-09-24 11:18:17: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":29952}' from endpoint 1 with groupID null
info  2022-09-24 11:18:17: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"linkquality":255,"max_brightness":7,"min_brightness":1,"state":"OFF"}'
debug 2022-09-24 11:18:17: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":30208}' from endpoint 1 with groupID null
info  2022-09-24 11:18:17: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"linkquality":255,"max_brightness":7,"min_brightness":1,"state":"OFF"}'
debug 2022-09-24 11:18:17: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":30464}' from endpoint 1 with groupID null
warn  2022-09-24 11:18:17: zigbee-herdsman-converters:fzDataPoints: dp 4 not defined for '_TZE200_1agwnems' with data {"dp":4,"datatype":4,"data":{"type":"Buffer","data":[0]}}
debug 2022-09-24 11:18:17: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":30720}' from endpoint 1 with groupID null
info  2022-09-24 11:18:17: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"linkquality":160,"max_brightness":7,"min_brightness":1,"state":"OFF"}'
debug 2022-09-24 11:18:17: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}],"seq":30976}' from endpoint 1 with groupID null
warn  2022-09-24 11:18:17: zigbee-herdsman-converters:fzDataPoints: dp 6 not defined for '_TZE200_1agwnems' with data {"dp":6,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
debug 2022-09-24 11:18:17: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":31232}' from endpoint 1 with groupID null
warn  2022-09-24 11:18:17: zigbee-herdsman-converters:fzDataPoints: dp 14 not defined for '_TZE200_1agwnems' with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}
debug 2022-09-24 11:18:18: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":29696}' from endpoint 1 with groupID null
info  2022-09-24 11:18:18: MQTT publish: topic 'zigbee2mqtt/Dining Room Dimmer Light', payload '{"brightness":254,"linkquality":255,"max_brightness":7,"min_brightness":1,"state":"OFF"}'
debug 2022-09-24 11:18:18: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":29952}' from endpoint 1 with groupID null
info  2022-09-24 11:18:18: MQTT publish: topic 'zigbee2mqtt/Dining Room Dimmer Light', payload '{"brightness":254,"linkquality":255,"max_brightness":7,"min_brightness":1,"state":"OFF"}'
[... cut ...]
warn  2022-09-24 11:18:28: zigbee-herdsman-converters:fzDataPoints: dp 14 not defined for '_TZE200_1agwnems' with data {"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}
debug 2022-09-24 11:18:28: Received Zigbee message from 'Back Lounge Blinds 3', type 'raw', cluster '64528', data '{"data":[13,99,18,48,128,73,0,0,0,0,2,0,0,0,151,214,66,0,0,182,65,1,255,165,0,0,0],"type":"Buffer"}' from endpoint 1 with groupID null
debug 2022-09-24 11:18:28: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,48,128,73,0,0,0,0,2,0,0,0,151,214,66,0,0,182,65,1,255,165,0,0,0],"type":"Buffer"}'
```

Koenkk, 2022-09-25:
- Does the brightness and state update correctly now?
- I updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96, all features should be supported now (added `light_type`, `countdown` and `power_on_behaviour`

shanelord01, 2022-09-25:
> * Does the brightness and state update correctly now?
> * I updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96, all features should be supported now (added `light_type`, `countdown` and `power_on_behaviour`

I don'ty believe so buit can't test as I am now getting this on startup of Z2M:

```
[18:40:37] INFO: Handing over control to Zigbee2MQTT Core ...
/app/dist/util/externally-loaded.js:68
    convertSet: async (entity, _, _, meta) => {
                                  ^
SyntaxError: Duplicate parameter name not allowed in this context
    at new Script (node:vm:100:7)
    at createScript (node:vm:257:10)
    at Object.runInNewContext (node:vm:298:10)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
    at start (/app/index.js:108:18)
```

shanelord01, 2022-09-28:
> * Does the brightness and state update correctly now?
> * I updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96, all features should be supported now (added `light_type`, `countdown` and `power_on_behaviour`

Any chance the error upon load as per above can be resolved? It doesn't happen with previous external converters only this latest with the _, _, section.

Koenkk, 2022-09-28:
Updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96

shanelord01, 2022-09-29:
> Updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96

Not sure what's happened:

```
debug 2022-09-29 10:58:26: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":65280}' from endpoint 1 with groupID null
error 2022-09-29 10:58:26: Exception while calling fromZigbee converter: No datapoints map defined}
debug 2022-09-29 10:58:26: Error: No datapoints map defined
    at Object.convert (/app/dist/util/externally-loaded.js:93:57)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:539:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:527:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at DeconzAdapter.emit (node:events:527:28)
debug 2022-09-29 10:58:26: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":65280}' from endpoint 1 with groupID null
error 2022-09-29 10:58:26: Exception while calling fromZigbee converter: No datapoints map defined}
debug 2022-09-29 10:58:26: Error: No datapoints map defined
    at Object.convert (/app/dist/util/externally-loaded.js:93:57)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:539:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:527:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at DeconzAdapter.emit (node:events:527:28)
debug 2022-09-29 10:58:26: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":513}' from endpoint 1 with groupID null
error 2022-09-29 10:58:26: Exception while calling fromZigbee converter: No datapoints map defined}
debug 2022-09-29 10:58:26: Error: No datapoints map defined
    at Object.convert (/app/dist/util/externally-loaded.js:93:57)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:539:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:527:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at DeconzAdapter.emit (node:events:527:28)
debug 2022-09-29 10:58:27: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1025}' from endpoint 1 with groupID null
error 2022-09-29 10:58:27: Exception while calling fromZigbee converter: No datapoints map defined}
debug 2022-09-29 10:58:27: Error: No datapoints map defined
    at Object.convert (/app/dist/util/externally-loaded.js:93:57)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:539:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:527:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at DeconzAdapter.emit (node:events:527:28)
debug 2022-09-29 10:58:27: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1281}' from endpoint 1 with groupID null
error 2022-09-29 10:58:27: Exception while calling fromZigbee converter: No datapoints map defined}
debug 2022-09-29 10:58:27: Error: No datapoints map defined
    at Object.convert (/app/dist/util/externally-loaded.js:93:57)
    at Receive.onDeviceMessage (/app/lib/extension/receive.ts:143:51)
    at EventEmitter.emit (node:events:539:35)
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:527:28)
[... cut ...]
    at EventBus.emitDeviceMessage (/app/lib/eventBus.ts:102:22)
    at Controller.<anonymous> (/app/lib/zigbee.ts:106:27)
    at Controller.emit (node:events:527:28)
    at Controller.selfAndDeviceEmit (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:515:14)
    at Controller.onZclOrRawData (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:726:18)
    at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:144:70)
    at DeconzAdapter.emit (node:events:527:28)

```

Koenkk, 2022-09-29:
Should be fixed now.

shanelord01, 2022-09-29:
> Should be fixed now.

Thanks - no errors now, but ON/OFF doesn't work anymore... brightness works.

```
info  2022-09-30 09:32:34: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
debug 2022-09-30 09:32:34: Received MQTT message on 'zigbee2mqtt/Back Lounge Dimmer Light/set' with data '{"state":"OFF"}'
debug 2022-09-30 09:32:34: Publishing 'set' 'state' to 'Back Lounge Dimmer Light'
info  2022-09-30 09:32:34: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"OFF"}'
info  2022-09-30 09:32:34: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
debug 2022-09-30 09:32:35: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":22016}' from endpoint 1 with groupID null
info  2022-09-30 09:32:35: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"OFF"}'
info  2022-09-30 09:32:35: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
debug 2022-09-30 09:32:35: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":22272}' from endpoint 1 with groupID null
info  2022-09-30 09:32:35: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"OFF"}'
info  2022-09-30 09:32:35: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
debug 2022-09-30 09:32:35: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}],"seq":22528}' from endpoint 1 with groupID null
info  2022-09-30 09:32:35: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"OFF"}'
info  2022-09-30 09:32:35: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
debug 2022-09-30 09:32:35: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":22784}' from endpoint 1 with groupID null
info  2022-09-30 09:32:35: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"OFF"}'
info  2022-09-30 09:32:35: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
debug 2022-09-30 09:32:37: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":29141}' from endpoint 1 with groupID null
info  2022-09-30 09:32:37: MQTT publish: topic 'zigbee2mqtt/Dining Room Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"OFF"}'
info  2022-09-30 09:32:37: MQTT publish: topic 'zigbee2mqtt/Dining Room Dimmer Light/type', payload '{"light":"led"}'
debug 2022-09-30 09:32:37: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,232],"type":"Buffer"},"datatype":2,"dp":2}],"seq":29397}' from endpoint 1 with groupID null
info  2022-09-30 09:32:37: MQTT publish: topic 'zigbee2mqtt/Dining Room Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"OFF"}'
info  2022-09-30 09:32:37: MQTT publish: topic 'zigbee2mqtt/Dining Room Dimmer Light/type', payload '{"light":"led"}'
debug 2022-09-30 09:32:37: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":29653}' from endpoint 1 with groupID null
info  2022-09-30 09:32:37: MQTT publish: topic 'zigbee2mqtt/Dining Room Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"OFF"}'
info  2022-09-30 09:32:37: MQTT publish: topic 'zigbee2mqtt/Dining Room Dimmer Light/type', payload '{"light":"led"}'
debug 2022-09-30 09:32:38: Received Zigbee message from 'Dining Room Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":29909}' from endpoint 1 with groupID null
[... cut ...]
debug 2022-09-30 09:32:44: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":24320}' from endpoint 1 with groupID null
info  2022-09-30 09:32:44: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":248,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
info  2022-09-30 09:32:44: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
```

Koenkk, 2022-09-30:
Updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96 again, please also check if the brightness still works.

shanelord01, 2022-09-30:
> Updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96 again, please also check if the brightness still works.

Yes On/Off and brightness now work. Now, however, brightness change causes the lights to go 100%, then dim (or brighten) to the level set.

benflux, 2022-09-30:
Isn’t this is what the devices do by default? I’ve heard this default behaviour can be changed.

shanelord01, 2022-09-30:
> Isn’t this is what the devices do by default? I’ve heard this default behaviour can be changed.
> […](#)

Not in the Tuya app that I used to get the codes, and not in previous ext_converter.js versions.

Koenkk, 2022-09-30:
@shanelord01 can you provide the herdsman debug log of just setting a brightness value of `100` with both the [old](https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1243346097) and [new](https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96) converter?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

shanelord01, 2022-09-30:
> @shanelord01 can you provide the herdsman debug log of just setting a brightness value of `100` with both the [old](https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1243346097) and [...]

I'm on Home Assistant OS and don't have the skills to access logs inside a container (I think I'd have to do this... and concerned I'll screw something up https://developers.home-assistant.io/docs/operating-system/debugging/#ssh-access-to-the-host)

Koenkk, 2022-09-30:
@shanelord01 thats not needed, it will show up in the log tab of the addon. Make sure you enable herdsman debug logging switch from the addon configuration page.

shanelord01, 2022-09-30:
> @shanelord01 thats not needed, it will show up in the log tab of the addon. Make sure you enable herdsman debug logging switch from the addon configuration page.

OK. One thing I just noticed. The jump to 100% only happens when doing it from Home Assistant Entity via Lovelace front end. This doesn't happen when changing brightness in Z2M "Exposes" tab of the device.

shanelord01, 2022-09-30:
Not sure if this helps:

```
Zigbee2MQTT:info  2022-09-30 20:25:08: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Zigbee2MQTT:info  2022-09-30 20:25:08: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
2022-09-30T10:25:08.013Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xbd46 EP:1 SeqNr. 74 request id: 93
2022-09-30T10:25:08.024Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 93 status: 0
2022-09-30T10:25:08.024Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-09-30T10:25:08.024Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2022-09-30T10:25:08.024Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-09-30T10:25:08.027Z zigbee-herdsman:deconz:driver query aps data confirm
2022-09-30T10:25:08.032Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 75
2022-09-30T10:25:08.034Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xbd46 request id: 93 confirm status: 0
2022-09-30T10:25:08.034Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-09-30T10:25:08.034Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 96
2022-09-30T10:25:08.034Z zigbee-herdsman:deconz:adapter false, true, false
2022-09-30T10:25:08.035Z zigbee-herdsman:deconz:adapter resolve request
2022-09-30T10:25:08.046Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-09-30T10:25:08.046Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-09-30T10:25:08.048Z zigbee-herdsman:deconz:driver query aps data indication
2022-09-30T10:25:08.053Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 76
2022-09-30T10:25:08.055Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 76 srcAddr: 0xbd46 destAddr: 0x0 profile id: 0x104 cluster id: 0xef00 lqi: 252
2022-09-30T10:25:08.055Z zigbee-herdsman:deconz:frameParser response payload: 9,97,1,1,6,14,4,0,1,0
2022-09-30T10:25:08.058Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":97,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"seq":1537,"dpValues":[{"dp":14,"datatype":4,"data":{"type":"Buffer","data":[0]}}]},"Command":{"ID":1,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataResponse"}},"address":48454,"endpoint":1,"linkquality":252,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-09-30 20:25:08: Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":1537}' from endpoint 1 with groupID null
2022-09-30T10:25:08.064Z zigbee-herdsman:controller:endpoint DefaultResponse 0x9035eafffecc7967/1 61184(1, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-09-30T10:25:08.065Z zigbee-herdsman:deconz:adapter no response expected
2022-09-30T10:25:08.065Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2022-09-30 20:25:08: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":252,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Zigbee2MQTT:info  2022-09-30 20:25:08: MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
2022-09-30T10:25:08.070Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xbd46 EP:1 SeqNr. 77 request id: 94
2022-09-30T10:25:08.075Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 94 status: 0
2022-09-30T10:25:08.076Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-09-30T10:25:08.090Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2022-09-30T10:25:08.090Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-09-30T10:25:08.096Z zigbee-herdsman:deconz:driver query aps data confirm
2022-09-30T10:25:08.101Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 78
2022-09-30T10:25:08.107Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xbd46 request id: 94 confirm status: 0
2022-09-30T10:25:08.108Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
[... cut ...] transSeq Nr.: 97
2022-09-30T10:25:08.110Z zigbee-herdsman:deconz:adapter false, true, false
2022-09-30T10:25:08.110Z zigbee-herdsman:deconz:adapter resolve request
2022-09-30T10:25:09.201Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 79
2022-09-30T10:25:09.204Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-09-30T10:25:09.206Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```

shanelord01, 2022-09-30:
I also don't have a copy of an older ext_converter.js sorry!

Koenkk, 2022-09-30:
If it only occurs in HA then the debug log is enough, so please disable the herdsman debug logging since it pollutes the log. I don't see you changing the brightness in this log, probably because to much herdsman debug log was written.

The old converter can be found here: https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1243346097

shanelord01, 2022-09-30:
Here is zh debug disabled:

```
Debug 2022-09-30 22:01:03Received MQTT message on 'zigbee2mqtt/Back Lounge Dimmer Light/set' with data '{"state":"ON","brightness":97}'
Debug 2022-09-30 22:01:03Publishing 'set' 'brightness' to 'Back Lounge Dimmer Light'
Info 2022-09-30 22:01:03MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":97,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:01:03MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:01:04Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1}' from endpoint 1 with groupID null
Info 2022-09-30 22:01:04MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":97,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:01:04MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:01:04Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,126],"type":"Buffer"},"datatype":2,"dp":2}],"seq":1}' from endpoint 1 with groupID null
Info 2022-09-30 22:01:04MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":97,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:01:04MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:01:05Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":1}' from endpoint 1 with groupID null
Info 2022-09-30 22:01:05MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":97,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:01:05MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:01:05Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,119],"type":"Buffer"},"datatype":2,"dp":2}],"seq":257}' from endpoint 1 with groupID null
Info 2022-09-30 22:01:05MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":95,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:01:05MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:01:05Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":513}' from endpoint 1 with groupID null
Info 2022-09-30 22:01:05MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":95,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:01:05MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:01:06Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":769}' from endpoint 1 with groupID null
Info 2022-09-30 22:01:06MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":95,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:01:06MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:01:06Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":1025}' from endpoint 1 with groupID null
Info 2022-09-30 22:01:06MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":95,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:01:06MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:01:06Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}],"seq":1281}' from endpoint 1 with groupID null
Info 2022-09-30 22:01:06MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":95,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
[... cut ...]
Debug 2022-09-30 22:01:10Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":1537}' from endpoint 1 with groupID null
Info 2022-09-30 22:01:10MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":254,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:01:10MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
```

Koenkk, 2022-09-30:
I expect it is related to the message payload order. Can you try the MQTT publishes as I suggested in https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1263439931

shanelord01, 2022-09-30:
> If it only occurs in HA then the debug log is enough, so please disable the herdsman debug logging since it pollutes the log. I don't see you changing the brightness in this log, probably because to much herdsman debug log was written.
> 
> The old converter can be found here: [#13800 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1243346097)
> 
> Also can you try the following. Use an MQTT client (or the HA mqtt publish service to publish a message to topic: `zigbee2mqtt/Back Lounge Dimmer Light` with payloads), vary the `brightness` values and check if the 100% jump occurs with the following payloads.
> 
> * `{"brightness": 200}`
> * `{"state": "ON", "brightness": 200}`
> * `{"brightness": 200, "state": "ON"}`

Weird. Using MQTT Explorer, I issue the payload {"state": "OFF"} when the light is on, and the switch in HA momentarily goes off (but the light physically doesn't) then turns on again.

Seems I'm doing it wrong?

Koenkk, 2022-09-30:
Can you provide the debug log of that?

shanelord01, 2022-09-30:
Top from Mosquitto logs, bottom from Z2M logs.

```
2022-09-30 22:55:26: Sending PUBLISH to 06gRVjwYvulrbWmUa6Umwo (d0, q0, r0, m0, 'zigbee2mqtt/Dining Room Dimmer Light', ... (147 bytes))
2022-09-30 22:55:26: Sending PUBLISH to mqttjs_3516dda1 (d0, q0, r0, m0, 'zigbee2mqtt/Dining Room Dimmer Light', ... (147 bytes))
2022-09-30 22:55:26: Received PUBLISH from mqttjs_3516dda1 (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (123 bytes))
2022-09-30 22:55:26: Sending PUBLISH to mqttjs_3516dda1 (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (123 bytes))
2022-09-30 22:55:26: Received PUBLISH from mqttjs_3516dda1 (d0, q0, r0, m0, 'zigbee2mqtt/Dining Room Dimmer Light/type', ... (15 bytes))
2022-09-30 22:55:26: Sending PUBLISH to mqttjs_3516dda1 (d0, q0, r0, m0, 'zigbee2mqtt/Dining Room Dimmer Light/type', ... (15 bytes))
2022-09-30 22:55:26: Received PUBLISH from 06gRVjwYvulrbWmUa6Umwo (d0, q0, r0, m0, 'zigbee2mqtt/Back Lounge Dimmer Light', ... (16 bytes))
2022-09-30 22:55:26: Sending PUBLISH to 06gRVjwYvulrbWmUa6Umwo (d0, q0, r0, m0, 'zigbee2mqtt/Back Lounge Dimmer Light', ... (16 bytes))
2022-09-30 22:55:26: Sending PUBLISH to mqttjs_3516dda1 (d0, q0, r0, m0, 'zigbee2mqtt/Back Lounge Dimmer Light', ... (16 bytes))

```
```
2022-09-30 22:58:31Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":11780}' from endpoint 1 with groupID null
Info 2022-09-30 22:58:31MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:58:31MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:58:31Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":12036}' from endpoint 1 with groupID null
Info 2022-09-30 22:58:31MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:58:31MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:58:31Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":6}],"seq":12292}' from endpoint 1 with groupID null
Info 2022-09-30 22:58:31MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:58:31MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:58:31Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":12548}' from endpoint 1 with groupID null
Info 2022-09-30 22:58:31MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:58:31MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:58:40Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":12548}' from endpoint 1 with groupID null
Info 2022-09-30 22:58:40MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:58:40MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:58:40Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,3,23],"type":"Buffer"},"datatype":2,"dp":2}],"seq":12804}' from endpoint 1 with groupID null
Info 2022-09-30 22:58:40MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:58:40MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
Debug 2022-09-30 22:58:40Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":13060}' from endpoint 1 with groupID null
Info 2022-09-30 22:58:40MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
[... cut ...]
Debug 2022-09-30 22:58:40Received Zigbee message from 'Back Lounge Dimmer Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":14}],"seq":14084}' from endpoint 1 with groupID null
Info 2022-09-30 22:58:40MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light', payload '{"brightness":201,"countdown":0,"light_type":"led","linkquality":255,"max_brightness":7,"min_brightness":1,"power_on_behavior":"off","state":"ON"}'
Info 2022-09-30 22:58:40MQTT publish: topic 'zigbee2mqtt/Back Lounge Dimmer Light/type', payload '{"light":"led"}'
```

Koenkk, 2022-09-30:
Whoops, instead of `zigbee2mqtt/Back Lounge Dimmer Light` you should use `zigbee2mqtt/Back Lounge Dimmer Light/set`, so please try the payloads from https://github.com/Koenkk/zigbee2mqtt/issues/13800#issuecomment-1263439931 again.

shanelord01, 2022-09-30:
> zigbee2mqtt/Back Lounge Dimmer Light/set

Hah... OK that works...

Now. Changed the brightness to 30 to be more noticeable.

1. {"brightness": 30}
If OFF, does not turn ON. If ON, the brightness adjusts with no jump beforehand (nice).

2. {"state": "ON", "brightness": 30} 
If OFF, turns ON at 100%, then dims to the level set in brightness. If ON, jumps to 100% then dims to level set in brightness.

3. {"brightness": 30, "state": "ON"}
If OFF, turns ON at 100%, then dims to the level set in brightness. If ON, jumps to 100% then dims to level set in brightness.

Hope this helps.

Koenkk, 2022-09-30:
What happens if the bulb is OFF, you publish `{"brightness": 30}` and shortly after that `{"state": ON}`, does the jump occur?

shanelord01, 2022-09-30:
Seems like {"state": "ON"} for this switch is ON/100%, then adjust to brightness setting, rather than start at previous brightness level.

Koenkk, 2022-09-30:
- So the OFF -> ON brightness jumps seems to be inevitable, just to be sure  can you check if it shows the same behavior with the TuYa gateway 
- The best thing we can do is not publish the ON when the switch is already ON and the brightness is changed. This prevents the brightness jumps when the device is already ON. I've updated https://gist.github.com/Koenkk/9c36da807c3768a85aaed9a352c74e96 so when the switch is already ON and you publish `{"state":"ON", "brightness": 30}` no jumping should occur.

shanelord01, 2022-10-01:
> * So the OFF -> ON brightness jumps seems to be inevitable, just to be sure  can you check if it shows the same behavior with the TuYa gateway
> * The best thing we can do is not publish the ON when [...]

Brilliant! This is all working perfectly. Thankyou!

Koenkk, 2022-10-01:
Great, I've integrated the changes which will be included in the next z2m release (1 hour from now)

iJuiceman, 2022-10-02:
I have been following this thread, as I have the same problems as @shanelord01 
I have updated my .js to the latest code, and while I am not seeing the errors anymore, I am being spammed in the debug/info log for this dimmer.  I am about ready to rip the thing off the wall.
Any suggestions ?

debug 2022-10-02 15:52:27Received Zigbee message from 'Toyroom Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,1,202],"type":"Buffer"},"datatype":2,"dp":2}],"seq":4609}' from endpoint 1 with groupID 0
Info 2022-10-02 15:52:27MQTT publish: topic 'zigbee2mqtt/Toyroom Light', payload '{"brightness":116,"countdown":0,"light_type":"led","linkquality":68,"max_brightness":6,"min_brightness":0,"power_on_behavior":"off","state":"ON"}'
Info 2022-10-02 15:52:27MQTT publish: topic 'zigbee2mqtt/Toyroom Light/type', payload '{"light":"led"}'
Debug 2022-10-02 15:52:27Received Zigbee message from 'Toyroom Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,1],"type":"Buffer"},"datatype":2,"dp":3}],"seq":4865}' from endpoint 1 with groupID 0
Info 2022-10-02 15:52:27MQTT publish: topic 'zigbee2mqtt/Toyroom Light', payload '{"brightness":116,"countdown":0,"light_type":"led","linkquality":68,"max_brightness":6,"min_brightness":0,"power_on_behavior":"off","state":"ON"}'
Info 2022-10-02 15:52:27MQTT publish: topic 'zigbee2mqtt/Toyroom Light/type', payload '{"light":"led"}'
Debug 2022-10-02 15:52:28Received Zigbee message from 'Toyroom Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":4,"dp":4}],"seq":5121}' from endpoint 1 with groupID 0
Info 2022-10-02 15:52:28MQTT publish: topic 'zigbee2mqtt/Toyroom Light', payload '{"brightness":116,"countdown":0,"light_type":"led","linkquality":72,"max_brightness":6,"min_brightness":0,"power_on_behavior":"off","state":"ON"}'
Info 2022-10-02 15:52:28MQTT publish: topic 'zigbee2mqtt/Toyroom Light/type', payload '{"light":"led"}'
Debug 2022-10-02 15:52:28Received Zigbee message from 'Toyroom Light', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,24],"type":"Buffer"},"datatype":2,"dp":5}],"seq":5377}' from endpoint 1 with groupID 0
Info 2022-10-02 15:52:28MQTT publish: topic 'zigbee2mqtt/Toyroom Light', payload '{"brightness":116,"countdown":0,"light_type":"led","linkquality":68,"max_brightness":6,"min_brightness":0,"power_on_behavior":"off","state":"ON"}'

[10 more replies]

closed 2022-11-08 by Koenkk