Koenkk/zigbee2mqtt #13801: Firmware updates fail for Aqara Curtain Motor E1 (ZNCLBL01LM) by Xiaomi

buhito81, 2022-09-01:
### What happened?

Zigbee2MQTT OTA Menu indicates firmware updates are available for the Aqara Curtain Motor E1 (ZNCLBL01LM).

[image]

However updating results in something that _appears_ to all go fine, but the firmware isn't updated in the end. I've tried power-switching the devices, deleting them and re-adding them - always the same result. 

`Zigbee2MQTT:info  2022-09-01 12:36:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Livingroom Curtain Front Right' at 99.89%, ≈ 0 minutes remaining","meta":{"device":"Livingroom Curtain Front Right","progress":99.89,"status":"update_progress"},"type":"ota_update"}'
Zigbee2MQTT:debug 2022-09-01 12:36:43: Got upgrade end request for '0x54ef44100050037e': {"status":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":6168}
Zigbee2MQTT:debug 2022-09-01 12:36:43: Update succeeded, waiting for device announce
Zigbee2MQTT:info  2022-09-01 12:36:43: Update of 'Livingroom Curtain Front Right' at 100.00%
Zigbee2MQTT:info  2022-09-01 12:36:43: MQTT publish: topic 'zigbee2mqtt/Livingroom Curtain Front Right', payload '{"action":null,"battery":98,"charging":false,"device_temperature":31,"hooks_state":"unlocked","linkquality":255,"motor_state":"stopped","position":0,"power_outage_count":5,"power_source":"battery","running":false,"state":"CLOSE","target_position":100,"update":{"progress":100,"remaining":2,"state":"updating"},"update_available":false,"voltage":3000}'
Zigbee2MQTT:info  2022-09-01 12:36:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Livingroom Curtain Front Right' at 100.00%","meta":{"device":"Livingroom Curtain Front Right","progress":100,"status":"update_progress"},"type":"ota_update"}'
Zigbee2MQTT:debug 2022-09-01 12:36:47: Received Zigbee message from 'Dressingroom Cover', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,23,5,33,2,0,9,33,0,1,10,33,207,199,11,32,0,12,32,1,13,35,29,15,0,0,17,35,1,0,0,0,100,32,76,101,32,58,102,32,128,103,32,19,105,32,2,106,33,10,29,107,32,0],"type":"Buffer"}}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2022-09-01 12:36:47: lumi.curtain.acn002: Processed buffer into data {"3":23,"5":2,"9":256,"10":51151,"11":0,"12":1,"13":3869,"17":1,"100":76,"101":58,"102":128,"103":19,"105":2,"106":7434,"107":0}
Zigbee2MQTT:debug 2022-09-01 12:36:47: lumi.curtain.acn002: unknown key 13 with value 3869
Zigbee2MQTT:debug 2022-09-01 12:36:47: lumi.curtain.acn002: unknown key 17 with value 1
Zigbee2MQTT:debug 2022-09-01 12:36:47: lumi.curtain.acn002: Processed data into payload {"device_temperature":23,"power_outage_count":1,"state":"OFF","battery":58}
Zigbee2MQTT:debug 2022-09-01 12:36:47: lumi.curtain.acn002: Processed data into payload {"device_temperature":23,"power_outage_count":1,"state":"OFF","battery":58}
Zigbee2MQTT:info  2022-09-01 12:36:47: MQTT publish: topic 'zigbee2mqtt/Dressingroom Cover', payload '{"battery":58,"charging_status":false,"device_temperature":23,"illuminance":0,"illuminance_lux":0,"linkquality":255,"motor_state":"pause","position":76,"power_outage_count":1,"running":false,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:debug 2022-09-01 12:36:55: Got device announce or timed out, call resolve
Zigbee2MQTT:debug 2022-09-01 12:36:55: Device 'Livingroom Curtain Front Right' announced itself
Zigbee2MQTT:info  2022-09-01 12:36:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Livingroom Curtain Front Right","ieee_address":"0x54ef44100050037e"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-09-01 12:36:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Livingroom Curtain Front Right"},"type":"device_announced"}'
Zigbee2MQTT:info  2022-09-01 12:36:55: Finished update of 'Livingroom Curtain Front Right'
Zigbee2MQTT:info  2022-09-01 12:36:55: MQTT publish: topic 'zigbee2mqtt/Livingroom Curtain Front Right', payload '{"action":null,"battery":98,"charging":false,"device_temperature":31,"hooks_state":"unlocked","linkquality":255,"motor_state":"stopped","position":0,"power_outage_count":5,"power_source":"battery","running":false,"state":"CLOSE","target_position":100,"update":{"state":"idle"},"update_available":false,"voltage":3000}'
Zigbee2MQTT:debug 2022-09-01 12:36:56: Received Zigbee message from 'Livingroom Curtain Front Right', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[1,33,184,11,3,40,31,5,33,7,0,10,33,152,1,12,32,1,13,35,24,25,0,0,100,32,2,101,32,196,102,32,3,103,32,0,104,33,25,1,105,32,1,106,32,0],"type":"Buffer"}}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2022-09-01 12:36:56: lumi.curtain.acn003,lumi.curtain.agl001: Processed buffer into data {"1":3000,"3":31,"5":7,"10":408,"12":1,"13":6424,"100":2,"101":196,"102":3,"103":0,"104":281,"105":1,"106":0}
Zigbee2MQTT:debug 2022-09-01 12:36:56: lumi.curtain.acn003,lumi.curtain.agl001: unknown key 13 with value 6424
Zigbee2MQTT:debug 2022-09-01 12:36:56: lumi.curtain.acn003,lumi.curtain.agl001: unknown key 104 with value 281
Zigbee2MQTT:debug 2022-09-01 12:36:56: lumi.curtain.acn003,lumi.curtain.agl001: Processed data into payload {"voltage":3000,"device_temperature":31,"power_outage_count":6,"battery":98}
Zigbee2MQTT:debug 2022-09-01 12:36:56: lumi.curtain.acn003,lumi.curtain.agl001: Processed data into payload {"voltage":3000,"device_temperature":31,"power_outage_count":6,"battery":98}
Zigbee2MQTT:info  2022-09-01 12:36:56: Configuring 'Livingroom Curtain Front Right'
Zigbee2MQTT:info  2022-09-01 12:36:56: MQTT publish: topic 'zigbee2mqtt/Livingroom Curtain Front Right', payload '{"action":null,"battery":98,"charging":false,"device_temperature":31,"hooks_state":"unlocked","linkquality":255,"motor_state":"stopped","position":0,"power_outage_count":6,"power_source":"battery","running":false,"state":"CLOSE","target_position":100,"update":{"state":"idle"},"update_available":false,"voltage":3000}'
Zigbee2MQTT:debug 2022-09-01 12:36:56: Received Zigbee message from 'Livingroom Curtain Front Right', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":196}' from endpoint 1 with groupID null
Zigbee2MQTT:info  2022-09-01 12:36:56: MQTT publish: topic 'zigbee2mqtt/Livingroom Curtain Front Right', payload '{"action":null,"battery":98,"charging":false,"device_temperature":31,"hooks_state":"unlocked","linkquality":255,"motor_state":"stopped","position":0,"power_outage_count":6,"power_source":"battery","running":false,"state":"CLOSE","target_position":100,"update":{"state":"idle"},"update_available":false,"voltage":3000}'
Zigbee2MQTT:debug 2022-09-01 12:36:56: Received Zigbee message from 'Livingroom Curtain Front Right', type 'readResponse', cluster 'aqaraOpple', data '{"1035":0}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2022-09-01 12:36:56: lumi.curtain.acn003,lumi.curtain.agl001: Processed data into payload {"voltage":0}
Zigbee2MQTT:info  2022-09-01 12:36:56: MQTT publish: topic 'zigbee2mqtt/Livingroom Curtain Front Right', payload '{"action":null,"battery":98,"charging":false,"device_temperature":31,"hooks_state":"unlocked","linkquality":255,"motor_state":"stopped","position":0,"power_outage_count":6,"power_source":"battery","running":false,"state":"CLOSE","target_position":100,"update":{"state":"idle"},"update_available":false,"voltage":0}'
Zigbee2MQTT:debug 2022-09-01 12:36:56: Received Zigbee message from 'Livingroom Curtain Front Right', type 'readResponse', cluster 'aqaraOpple', data '{"1064":0}' from endpoint 1 with groupID null
Zigbee2MQTT:debug 2022-09-01 12:36:56: lumi.curtain.acn003,lumi.curtain.agl001: Processed data into payload {"hooks_state":"unlocked"}
Zigbee2MQTT:info  2022-09-01 12:36:56: MQTT publish: topic 'zigbee2mqtt/Livingroom Curtain Front Right', payload '{"action":null,"battery":98,"charging":false,"device_temperature":31,"hooks_state":"unlocked","linkquality":255,"motor_state":"stopped","position":0,"power_outage_count":6,"power_source":"battery","running":false,"state":"CLOSE","target_position":100,"update":{"state":"idle"},"update_available":false,"voltage":0}'
Zigbee2MQTT:debug 2022-09-01 12:36:56: Received Zigbee message from 'Livingroom Curtain Front Right', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID null
[... cut ...]
Zigbee2MQTT:info  2022-09-01 12:38:29: MQTT publish: topic 'zigbee2mqtt/Diningroom Light Sensor', payload '{"battery":100,"illuminance":32436,"illuminance_lux":1752,"linkquality":255,"power_outage_count":16,"voltage":3090}'`

### What did you expect to happen?

...updated firmware

### How to reproduce it (minimal and precise)

Press Update Firmware on ZNCLBL01LM device in Zigbee2MQTT and wait

### Zigbee2MQTT version

1.26.0-1, 1.27.0-1, 1.27.1-1

### Adapter firmware version

 0x26720700

### Adapter

 ConBee2/RaspBee2

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9469339/log.txt)

Koenkk, 2022-09-05:
I think the device rejects the firmware, @otnow do you know if this is the correct firmware for the ZNCLBL01LM?

Otnow, 2022-09-05:
> I think the device rejects the firmware, @Otnow do you know if this is the correct firmware for the ZNCLBL01LM?

At least there are no errors when checking the OTA file with [zigpy](https://github.com/zigpy/zigpy-cli#ota) 🤷‍♂️:

```console
>zigpy ota info 20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota

Header: OTAImageHeader(upgrade_file_id=200208670, header_version=256, header_length=56, field_control=<FieldControl.0: 0>, manufacturer_id=4447, image_type=4105, file_version=6168, stack_version=2, header_string='CURTAIN-OCC-JN5189---ENCRYPTED00', image_size=288213, *device_specific_file=False, *hardware_versions_present=False, *key=ImageKey(manufacturer_id=4447, image_type=4105), *security_credential_version_present=False)
Number of subelements: 2
Validation result: ValidationResult.UNKNOWN
```

---

Judging by the [log](https://github.com/Koenkk/zigbee2mqtt/files/9469339/log.txt), the firmware update is going well, but it seems that this is the same firmware version.

It seems to me that the problem is that the device reports an incorrect version of its own firmware (`"fileVersion":24, "dateCode":"Apr 22 2022", "swBuildId":"0.0.0_0024"`) compared to the one contained in the headers of the OTA update file (`"fileVersion":6168, 0.0.0_2424_20220422`), although this file was flashed into the device.

And because the version in the headers of the OTA update file is higher than the device reports, then _z2m_ constantly reports an available update.

```js
debug 2022-09-01 12:01:11: Received Zigbee message from 'Livingroom Curtain Front Right', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":24,"imageType":4105,"manufacturerCode":4447}' from endpoint 1 with groupID null
debug 2022-09-01 12:01:11: Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":24}'
debug 2022-09-01 12:01:11: ZigbeeOTA: downloaded main index
debug 2022-09-01 12:01:11: Is new image available for '0x54ef44100050037e', current '{"fieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":24}', latest meta '{"fileVersion":6168,"fileSize":288213,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota","sha512":"0d1bae25759420898d381b98e97858d2e47f48cd3495abd8754661d0e1957eb5b570a2ae7435acc435d5a18665409e07148abae6f1f4a190c9af786d94ec725d"}'
debug 2022-09-01 12:01:11: Update available for '0x54ef44100050037e': YES
info  2022-09-01 12:01:11: Update available for 'Livingroom Curtain Front Right'

...

debug 2022-09-01 12:02:22: Received Zigbee message from 'Livingroom Curtain Front Right', type 'readResponse', cluster 'genBasic', data '{"dateCode":"Apr 22 2022","swBuildId":"0.0.0_0024"}' from endpoint 1 with groupID null
debug 2022-09-01 12:02:22: Updating to latest '0x54ef44100050037e' (lumi.curtain.agl001)
debug 2022-09-01 12:02:22: Using endpoint '1'
debug 2022-09-01 12:02:23: Received Zigbee message from 'Livingroom Curtain Front Right', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":24,"imageType":4105,"manufacturerCode":4447}' from endpoint 1 with groupID null
debug 2022-09-01 12:02:23: Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":24}'
debug 2022-09-01 12:02:23: ZigbeeOTA: downloaded main index
debug 2022-09-01 12:02:23: getNewImage for '0x54ef44100050037e', meta {"fileVersion":6168,"fileSize":288213,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota","sha512":"0d1bae25759420898d381b98e97858d2e47f48cd3495abd8754661d0e1957eb5b570a2ae7435acc435d5a18665409e07148abae6f1f4a190c9af786d94ec725d"}
debug 2022-09-01 12:02:23: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota
debug 2022-09-01 12:02:24: OTA update checksum validation succeeded for '0x54ef44100050037e'
debug 2022-09-01 12:02:24: getNewImage for '0x54ef44100050037e', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":6168,"zigbeeStackVersion":2,"otaHeaderString":"CURTAIN-OCC-JN5189---ENCRYPTED00","totalImageSize":288213}
debug 2022-09-01 12:02:24: Got new image for '0x54ef44100050037e'
debug 2022-09-01 12:02:24: Starting upgrade
debug 2022-09-01 12:02:24: Received Zigbee message from 'Livingroom Curtain Front Right', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":24,"imageType":4105,"manufacturerCode":4447}' from endpoint 1 with groupID null
debug 2022-09-01 12:02:25: OTA update at 0%, remaining Infinity seconds
info  2022-09-01 12:02:25: Update of 'Livingroom Curtain Front Right' at 0.00%

...

debug 2022-09-01 12:36:43: Got upgrade end request for '0x54ef44100050037e': {"status":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":6168}
debug 2022-09-01 12:36:43: Update succeeded, waiting for device announce
info  2022-09-01 12:36:43: Update of 'Livingroom Curtain Front Right' at 100.00%

...

debug 2022-09-01 12:36:55: Got device announce or timed out, call resolve
[... cut ...]
debug 2022-09-01 13:06:22: Is new image available for '0x54ef44100050037e', current '{"fieldControl":0,"manufacturerCode":4447,"imageType":4105,"fileVersion":24}', latest meta '{"fileVersion":6168,"fileSize":288213,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20220506181329_lumi.curtain.agl001_Multi_JN5189_FMSH_0.0.0_2424_20220422_1aa302.ota","sha512":"0d1bae25759420898d381b98e97858d2e47f48cd3495abd8754661d0e1957eb5b570a2ae7435acc435d5a18665409e07148abae6f1f4a190c9af786d94ec725d"}'
debug 2022-09-01 13:06:22: Update available for '0x54ef44100050037e': YES
info  2022-09-01 13:06:22: Update available for 'Livingroom Curtain Front Right'

```

buhito81, 2022-09-06:
But is it actually different firmware? If yes, there must be a way to fiddle with code a bit or alter the check for the firmware slightly to update the firmware. FileVersion is different, if I understand you correctly...

Otnow, 2022-09-06:
> But is it actually different firmware? If yes, there must be a way to fiddle with code a bit or alter the check for the firmware slightly to update the firmware. FileVersion is different, if I understand you correctly...

You have an up-to-date firmware version on the devices (the same as in the OTA update file), but due to the fact that they report an incorrect version number (`0.0.0_0024` instead of `0.0.0_2424`) _z2m_ constantly reports an available update.

This is Aqara's fault and there's nothing to be done about it other than just ignore it.

closed 2022-09-06 by buhito81

YPyltiai, 2023-05-29:
@Otnow , check this out:

```
Debug 2023-05-28 23:36:07lumi.curtain.acn003,lumi.curtain.agl001: unknown key 13 with value 6168
```
^ this shows up in debug log. I guess key 13 might actually be the installed file version and then the one currently used as id is just a title/name. In this case this key 13 would match latest version and likely wouldn’t cause this miss-match. It’s just a guess though but I cannot understand why this “unknown” key would consistently be advertising the same number as the latest file version is.

Otnow, 2024-01-31:
> @Otnow , check this out:
>
> ```
> Debug 2023-05-28 23:36:07lumi.curtain.acn003,lumi.curtain.agl001: unknown key 13 with value 6168
> ```
>
> ^ this shows up in debug log. I guess key 13 might actually be the [...]

Yes, that's right, key 13, among other things, was used in merged [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6983), which fixing current problem.

Note for PR: when pairing, device will immediately display correct firmware version, otherwise you need to wait until device sends next heartbeat message (~ 6-6,5 minutes) or perform its manual reconfiguration and then restart zigbee2mqtt (so that correct version is displayed in web interface).