Koenkk/zigbee2mqtt #13805: TuYa TS130F does not react on open/close command

YuriyDY, 2022-09-01:
### What happened?

TuYa TS130F does not react on open/close command after upgrade to version 1.27.2-1. 
In version 1.27.0-1 everething work as expected.

### How to reproduce it (minimal and precise)

I send open/close command from zigbee2mqtt or home assistant to the TS130F switch but it is not rected on it. 

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

20220301

### Adapter

DIY CC2652 based

convicte, 2022-09-01:
I have identified a similar problem and brought it up in the Z2m discord.
In the meantime, I've managed to figure out the most probable issue you are having - since I had it as well.

In the recent update, even though it's not tracked - the cover_invert parameter was flipped for this shutter.
At the same time the motor_revers parameter didn't so what ends up happening is that your physical state of the shutter (open/closed) ends up desynchronized with the programmatic state.

In my case, I would click, for example, close, and the shutter would show moving up (not down) while the shutter wouldn't be moving. The motor would have been already fully up, so the power would have been cut, while the timer inside the shutter controller was running, until it stopped and showed the reversed position (open).

I know its a very poor description of the state of affairs, but please try to change the cover_inverse and reverse_motor to the opposite states as compared to what they used to be. Then actuate the shutter manually to do a full up and down cycle. This should realign the physical state with the programmatic state of the shutters.

An absolute mess that took hours to figure out, but it worked for me.

YuriyDY, 2022-09-01:
No, your description is enough :)
I confirme that cover_invert parameter is flip now, but in my case looks like I have some additional issue.

In version 1.27.0-1 when I click to open shutters in log I have this messages:
`Info 2022-09-01 22:58:56MQTT publish: topic 'zigbee2mqtt/Living room Shutter role/mode', payload '{"backlight":"LOW"}'
Info 2022-09-01 22:59:02MQTT publish: topic 'zigbee2mqtt/Living room Shutter role', payload '{"backlight_mode":"LOW","calibration":"OFF","calibration_time":15.1,"last_seen":"2022-09-01T19:59:01.635Z","linkquality":32,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"CLOSE"}'
Info 2022-09-01 22:59:02MQTT publish: topic 'zigbee2mqtt/Living room Shutter role/mode', payload '{"backlight":"LOW"}'
Info 2022-09-01 22:59:02MQTT publish: topic 'zigbee2mqtt/Living room Shutter role', payload '{"backlight_mode":"LOW","calibration":"OFF","calibration_time":15.1,"last_seen":"2022-09-01T19:59:01.655Z","linkquality":32,"motor_reversal":"OFF","moving":"UP","position":0,"state":"CLOSE"}'
Info 2022-09-01 22:59:02MQTT publish: topic 'zigbee2mqtt/Living room Shutter role/mode', payload '{"backlight":"LOW"}'
Info 2022-09-01 22:59:17MQTT publish: topic 'zigbee2mqtt/Living room Shutter role', payload '{"backlight_mode":"LOW","calibration":"OFF","calibration_time":15.1,"last_seen":"2022-09-01T19:59:16.831Z","linkquality":25,"motor_reversal":"OFF","moving":"STOP","position":100,"state":"CLOSE"}'
Info 2022-09-01 22:59:17MQTT publish: topic 'zigbee2mqtt/Living room Shutter role/mode', payload '{"backlight":"LOW"}'`

In a Logbook of HomeAssistant:
[image]

In version 1.27.2-1 when I click to open shutters in log I have this messages:
`01T19:48:08.777Z","linkquality":25,"motor_reversal":"ON","moving":"UP","position":0,"state":"CLOSE"}'
Info 2022-09-01 22:48:09MQTT publish: topic 'zigbee2mqtt/Living room Shutter role/mode', payload '{"backlight":"LOW"}'
Info 2022-09-01 22:48:09MQTT publish: topic 'zigbee2mqtt/Living room Shutter role', payload '{"backlight_mode":"LOW","calibration":"OFF","calibration_time":15.1,"last_seen":"2022-09-01T19:48:08.811Z","linkquality":25,"motor_reversal":"ON","moving":"UP","position":0,"state":"CLOSE"}'
Info 2022-09-01 22:48:09MQTT publish: topic 'zigbee2mqtt/Living room Shutter role/mode', payload '{"backlight":"LOW"}'
Info 2022-09-01 22:48:09MQTT publish: topic 'zigbee2mqtt/Living room Shutter role', payload '{"backlight_mode":"LOW","calibration":"OFF","calibration_time":15.1,"last_seen":"2022-09-01T19:48:08.911Z","linkquality":29,"motor_reversal":"ON","moving":"STOP","position":0,"state":"CLOSE"}'
Info 2022-09-01 22:48:09MQTT publish: topic 'zigbee2mqtt/Living room Shutter role/mode', payload '{"backlight":"LOW"}'
`

In a Logbook of HomeAssistant:
[image]

Please look at timestamp of each tryes on a screenshoots. In the 1.27.2-1 version shutters swiched from "moving" station to "stop" very quickly. In a fact it doesn't move.

convicte, 2022-09-01:
Actually I spoke too soon, since I still am suffering from the state being flipped, even with all the things changed

```yaml

Info 2022-09-01 23:45:55MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":32,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"CLOSE"}'
Info 2022-09-01 23:45:55MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-01 23:46:02MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":32,"motor_reversal":"OFF","moving":"DOWN","position":100,"state":"OPEN"}'
Info 2022-09-01 23:46:02MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-01 23:46:24MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":36,"motor_reversal":"OFF","moving":"STOP","position":100,"state":"OPEN"}'
Info 2022-09-01 23:46:24MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-01 23:46:29MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":29,"motor_reversal":"OFF","moving":"UP","position":0,"state":"CLOSE"}'
Info 2022-09-01 23:46:29MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-01 23:46:50MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":29,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"CLOSE"}'
Info 2022-09-01 23:46:50MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-01 23:46:58MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":32,"motor_reversal":"OFF","moving":"DOWN","position":100,"state":"OPEN"}'
Info 2022-09-01 23:46:58MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-01 23:47:19MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":32,"motor_reversal":"OFF","moving":"STOP","position":100,"state":"OPEN"}'
Info 2022-09-01 23:47:19MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
```

3 actions:
Open -> Closed
Closed -> Open
Open -> Closed
Same outcome - state reversed as compared to physical cover state.

convicte, 2022-09-01:
For the sake of completeness I've rerun the test with the same open->close->open scenario and the position logic is correct for the inverse_cover:true setting but ther state is equally incorrect:

```yaml
Info 2022-09-02 00:05:11MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":32,"motor_reversal":"OFF","moving":"DOWN","position":0,"state":"OPEN"}'
Info 2022-09-02 00:05:11MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-02 00:05:33MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":32,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"OPEN"}'
Info 2022-09-02 00:05:33MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-02 00:05:44MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":32,"motor_reversal":"OFF","moving":"UP","position":100,"state":"CLOSE"}'
Info 2022-09-02 00:05:44MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-02 00:06:06MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":36,"motor_reversal":"OFF","moving":"STOP","position":100,"state":"CLOSE"}'
Info 2022-09-02 00:06:06MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-02 00:06:13MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":32,"motor_reversal":"OFF","moving":"DOWN","position":0,"state":"OPEN"}'
Info 2022-09-02 00:06:13MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'
Info 2022-09-02 00:06:34MQTT publish: topic 'zigbee2mqtt/Office shutter', payload '{"calibration":"OFF","calibration_time":null,"last_seen":"2022-08-12T18:05:32+02:00","linkquality":32,"motor_reversal":"OFF","moving":"STOP","position":0,"state":"OPEN"}'
Info 2022-09-02 00:06:34MQTT publish: topic 'zigbee2mqtt/Office shutter/mode', payload '{}'

```

McGiverGim, 2022-09-05:
I continue here with my investigation: the problem was with this commits:
https://github.com/Koenkk/zigbee-herdsman-converters/commit/1cc7c2301d58233ef541d7e1ecb746414d02ca8a
https://github.com/Koenkk/zigbee-herdsman-converters/commit/2bc158c3cc20ee45bd8b39fdf84df49f21fc0175

Now, it has a difference, if the cover is inverted, it must have a `meta: {coverInverted: true},` in the definition of the model. The all `invert_cover` is only used to change the position of the cover, to let 100 be open or close, but not the open/close state. The state is get from the position and the meta tag. So now we can invert the "position" but not the "open/close" state, that must be defined in the meta tags.

I've tried to add the meta tag, and REMOVE the `invert_cover` and it works again. I will push a PR adding this meta tag, but is a breaking change and the `invert_cover` must be removed from the configuration. I'm not too sure if this is the correct change, but I will push it and see what @Koenkk has to say.

closed 2022-09-06 by Koenkk

Jens-Wymeersch, 2022-09-06:
@McGiverGim  @convicte So what is the solution to this problem because in my case it doesn't feel like cover_invert doesn't do anything for me.

McGiverGim, 2022-09-06:
You can install the edge/develop version of zigbee2mqtt, or wait until stable version next month. The fix is in the new code.

Jens-Wymeersch, 2022-09-06:
@McGiverGim thank you for your quick reply. How is the fix exactly implemented ? What do I need to do to make this work correctly ?

McGiverGim, 2022-09-06:
The change in the code is linked above, in the PR.