Koenkk/zigbee2mqtt #13806: [New device support]: BAC-3000-ALZB Fan Coil Thermostat

hanfreakingsolo, 2022-09-01:
### Link

https://www.aliexpress.com/item/1005002593757973.html

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0x2c1165fffe99ebcd","nwkAddr":32547,"manufId":4098,"manufName":"_TZE200_dzuqwsyg","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0007","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_dzuqwsyg","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genGroups":{"attributes":{"nameSupport":0}},"genPowerCfg":{"attributes":{}},"genScenes":{"attributes":{"currentScene":0}},"genLevelCtrl":{"attributes":{}},"hvacThermostat":{"attributes":{}},"genIdentify":{"attributes":{}},"hvacPumpCfgCtrl":{"attributes":{}},"manuSpecificTuya_3":{"attributes":{}},"genDeviceTempCfg":{"attributes":{}},"hvacUserInterfaceCfg":{"attributes":{}}},"binds":[{"cluster":61184,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0bcf6","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1662067468154,"defaultSendRequestWhen":"immediate"}

### Comments

This is my first time adding a new device, and I'm totally new to JS so bear with me. I bought this thermostat on Aliexpress, and it showed up as a supported device in zigbee2mqtt with model number BAC-002-ALZB. Problem is, it's model BAC-3000-AZLB.
The default converter sort of worked, but was missing the heat function and allowed 0.5 degree steps. I found a converter someone made to fix these things for the BAC-002 model, and adjusted it a bit. 
Adjustments made:
- Removed 'ON OFF' as a separate state switch, and combined it with system_mode (now 'off', 'cool', 'heat', and 'fan_only) *Personal Preference!
- Fixed controls trying to send half degree adjustments by adding rounding to occupied_heating_heatpoint value. Although I limited the steps to whole degrees, google home still allowed half degree increments which caused the device to spaz out and set itself to zero.

OUTSTANDING PROBLEMS:
- Model number! I can't figure this one out. It identifies as a BAC-002-ALZB because the modelID and manufacturerName are the same. How can the device be identified as a different model? **NOTE** I left the incorrect model in the converter below, that needs to be changed
- Combining off into system_mode works fine, but the device still returns the state 'ON' when it's actually off
- When selecting the device modes in google home, they don't match up and still show 'ON' as an option. The Fan mode also doesn't have an icon. (see screenshot below)
- Half degree increments! Is this possible? The device itself allows this through the touchscreen, but after some reading online I think it's a tuya zigbee limitation? Any way to get around this?

[image]

See the converter below. Things I disabled are left commented out in case I messed anything up, so apologies as the code needs some cleaning.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const ota = require('zigbee-herdsman-converters/lib/ota');
const {repInterval, defaultBindGroup, OneJanuary2000} = require('zigbee-herdsman-converters/lib/constants');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fz_moes_thermostat_custom = {
	cluster: 'manuSpecificTuya',
	type: ['commandGetData', 'commandSetDataResponse', 'commandDataResponse'],
	convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'moes_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);
		let temperature;
		/* See tuyaThermostat above for message structure comment */
		switch (dp) {
		case tuya.dataPoints.moesSchedule:
			return {
				program: [
					{weekdays_p1: value[0] + 'h:' + value[1] + 'm ' + value[2]/2 + '°C'},
					{weekdays_p2: value[3] + 'h:' + value[4] + 'm ' + value[5]/2 + '°C'},
					{weekdays_p3: value[6] + 'h:' + value[7] + 'm ' + value[8]/2 + '°C'},
					{weekdays_p4: value[9] + 'h:' + value[10] + 'm ' + value[11]/2 + '°C'},
					{saturday_p1: value[12] + 'h:' + value[13] + 'm ' + value[14]/2+ '°C'},
					{saturday_p2: value[15] + 'h:' + value[16] + 'm ' + value[17]/2 + '°C'},
					{saturday_p3: value[18] + 'h:' + value[19] + 'm ' + value[20]/2 + '°C'},
					{saturday_p4: value[21] + 'h:' + value[22] + 'm ' + value[23]/2 + '°C'},
					{sunday_p1: value[24] + 'h:' + value[25] + 'm ' + value[26]/2 + '°C'},
					{sunday_p2: value[27] + 'h:' + value[28] + 'm ' + value[29]/2 + '°C'},
					{sunday_p3: value[30] + 'h:' + value[31] + 'm ' + value[32]/2 + '°C'},
					{sunday_p4: value[33] + 'h:' + value[34] + 'm ' + value[35]/2 + '°C'},
				],
			};
		case 1: // system mode (OFF, ON)
			if(value == 0) {
[... 120 lines cut ...]
    {
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_dzuqwsyg'}],
        model: 'BAC-002-ALZB',
        vendor: 'HKGK',
        description: 'BAC series thermostat',
        fromZigbee: [fz_moes_thermostat_custom],
        toZigbee: [tz_system_mode, tz_fan_mode, tz_temperature_setpoint, tz.moes_thermostat_child_lock, tz.moes_thermostat_current_heating_setpoint, tz.moes_thermostat_mode,
            tz.hgkg_thermostat_standby, tz.moes_thermostat_sensor, tz.moes_thermostat_calibration,
            tz.moes_thermostat_deadzone_temperature, tz.moes_thermostat_max_temperature_limit],
        exposes: [e.child_lock(), e.deadzone_temperature(), e.max_temperature_limit(),
			//exposes.switch()
			//	.withState('state', false, 'OFF/ON state'),
            exposes.climate()
				.withSetpoint('occupied_heating_setpoint', 16, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
				.withLocalTemperatureCalibration(ea.STATE_SET)
                .withSystemMode(['off', 'cool', 'heat', 'fan_only'], ea.STATE_SET)
				.withFanMode(['low', 'medium', 'high', 'auto'], ea.STATE_SET)
				.withRunningState(['idle', 'heat', 'cool'], ea.STATE)
                .withPreset(['hold', 'program'])
				.withSensor(['IN', 'AL', 'OU'], ea.STATE_SET)],
        onEvent: tuya.onEventSetLocalTime,
    },
];
```

hanfreakingsolo, 2022-10-02:
Anyone able to help with this?

hanfreakingsolo, 2022-11-03:
Not stale, please help

closed 2022-12-13 by github-actions[bot]

MBrognoli, 2023-01-12:
I have the same problem, can anyone help us?

hanfreakingsolo, 2023-03-28:
Ok changes to the herdsman converters have messed this up even more. Please help, my thermostat is now showing -6000 room temp