Koenkk/zigbee2mqtt #13807: Sonoff ZBDongle-E: Fails to add devices or not responding.

kdober, 2022-09-01:
### What happened?

I changed to the new Sonoff ZBDongle-E and I'm having issues making my devices to work.
I don't have many, just a few aqara roller shade E1 and the cube.
I tried to pair the aqara roller shade E1 many times, which is supported (had it working with the conbee II stick before).
However, I cannot make it work. On the logs you can see two attemps to join it. It joins and the device shows up, but no response to commands and acording to logs it failed to configure it.

I also tried to join the cube. On first few times I tried it was not recognized. Regredly I didn't save the logs and now it doesn't even join to the network (i suspect the issue here is the cube stopped working...)

Thanks :)

### What did you expect to happen?

To work :)

### How to reproduce it (minimal and precise)

Just added the Aqara roller shade E1

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff ZBDongle-E

### Debug log

error 2022-09-01 18:44:01Failed to configure '0x54ef4410003a2d5c', attempt 1 (Error: Read 0x54ef4410003a2d5c/1 aqaraOpple([1034], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34010 - 1 - 10 - 64704 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))
error 2022-09-01 18:44:51Publish 'get' 'position' to '0x54ef4410003a2d5c' failed: 'Error: Read 0x54ef4410003a2d5c/1 genAnalogOutput([85], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34010 - 1 - 11 - 13 - 1 after 10000ms)'
error 2022-09-01 18:45:41Failed to configure '0x54ef4410003a2d5c', attempt 2 (Error: Read 0x54ef4410003a2d5c/1 aqaraOpple([1034], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34010 - 1 - 12 - 64704 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))
error 2022-09-01 18:49:32Publish 'set' 'state' to '0x54ef4410003a2d5c' failed: 'Error: Write 0x54ef4410003a2d5c/1 genAnalogOutput({"85":{"value":100,"type":57}}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34010 - 1 - 15 - 13 - 4 after 10000ms)'
info 2022-09-01 18:51:12Succesfully changed options
info 2022-09-01 18:51:12MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"pz949-1"}'
info 2022-09-01 18:51:19Successfully interviewed '0x54ef4410003a2d5c', device has successfully been paired
info 2022-09-01 18:51:19Device '0x54ef4410003a2d5c' is supported, identified as: Xiaomi Aqara roller shade companion E1 (ZNJLBL01LM)
info 2022-09-01 18:51:19MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara roller shade companion E1","exposes":[{"features":[{"access":7,"name":"state","property":"state","type":"enum","values":["OPEN","CLOSE","STOP"]},{"access":7,"description":"Position of this cover","name":"position","property":"position","type":"numeric","value_max":100,"value_min":0}],"type":"cover"},{"access":5,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Temperature of the device","name":"device_temperature","property":"device_temperature","type":"numeric","unit":"°C"},{"access":5,"description":"The current charging status.","name":"charging_status","property":"charging_status","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"The current state of the motor.","name":"motor_state","property":"motor_state","type":"enum","values":["declining","rising","pause","blocked"]},{"access":1,"description":"Whether the motor is moving or not","name":"running","property":"running","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZNJLBL01LM","options":[{"access":2,"description":"Inverts the cover position, false: open=100,close=0, true: open=0,close=100 (default false).","name":"invert_cover","property":"invert_cover","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"Calibrates the device_temperature value (absolute offset), takes into effect on next report of device.","name":"device_temperature_calibration","property":"device_temperature_calibration","type":"numeric"}],"supports_ota":true,"vendor":"Xiaomi"},"friendly_name":"0x54ef4410003a2d5c","ieee_address":"0x54ef4410003a2d5c","status":"successful","supported":true},"type":"device_interview"}'
info 2022-09-01 18:51:19Configuring '0x54ef4410003a2d5c'
info 2022-09-01 18:51:20MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Aqara roller shade companion E1","friendly_name":"0x54ef4410003a2d5c","model":"ZNJLBL01LM","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'
error 2022-09-01 18:51:40Failed to configure '0x54ef4410003a2d5c', attempt 3 (Error: Read 0x54ef4410003a2d5c/1 aqaraOpple([1034], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 34010 - 1 - 17 - 64704 - 1 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))
info 2022-09-01 18:52:19Zigbee: disabling joining new devices.
info 2022-09-01 18:52:19MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":false},"status":"ok","transaction":"169zc-2"}'

alexberdnik, 2022-09-04:
it looks like the joining issue solved in current dev branch - could you please check? At least it works for me. See this one: https://github.com/Koenkk/zigbee2mqtt/issues/13607

sujithg9, 2022-09-09:
@kdober I might be having a similar issue with one of my cubes. One is working perfectly fine but the other is able to join the network but nothing is being reported like battery, temp anything. 

[image]

Loic691, 2022-09-11:
I have the same problem on Legrand 067647 device.
Nothing is reported with ZBdongle E . It' ok with conbee2

canedje, 2022-09-21:
I don't know if this is the right threath but:

I did install a new sonoff zigbee 3.0 USB dongle plus-ZBDongle -E (based on EFR32MG21)
And did migrate from Deconz to Z2M by repairing all devices one by one (57 devices)
At the and I succeeded, but did counter a lot of pairing trouble with the aqara door sensors (18 devices in my network)
The first ones did paire easely but pairing getting more and more difficult after adding new devices.
The last one took about 15 minutes and severall tries to get it paired

I do use more then enough routers and none of the door sensors where far away or obstructed

Loic691, 2022-09-21:
And same with MOES sensors device which was working well with conbee2

kirovilya, 2022-09-29:
@Loic691 @canedje @kdober @sujithg9 Please check on the latest/edge dev version of z2m - changes have been made to improve device pairing.
But! You need to create a new network! To do this, you should change the channel or panId - in this case, the list of devices will be cleared and you will have to pair the devices again.

canedje, 2022-09-29:
Thanks. That’s not handy.
I’m not willing to repair 57 devices again

kirovilya, 2022-09-29:
@canedje sorry, but at the moment I have not come up with a way to recreate the network and save the existing devices. I will think about how I can do it, but I do not promise.

canedje, 2022-09-29:
Thanks anyway for the effort made.
I do like Z2M much more then Deconz at the moment

Loic691, 2022-09-29:
> Thanks. That’s not handy. I’m not willing to repair 57 devices again. Coming from Deconz I could change device by device running both parallel. Is there no migration option? How do I change channel or panId?

If you want to use 2 zigbee networks in parallel, you have to change the PANID of one of the network.
You can keep the same channel even it's not optimal. For my I had not the choice because of my legrand device wich compatible only with channel 11.

canedje, 2022-09-30:
How can I change a PANID?

kirovilya, 2022-09-30:
@canedje in configuration.yml flie https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#network-config

closed 2022-11-07 by github-actions[bot]

latunnqj-shtangist, 2024-03-23:
The problem persists. The buton is pairing successfully, but doesn't give back battery level, and going to not responding after few hours.