Koenkk/zigbee2mqtt #13809: OTA crashed for IKEA Syrbar Button with unhandled promise rejection

arteck, 2022-09-02:
### What happened?

start ota and look for errors 
2022-09-01 17:58:53.956 - error: zigbee.0 (10519) Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
2022-09-01 17:58:53.957 - error: zigbee.0 (10519) unhandled promise rejection: Timeout - 63764 - 1 - null - 25 - 1 after 30000ms
2022-09-01 17:58:53.959 - error: zigbee.0 (10519) Error: Timeout - 63764 - 1 - null - 25 - 1 after 30000ms
at Timeout._onTimeout (/opt/iobroker/node_modules/**zigbee-herdsman/src/utils/waitress.ts:64:35**)
at listOnTimeout (internal/timers.js:557:17)
at processTimers (internal/timers.js:500:7)
2022-09-01 17:58:53.960 - error: zigbee.0 (10519) Timeout - 63764 - 1 - null - 25 - 1 after 30000ms
2022-09-01 17:58:53.983 - info: zigbee.0 (10519) cleaned everything up...
2022-09-01 17:58:54.339 - info: zigbee.0 (10519) Zigbee: disabling joining new devices.
2022-09-01 17:58:54.368 - info: zigbee.0 (10519) Failed to check if update available for '0x14b457fffe91707c' (Device didn't respond to OTA request)
2022-09-01 17:58:54.485 - info: zigbee.0 (10519) terminating

### How to reproduce it (minimal and precise)

aou need the Button and start ota 

### Zigbee2MQTT version

new

### Adapter firmware version

new

### Adapter

all

arteck, 2022-09-02:
https://github.com/ioBroker/ioBroker.zigbee/issues/1518

tugsi, 2022-09-04:
I entered the issue at ioBroker.
But I still have a more up-to-date Styrbar (with a silver lid), when I register it and try OTA, everything works normally, it checks and says that there is no new firmware (date code: 20200708
build: 1.0.024), but the old Styrbar has the same data.
So I guess it's the switch itself that's causing the adapter to crash.

closed 2022-10-12 by github-actions[bot]