Koenkk/zigbee2mqtt #13812: TS0601_vibration_sensor contact boolean is inverted

Mattg0, 2022-09-02:
### What happened?

I added 2 [Tuya vibration/door sensors](https://www.zigbee2mqtt.io/devices/TS0601_vibration_sensor.html#tuya-ts0601_vibration_sensor) but the contact value is inverted for both devices:

When door is closed = > Contact = false
When doos is open => Contact = true

### What did you expect to happen?

Contact is not inverted

When door is closed = > Contact = true
When doos is open => Contact = false

### How to reproduce it (minimal and precise)

N/A

### Zigbee2MQTT version

 1.27.2 commit: 7dc48fb6

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

### Debug log

Opening the door:
debug 2022-09-02 09:24:59: Received Zigbee message from 'Petite Porte', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4352}' from endpoint 1 with groupID null
info  2022-09-02 09:24:59: MQTT publish: topic 'zigbee2mqtt/Petite Porte', payload '{"battery":92,"contact":true,"linkquality":255,"vibration":false}'

Closing the door:
debug 2022-09-02 09:25:03: Received Zigbee message from 'Petite Porte', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":1}],"seq":4608}' from endpoint 1 with groupID null
info  2022-09-02 09:25:03: MQTT publish: topic 'zigbee2mqtt/Petite Porte', payload '{"battery":92,"contact":false,"linkquality":255,"vibration":false}'

Koenkk, 2022-09-05:
> When door is closed = > Contact = false
When doos is open => Contact = true

@flatsiedatsie can you confirm this?

flatsiedatsie, 2022-09-05:
I don't know. This device is very low quality (very big, poor quality plastic), so I stopped using it.

Koenkk, 2022-09-06:
@Mattg0 pushed a fix! let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-10-15 by github-actions[bot]