Koenkk/zigbee2mqtt #13813: Cover state TS130F incorrect

McGiverGim, 2022-09-02:
### What happened?

I've several TS130F cover switch. They work well, but since latest version, they appear with estate "CLOSE" when they are totally opened. The numeric position is right, the state is wrong.

### What did you expect to happen?

Have the state correct.

### How to reproduce it (minimal and precise)

Open the cover. Look at the state of the device. It will appear as CLOSE.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

Egony E72

### Debug log

I will attach it later.

McGiverGim, 2022-09-02:
I forgot to say that I have `invert_cover: true` so maybe is affecting...

McGiverGim, 2022-09-02:
And it seems the correct state, must be STOP, not OPEN or CLOSE.

McGiverGim, 2022-09-02:
According to the oficial documentation of Home Assistant, the correct states must be one of:
- open, opening, closing and closed
Now the state is "closed" when the cover is totally open. Home Assistant seems to ignore it and uses only the position, so it works well, but some custom component uses the state and shows the cover closed when it is open.
I will investigate further and come back with more information if I can.

McGiverGim, 2022-09-02:
This two commit seem related to the problem. I'm not being able to test changes in local, I will keep trying...
https://github.com/Koenkk/zigbee-herdsman-converters/commit/1cc7c2301d58233ef541d7e1ecb746414d02ca8a
https://github.com/Koenkk/zigbee-herdsman-converters/commit/2bc158c3cc20ee45bd8b39fdf84df49f21fc0175

McGiverGim, 2022-09-02:
@Koenkk I can confirm that reverting the two commits, fixes the bug. But I'm not too sure what is wrong here. I'm a little lost with the invert and the metaInvert...

jvcsw, 2022-09-03:
Same problem here.

convicte, 2022-09-04:
#13805

stefanuytterhoeven, 2022-09-04:
Thx @McGiverGim  for having a look at it!

McGiverGim, 2022-09-05:
I continue in https://github.com/Koenkk/zigbee2mqtt/issues/13805 to have all under the same post

closed 2022-09-06 by Koenkk

stefanuytterhoeven, 2022-10-06:
Zigbee2MQTT
Current version: 1.28.0-1
Still having the same problem: wrong state. 

The cover is OPEN:
when invert_cover : true
position=  100
state= CLOSE
when invert_cover: false
position=0
state= CLOSE
is guess correct setting need to be
invert_cover:true
position=100
State=OPEN

stefanuytterhoeven, 2022-10-10:
> I think you simply have the wires inversed. The open/close state does not change with the invert_cover, only the position.

Ok, maybe that's possible. 
Before I used z2m, I was using Deconz/phoscon. Deconz/phoscon didn't have that problem. 
I will have a look at the wiring though...
thx

McGiverGim, 2022-10-10:
You can invert the wires and the motor_inverted in the worst case

stefanuytterhoeven, 2022-10-10:
reversing the wires for up/down didn't help. When I pushed the UP button on the switch itself, the cover goes down and vice versa. So, I reversed the wires as they were.
While testing in z2m I found something strange (maybe it's normal, don't know)
In the page you get in Z2M when you click on the device, you get the options; About, exposes,bind, 
When I click on "exposes", I see State (Open,Close.Stop), Position, lqi
Current state of the cover itself: OPEN
(Invert_cover: False , but I see it only applies to the percentage)
Z2M: Exposes has the State: Close (so, ZM2 assumes the cover is closed)
Z2M: When I click on "Open" , nothing happens 
Z2M; When I click on "Close", the cover closes. The state changes to Open

The only thing I can do, i try to reverse the wires that connect to Live and neutral. I'm not en electrician, so I don't know if that will help.

McGiverGim, 2022-10-10:
> reversing the wires for up/down didn't help. When I pushed the UP button on the switch itself, the cover goes down and vice versa. So, I reversed the wires as they were.

Reverse the wires as you did, and enable de `motor_reversal` in the configuration of the device. This must work if I'm not wrong. Don't touch the neutral wire.

stefanuytterhoeven, 2022-10-10:
Thanks for your patience :)
where do I find motor_reversal ? Somewhere in Z2M ?

McGiverGim, 2022-10-10:
In the exposes tab of the device must appear, at least in my case.

stefanuytterhoeven, 2022-10-10:
Not in my case. The exposes tab only gives the states, position, and lqi.
 I tried to asdd motor_reversal: true in the settings of the device in the configuration.yaml of Z2M, but it didn't help...
It's strange; when i tell Google Assistant to open the cover, it opens the cover (and vice versa). WHen I ask Google assistant the state, it gives the correct state (probably GA looks at the percentage?)
The cover shutter switch itself works also ok. Only, something is wrong with the "state" in Z2M. Maybe some incompatibilty of the switch? Anyway, Deconz/Phoscon did it right, but maybe they had a workaround.
Thanks for the help. Maybe in the future I find a solution.

McGiverGim, 2022-10-10:
Take a log in debug mode to see if there is some parameter missing that can give the position. z2m only uses the position to decide if it is open or close. 
The problem with tuya devices is that the device is produced by tuya, and it's sold by different brands, with some little adjustments.
z2m uses by default 100 as open and 0 as close, but this device, internally, sends 100 as close and 0 as open (and z2m maps this to close/open). In a debug log you can see if this is true for your device or not.

stefanuytterhoeven, 2022-10-10:
its a https://www.zigbee2mqtt.io/devices/067776.html#legrand-067776 
I'll have al look at it.
thanks...

McGiverGim, 2022-10-10:
Maybe this information was the first you must comment... this issue was only for TS130F. For Legrand, you must do the same that I did here: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609 but here https://github.com/Koenkk/zigbee-herdsman-converters/blob/46fa107af76b08c19f7da1e5626dbc678570de3d/devices/legrand.js#L96-L116

McGiverGim, 2022-10-10:
Nobody has reported it, so it's a change that nobody has done... If you know how to do a PR, you can do it yourself.

stefanuytterhoeven, 2022-10-10:
ow, And I have no idea how to do that...
so, if you update z2m, you have to re-apply your changes?

jvcsw, 2022-10-10:
Hi,

I have several TS130F devices and with the latest version of z2m all covers works fine (deleting invert_cover=true before).

McGiverGim, 2022-10-10:
> ow, And I have no idea how to do that... so, if you update z2m, you have to re-apply your changes?

I can open it later.

McGiverGim, 2022-10-11:
@stefanuytterhoeven PR done here https://github.com/Koenkk/zigbee-herdsman-converters/pull/4770

After merged, it will be available in the Edge version of zigbee2mqtt some hours later, and in the next stable version at the beginning of next month.

stefanuytterhoeven, 2022-10-11:
> Koenkk/zigbee-herdsman-converters#4770

Ha, that would be very nice!
Thanks!

jvcsw, 2022-12-12:
In my case, all working fine.

stefanuytterhoeven, 2022-12-12:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4770 was the trick i guess.
The release after this updates fixed it. But now, with one of the latest releases, this fix is gone? 
so, position 100% is open now while the window cover is closed. Inverted or not, same result :(

JakubMartenka, 2023-01-13:
The same problem...

Koenkk, 2023-01-14:
For those having an inverted state with z2m >= 1.29.0. Can you provide the data/database.db entry of your device?

rbickel, 2023-05-31:
> In my case, all works fine.

Can you share which ts130f you are using? In my cases they all act weird and different. Waiting my tuya hub to upgrade the firmware...

jvcsw, 2023-06-01:
I'm using this (internally is TS130F)

https://www.loratap.com/products/sc400zb-us

JoelSternke, 2024-10-08:
Looks like it's been a while since you posted this so maybe you've already solved this.  I recently ran into this issue and came across your post while searching for a solution.  I had the same problem where the cover state was reversed (said closed when open and open when closed) and when I clicked the up button in the UI, the cover went down and vice versa. I thought I’d share what I did in case it helps others.  I solved this by creating a template cover that referenced the original cover device in Zigbee2MQTT.  I renamed all of my shades in ZigbeeMQTT to add a prefix of MQTT and set them up with Invert Cover option set to True.  In my template, I simply override close_cover and oper_cover to reverse the buttons and the open/close state of the Cover. I now use this template cover in my Automations, Dashboards, and to share with Alexa through emulated Hue and everything works as expected.  
Here is my template.  This just goes in your configuration.yaml file.  In this example, my shade was originally called "Office Blind" and I renamed it in ZigbeeMQTT to "MQTT Office Blind".  This template creates "Office Blind" (cover.office_blind) which references (controls) the MQTT version of it.  But this reverses the open and close actions.  for additional covers, just add them below this one starting with the same indentation as office_blind.  Note that spacing is critical.
`
cover:
  - platform: template
    covers:
      office_blind:
        device_class: shade
        friendly_name: "Office Blind"
        # just take the current position of the MQTT version as the position since it's already inverted there.
        position_template: >-
          {{ state_attr('cover.mqtt_office_blind', 'current_position') | int }}
        # The following reverses the behavior of the up/down buttons and Closed/Open
        open_cover:
          - condition: numeric_state
            entity_id: cover.mqtt_office_blind
            attribute: current_position
            below: 100
          - action: cover.close_cover
            target:
              entity_id: cover.mqtt_office_blind
        close_cover:
          - condition: numeric_state
            entity_id: cover.mqtt_office_blind
            attribute: current_position
            above: 0
          - action: cover.open_cover
            target:
              entity_id: cover.mqtt_office_blind
        # For stop and set cover position, just send these on to the MQTT version
        stop_cover:
          action: cover.stop_cover
          target:
            entity_id: cover.mqtt_office_blind
        set_cover_position:
          action: cover.set_cover_position
          target:
            entity_id: cover.mqtt_office_blind
          data:
            position: "{{ position }}"
        # When the shade is completely open, use the blinds-open icon because it looks cool
        icon_template: >-
          {% if state_attr('cover.mqtt_office_blind' , 'current_position') | int > 99 %}
            mdi:blinds-open
          {% endif %}
        optimistic: false
`

McGiverGim, 2024-10-09:
@JoelSternke in my case it was fixed directly into zigbee2mqtt.

ebozonne, 2025-03-25:
For those like me who still have incorrect position state (not updated) after using the physical button or HA reboot (until you click on stop button), here is a small workaround with an automation (blueprint), until there's a fix:
[https://github.com/ebozonne/ha-blueprints](https://github.com/ebozonne/ha-blueprints)