Koenkk/zigbee2mqtt #13814: Xiaomi devices were showing correct battery percentage before updating to 1.27.2-1

Koltsz, 2022-09-02:
### What happened?

Xiaomi devices were showing the correct or at least a true indication of their battery. since upgrading ZTMQTT to 1.27.2-1 in Home Assistant everything is showing 100% again.

### What did you expect to happen?

the battery percentage to continue working correctly

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0

MichalRybecky, 2022-09-02:
Same issue here, all Aqara (Xiaomi) devices are reporting 100% battery after latest update.

mrochette01, 2022-09-02:
I have seen the same behavior.

sjorge, 2022-09-02:
See the super long and lengthly discussion https://github.com/Koenkk/zigbee2mqtt/issues/8499

Koenkk, 2022-09-05:
As @sjorge mentioned, the battery calculation has been changed.

Battery % calculated is based on the voltage, above 3000mv = 100%, below 2800 mv = 0%, anything in between is linearly scaled.

closed 2022-09-05 by Koenkk