Koenkk/zigbee2mqtt #13818: [New device support]:  Tuya 2022  6 Scene switch  Wired

mahdi89X, 2022-09-02:
### Link

https://www.alibaba.com/product-detail/Home-Intelligent-Multifunctional-System-Tuya-Smart_1600601631848.html?spm=a2700.shop_pl.41413.36.253772ce6Iigcb

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0xa4c138fd1f6495b3","nwkAddr":24471,"manufId":4417,"manufName":"_TZ3002_m3aeya68","powerSource":"Mains (single phase)","modelId":"TS0026","epList":[1,2,3,4,5,6,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":4,"inClusterList":[4,5,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0026","manufacturerName":"_TZ3002_m3aeya68","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genScenes":{"attributes":{"count":0}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc4259","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc4259","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc4259","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc4259","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":4,"inClusterList":[4,5,57344],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":4,"inClusterList":[4,5,57344],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":4,"inClusterList":[4,5,57344],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":4,"inClusterList":[4,5,57344],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":260,"epId":6,"devId":4,"inClusterList":[4,5,57344],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1662128837126,"defaultSendRequestWhen":"immediate"}

### Comments

its a wired  6 scenes switch  from Tuya 
example
1st button : 
 2022-09-02 17:36:26Received Zigbee message from '0xa4c138fd1f6495b3', type 'raw', cluster 'genOnOff', data '{"data":[1,52,253,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Debug 2022-09-02 17:36:26No converter available for 'TS0026' with cluster 'genOnOff' and type 'raw' and data '{"data":[1,52,253,0],"type":"Buffer"}'
2nd button : 
Debug 2022-09-02 17:37:22Received Zigbee message from '0xa4c138fd1f6495b3', type 'raw', cluster 'genOnOff', data '{"data":[1,55,253,0],"type":"Buffer"}' from endpoint 2 with groupID 0
Debug 2022-09-02 17:37:22No converter available for 'TS0026' with cluster 'genOnOff' and type 'raw' and data '{"data":[1,55,253,0],"type":"Buffer"}

please help me add its.. as i bout like 6 pcs to use in my new house :/ and am failing to make a converter

Jeff89L, 2022-09-04:
i need help with the same device  !

mahdi89X, 2022-10-06:
dont close :/

closed 2022-11-15 by github-actions[bot]