Koenkk/zigbee2mqtt #13823: JARVIS / Merrytek _TZE200_i0b1dbqu - led_enable is broken

BradleyFord, 2022-09-02:
### What happened?

Recently purchased and adopted device.

Most data is working correctly, except the "led_enable" does not work and the led keeps flashing

Also when I turn it on or off, in the logs it keeps showing "Data":[1]". Hence confirming that it is still turned on.

### What did you expect to happen?

LED show turn off when it is disabled... (and the bool should go to zero)

### How to reproduce it (minimal and precise)

It seems that their is a request being send, but nothing is changing within the device.

When I look at the toZigbee converter it seems that the developer is trying to target 2 different devices with the same converter by using a if statement. Due to this is it confusing to debug.

The "devices" is searching for 2 difference devices; but then in the converter it is assuming one device, and then dumping all other devices into else.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

20220219

### Adapter

Zigstar

### Debug log

2022-09-03 00:29:20Received MQTT message on 'zigbee2mqtt/0x2c1165fffee14910/set' with data '{"led_enable":false}'
2022-09-03 00:29:20Publishing 'set' 'led_enable' to '0x2c1165fffee14910'
2022-09-03 00:29:20Received Zigbee message from '0x2c1165fffee14910', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":103}],"seq":3080}' from endpoint 1 with groupID 0
2022-09-03 00:29:21Received MQTT message on 'zigbee2mqtt/0x2c1165fffee14910/set' with data '{"led_enable":true}'
2022-09-03 00:29:21Publishing 'set' 'led_enable' to '0x2c1165fffee14910'
2022-09-03 00:29:21Received Zigbee message from '0x2c1165fffee14910', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":1,"dp":103}],"seq":3336}' from endpoint 1 with groupID 0

BradleyFord, 2022-09-02:
@dinhchinh82 @TheEnd98 @chinhhut

dinhchinh82, 2022-09-06:
@BradleyFord we will check and make an update soon.

closed 2022-10-15 by github-actions[bot]