Koenkk/zigbee2mqtt #13825: Philips Hue Tap dial no events

Yasp0, 2022-09-02:
### What happened?

I connected the Philips Hue Tap dial and updated the firmware to the latest version. When I listen to events no events are fired when pushing a button or turning the dial. Also the battery percentage shows 0%, I tried a fresh new battery, same result.

When I setup an automation in Home Assistant I see 3 triggers:
- battery level changes
- got an update available
- became up-to-date 

### What did you expect to happen?

That when I listen to events in Z2M the buttons and dial respond. That there are triggers in Home Assistant like button 1/2/3/4 and the dial.

### How to reproduce it (minimal and precise)

- In Zigbee2MQTT click 'Permit join (All)' button
- The Philips device is getting connected
- Check the battery percentage (says 0% in my case)
- In Home Assistant go to Settings> Integrations > Mosquitto broker > Configure > Listen to a topic (Iisten to #)
- I got no response on any button

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

20220219

### Adapter

Sonoff ZigBee 3.0 USB Dongle Plus (CC2652P)

### Debug log

```
Info 2022-09-02 21:05:53Device '0x001788010d34ba82' joined
Info 2022-09-02 21:05:53MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d34ba82","ieee_address":"0x001788010d34ba82"},"type":"device_joined"}'
Info 2022-09-02 21:05:53MQTT publish: topic 'homeassistant/sensor/0x001788010d34ba82/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x001788010d34ba82 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x001788010d34ba82","unique_id":"0x001788010d34ba82_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Info 2022-09-02 21:05:53MQTT publish: topic 'homeassistant/sensor/0x001788010d34ba82/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0x001788010d34ba82 action","state_topic":"zigbee2mqtt/0x001788010d34ba82","unique_id":"0x001788010d34ba82_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Info 2022-09-02 21:05:53MQTT publish: topic 'homeassistant/sensor/0x001788010d34ba82/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x001788010d34ba82 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x001788010d34ba82","unique_id":"0x001788010d34ba82_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2022-09-02 21:05:53MQTT publish: topic 'homeassistant/sensor/0x001788010d34ba82/update_state/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"0x001788010d34ba82 update state","state_topic":"zigbee2mqtt/0x001788010d34ba82","unique_id":"0x001788010d34ba82_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Info 2022-09-02 21:05:53MQTT publish: topic 'homeassistant/binary_sensor/0x001788010d34ba82/update_available/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"device_class":"update","enabled_by_default":true,"entity_category":"diagnostic","name":"0x001788010d34ba82 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x001788010d34ba82","unique_id":"0x001788010d34ba82_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
Info 2022-09-02 21:05:53Starting interview of '0x001788010d34ba82'
Info 2022-09-02 21:05:53MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010d34ba82","ieee_address":"0x001788010d34ba82","status":"started"},"type":"device_interview"}'
Debug 2022-09-02 21:05:53Received MQTT message on 'homeassistant/sensor/0x001788010d34ba82/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0x001788010d34ba82 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x001788010d34ba82","unique_id":"0x001788010d34ba82_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Debug 2022-09-02 21:05:53Received MQTT message on 'homeassistant/sensor/0x001788010d34ba82/action/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"0x001788010d34ba82 action","state_topic":"zigbee2mqtt/0x001788010d34ba82","unique_id":"0x001788010d34ba82_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Debug 2022-09-02 21:05:53Received MQTT message on 'homeassistant/sensor/0x001788010d34ba82/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0x001788010d34ba82 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x001788010d34ba82","unique_id":"0x001788010d34ba82_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Debug 2022-09-02 21:05:53Received MQTT message on 'homeassistant/sensor/0x001788010d34ba82/update_state/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","name":"0x001788010d34ba82 update state","state_topic":"zigbee2mqtt/0x001788010d34ba82","unique_id":"0x001788010d34ba82_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
Debug 2022-09-02 21:05:53Received MQTT message on 'homeassistant/binary_sensor/0x001788010d34ba82/update_available/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0x001788010d34ba82"],"manufacturer":"Philips","model":"Hue Tap dial switch (8719514440937/8719514440999)","name":"0x001788010d34ba82","sw_version":"2.59.19"},"device_class":"update","enabled_by_default":true,"entity_category":"diagnostic","name":"0x001788 [... cut ...]
Info 2022-09-02 21:06:00Successfully configured '0x001788010d34ba82'
Debug 2022-09-02 21:06:02Received Zigbee message from '0x001788010d34ba82', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":0}' from endpoint 1 with groupID 0
Info 2022-09-02 21:06:02MQTT publish: topic 'zigbee2mqtt/0x001788010d34ba82', payload '{"action":null,"battery":0,"linkquality":25,"update":{"state":null},"update_available":null}'
Debug 2022-09-02 21:06:22Received MQTT message on 'homeassistant/sensor/pc_jasper/availability' with data 'online'
```

NicBrain, 2022-09-09:
I see exactly the same behaviour. I don't use Home Assistant. I've not done any debugging yet, but wanted to add a concurrence to the issue.

closed 2022-10-17 by github-actions[bot]

nareddyt, 2023-02-11:
Did you ever find the root cause? I'm still seeing the same problem

nareddyt, 2023-02-11:
Nevermind, it is because I had `legacy_triggers` disabled in Zigbee2MQTT

https://www.zigbee2mqtt.io/guide/configuration/homeassistant.html#advanced-configuration