Koenkk/zigbee2mqtt #13827: Cover status now inverted (open/closed) for inverted covers

dominicusmento, 2022-09-02:
### What happened?

After update to latest version (1.27.2-1) Covers "end" status not shown correctly in Lovelace.
It is wrong only for covers which have invert: true

It was working fine until now.
Percentages are fine and correct.
Only Open/Closed statuses are now switched.

My covers are TS130F.

### What did you expect to happen?

It was working fine previously.

### How to reproduce it (minimal and precise)

having this in devices.yaml is enough for me..
"0x2c1122fffe12345f":
  friendly_name: Room Cover
  **invert_cover: true**

and then using the slider button card visual on dasboard:
  - type: custom:slider-button-card
    entity: cover.room_cover
    slider:
      direction: top-bottom
      background: striped
    icon:
      tap_action:
        action: more-info
      icon: ''
    action_button:
      mode: custom
      icon: mdi:swap-vertical
    name: Bedroom
    compact: false

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

20211217

### Adapter

Tube's Zigbee Gateway (CC2652P2)

adrienvsj, 2022-09-02:
I confirm the same problem here

jvcsw, 2022-09-03:
Yes, same problem here too.

stefanuytterhoeven, 2022-09-03:
Aha, same here. Glad I'm not the only one. 
Exactly the same problem.

touzenesmy, 2022-09-03:
I have the same problem, even though I have inverted the position of some covers in the settings, they still report as closed when they're open. Can you please help ?
[https://i.imgur.com/YVw2Qbm.png](url)

dominicusmento, 2022-09-04:
#13847
#13813 

seems these are the same..

convicte, 2022-09-05:
FYI
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609

closed 2022-09-06 by Koenkk