Koenkk/zigbee2mqtt #13831: Remote adapter

Jens-Wymeersch, 2022-09-03:
### What happened?

Basically, I'm planning to move my Z2M to my main HA environment and have a remote adapter located where it is now. Before moving, i set up a test environment to see how to make it work before we bring it to production.
I've tested my ZZH! Stick with an HA environment and everything works as expected. So no errors with the FW or identification of the right port. 
I followed the guide https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_adapter.html#_2-b-configure-ser2net-4-0 to the letter

- configuration 2B 

`connection: &con01
  accepter: tcp,20108
  connector: serialdev,/dev/ttyUSB0,115200n81,local,dtr=off,rts=off
  options:
    kickolduser: true`

and added the following to my test Home Assistant environment (where Z2M and MQTT are running)
```
serial:
  port: 'tcp://192.168.2.13:20108'
advanced:
  rtscts: false    
```

### What did you expect to happen?

I expected the Z2M to working

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

zzh!

### Debug log

```
[12:14:32] INFO: Handing over control to Zigbee2MQTT Core ...
Zigbee2MQTT:info  2022-09-03 12:14:32: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-03.12-14-32' filename: log.txt
Zigbee2MQTT:info  2022-09-03 12:14:32: Starting Zigbee2MQTT version 1.27.2 (commit #unknown)
Zigbee2MQTT:info  2022-09-03 12:14:32: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:error 2022-09-03 12:14:32: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-03 12:14:32: Failed to start zigbee
Zigbee2MQTT:error 2022-09-03 12:14:32: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-03 12:14:32: Exiting...
Zigbee2MQTT:error 2022-09-03 12:14:32: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:198:24)
    at Socket.emit (node:events:539:35)
    at emitErrorNT (node:internal/streams/destroy:157:8)
    at emitErrorCloseNT (node:internal/streams/destroy:122:3)
    at processTicksAndRejections (node:internal/process/task_queues:83:21)
 
```

charraeus, 2022-09-27:
I have the same issue. 

**My Home Assistant:**

> * Raspberry Pi 3 B+
> * Home Assistant 2022.9.7
> * Supervisor 2022.09.1
> * Operating System 9.0
> * Frontend 20220907.2 - latest
> * zigbee2mqtt 1.27.2-1

**remote computer:**

> * Raspberry Pi 3 B+ with current Raspian
> * ser2net v4.3.3-1
> * CC2652RB stick
> * zigbee2mqtt installed but disabled. When enabled the zigbee2mqtt-config works flawlessly on this computer (of course another port is configured)

----

Here are my configs and logs. I first tried with socat on HA-computer disabled. Then with socat enabled (see below).

**zigbee2mqtt-config  (only the relevant lines):** 
```
data_path: /config/zigbee2mqtt
socat:
  enabled: false
  master: pty,raw,echo=0,link=/root/ttyZ2M,mode=777,end-close
  slave: tcp4:192.168.1.15:8485,ignoreeof
  options: "-d -d"
  log: true
  restartdelay: 1
  initialdelay: 1
serial:
  port: tcp://192.168.1.15:20108
  adapter: zstack
  disable_led: false
advanced:
  baudrate: 115200
  rtscts: false
  legacy_api: false
device_options:
  legacy: false
```

***/etc/ser2net.yaml* on the remote computer**
```
pi@magicmirror1:~ $ cat /etc/ser2net.yaml 
%YAML 1.1
---
# This is a ser2net configuration file, tailored to be rather
# simple.

define: &banner \r\nser2net port \p device \d [\B] (Debian GNU/Linux)\r\n\r\n

connection: &con01
    accepter: tcp,20108
    options:
      kickolduser: true
    connector: serialdev,/dev/serial/by-id/usb-Silicon_Labs_slae.sh_cc2652rb_stick_-_slaesh_s_iot_stuff_00_12_4B_00_23_93_37_48-if00-port0,
              115200n81,local
```

**ser2net-start on the remote computer**

> pi@magicmirror1:~ $ sudo /usr/sbin/ser2net -d -c /etc/ser2net.yaml -P /run/ser2net.pid

**The resulting log:**
```
16:36:59] INFO: Zigbee Herdsman debug logging enabled
[... 83 lines cut ...]
2022-09-27T12:08:09.530Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2022-09-27T12:08:09.544Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-09-27T12:08:09.555Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-09-27T12:08:09.817Z zigbee-herdsman:adapter:zStack:znp:log Writing CC2530/CC2531 skip bootloader payload
2022-09-27T12:08:09.820Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2022-09-27T12:08:10.824Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-09-27T12:08:10.827Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-09-27T12:08:11.079Z zigbee-herdsman:adapter:zStack:znp:log Skip bootloader for CC2652/CC1352
2022-09-27T12:08:11.540Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-09-27T12:08:11.545Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-09-27T12:08:17.547Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-09-27T12:08:17.550Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2022-09-27T12:08:23.558Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2022-09-27T12:08:23.563Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
Zigbee2MQTT:error 2022-09-27 14:08:29: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-27 14:08:29: Failed to start zigbee
Zigbee2MQTT:error 2022-09-27 14:08:29: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-27 14:08:29: Exiting...
Zigbee2MQTT:error 2022-09-27 14:08:30: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:103:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:132:29)
    at Zigbee.start (/app/lib/zigbee.ts:58:27)
    at Controller.start (/app/lib/controller.ts:101:27)
    at start (/app/index.js:109:5)
```

charraeus, 2022-10-28:
Any news? How can I help you? Which additional information do you need?

closed 2022-12-05 by github-actions[bot]