Koenkk/zigbee2mqtt #13833: Z2M not starting after update "no such file or directory, open 'null/configuration.yaml'"

kkuriata, 2022-09-03:
### What happened?

Z2M stopped working after updating to `1.27.2-1`. I'm using it around 2 years and everything was fine so far. It is installed on Home Assistant (trying to get all my plugins up-to-date, but something went wrong after last update of Z2M).

My HA version:
```
Home Assistant 2022.8.7
Supervisor 2022.08.5
Operating System 8.5
Interfejs użytkownika: 20220802.0 - latest
```

Z2M settings:
```json
// zigbee2mqtt/configuration.yaml
{
  "external_converters": [],
  "devices": [
    "devices.yaml"
  ],
  "groups": [
    "groups.yaml"
  ],
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://core-mosquitto",
    "user": "homeassistant",
    "password": "some_password"
  },
  "serial": {
    "port": "/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0"
  },
  "advanced": {
    "log_level": "warn",
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
         // I removed net key to not share it
    ],
    "availability_blocklist": [],
    "availability_passlist": []
  },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "frontend": {
    "port": 8099
  },
  "experimental": {},
  "availability": false,
  "zigbee_herdsman_debug": true
}
```

When I try to open Z2M UI from left menu panel, I get `502: Bad Gateway`

### What did you expect to happen?

Z2M should start as expected in HA.

### How to reproduce it (minimal and precise)

In my case:
1. Update HA to the given version (currently the latest one).
2. Update Z2M.

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

Not sure how to check that as Z2M UI isn't working

### Adapter

QinHeng Electronics HL-340 USB-Serial adapter / CC2652P

### Debug log

[18:27:04] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)

rwjack, 2022-09-05:
Same, send help

Knodd, 2022-09-05:
Came here hoping for a permanent fix, but here is a temporary solution:
Seems to be a problem with how data path is read so you can open the addon config, choose "edit in yaml", put quotes around the data path 
-> change data_path: /config/zigbee2mqtt to "data_path: /config/zigbee2mqtt"
Save and restart addon

Next HA restart the problem is back...

rwjack, 2022-09-05:
https://github.com/Koenkk/zigbee2mqtt/issues/12801

Stooovie, 2022-09-06:
Same. How does this happen? There was no Z2M update nor HA update.

closed 2022-10-15 by github-actions[bot]