Koenkk/zigbee2mqtt #13835: _TZ3000_xkap8wtb paired as TS0001

pabloaugusto, 2022-09-03:
### What happened?

My Tuya /Aubess switch have power monitoring, and TS0001 don't have

### What did you expect to happen?

Z2M must expose power entities from switch

### How to reproduce it (minimal and precise)

Just pair with z2m and will be paired with wrong model, without power monitoring

### Zigbee2MQTT version

1.27.2-dev commit: 178f72c

### Adapter firmware version

20220507

### Adapter

Tuya  _TZ3000_xkap8wtb

### Debug log

This is the database.db row from my switch 

{"id":19,"type":"Router","ieeeAddr":"0xa4c138296d0d40d5","nwkAddr":25199,"manufId":4417,"manufName":"_TZ3000_xkap8wtb","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":126,"rmsCurrent":0,"activePower":0}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":2754918,"53251":125,"53252":11118,"53253":1306,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b8081e","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1662228938281,"defaultSendRequestWhen":"immediate"}

And This is the database.db row from other of my tuya switch with power monitor that works ok in the same enviroment

{"id":14,"type":"Router","ieeeAddr":"0xa4c138839e183149","nwkAddr":26140,"manufId":4417,"manufName":"_TZ3000_cjrngdr3","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�NG*\u0017�NG*\u0013\u0000OG*\u0012\u0012\u0000\u0000\u0000\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3000_cjrngdr3","powerSource":1,"zclVersion":3,"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":128,"rmsCurrent":295,"activePower":21}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,2603],"status":0,"unitOfMeasure":0,"summaFormatting":0,"meteringDeviceType":0}},"genOnOff":{"attributes":{"32768":1,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"powerOnBehavior":1,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b8081e","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b8081e","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b8081e","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":"5"},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":"20"},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":"2"},{"cluster":1794,"attrId":0,"minRepIntval":"5","maxRepIntval":3600,"repChange":"0"},{"cluster":1794,"attrId":512,"minRepIntval":"5","maxRepIntval":3600,"repChange":0},{"cluster":1794,"attrId":768,"minRepIntval":"5","maxRepIntval":3600,"repChange":0},{"cluster":1794,"attrId":771,"minRepIntval":"5","maxRepIntval":3600,"repChange":0},{"cluster":1794,"attrId":774,"minRepIntval":"5","maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":77,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1798312590},"lastSeen":1662229137941,"defaultSendRequestWhen":"immediate"}

pabloaugusto, 2022-09-03:
The photo of the switch https://i.im.ge/2022/09/04/ObBbSC.c0d0dbfc-cd32-43ee-9ebf-5bf9e08f2d35.jpg

tomekso, 2022-09-05:
I have exactly the same problem with this switch

k7franklin, 2022-09-07:
I also have exactly the same problem. Aubess switch.

Koenkk, 2022-09-08:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition =     {
    fingerprint: [{modelID: 'TS0001', manufacturerName: '_TZ3000_xkap8wtb'}],
    model: 'TS0001_power',
    description: 'Relay with power monitoring',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE)],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

k7franklin, 2022-09-08:
Thank you very much. It Works! 

(https://drive.google.com/file/d/1FMNKT7L9YaO-JDoihze17e9XlJV8t-hX/view?usp=sharing)

Koenkk, 2022-09-09:
- Can you provide me a link of this device? (aliexpress?)
- is there any model number written on it?

k7franklin, 2022-09-09:
https://pt.aliexpress.com/item/1005004605451027.html?spm=a2g0o.order_list.0.0.216dcaa49LnDyT&gatewayAdapt=glo2bra

This is the exact model. Choose the "with power metering" version.

Absolutely no model number or other identification, only the original image shows.

pabloaugusto, 2022-09-10:
Sorry for delay. Yes this external converter fix the problem

pabloaugusto, 2022-09-10:
> * Can you provide me a link of this device? (aliexpress?)
> * is there any model number written on it?

This switch has no print info about model: https://i.im.ge/2022/09/04/ObBbSC.c0d0dbfc-cd32-43ee-9ebf-5bf9e08f2d35.jpg

Buyed from https://pt.aliexpress.com/item/1005002278683118.html?gatewayAdapt=glo2bra

Koenkk, 2022-09-10:
Added it!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-10 by Koenkk

Gerd2809, 2022-09-26:
Hej,
I use two of this devices, they have turns off by itself like the TS011F.

So I start a test.
I activate at the TS0001_power the ota update in tuya.js by insert ota.zigbeeOTA,
then I update the devices and after update they will resport the modellID TS011F.
After that I edit the tuya.js a second time and update the fingerprint of the TS011F din smart relay.

`fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_8bxrzyxz'}, {modelID: 'TS011F', manufacturerName: '_TZ3000_ky0fq4ho'},
        **{modelID: 'TS011F', manufacturerName: '_TZ3000_xkap8wtb'}],**
        model: 'TS011F_din_smart_relay',
        description: 'Din smart relay (with power monitoring)',
        vendor: 'TuYa',
        fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report, fz.tuya_switch_power_outage_memory,
            fz.tuya_relay_din_led_indicator],
        toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory, tz.tuya_relay_din_led_indicator],
        whiteLabel: [{vendor: 'MatSee Plus', model: 'ATMS1602Z'}],
        ota: ota.zigbeeOTA,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
            await reporting.rmsVoltage(endpoint, {change: 5});
            await reporting.rmsCurrent(endpoint, {change: 50});
            await reporting.activePower(endpoint, {change: 10});
            await reporting.currentSummDelivered(endpoint);
            endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
            endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            device.save();
        },
        exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE),
            e.energy(), exposes.enum('power_outage_memory', ea.ALL, ['on', 'off', 'restore'])
                .withDescription('Recover state after power outage'),
            exposes.enum('indicator_mode', ea.STATE_SET, ['off', 'on_off', 'off_on'])
                .withDescription('Relay LED indicator mode')],
    },`

Now the device work fine.

Bets regards 
Gerd