Koenkk/zigbee2mqtt #13843: LELLKI WP30-EU : each switch control change state of all switches

baegopooh, 2022-09-04:
### What happened?

I started using LELLKI WP30-EU with home assistant via zigbee2mqtt(standalone linux installation).

When I tried to switch seperate switch[Switch (l1~l3 endpoint)] every swiches change state.

I tried in HA, in zigbee2mqtt dashboard, with mqtt msg(publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"state_l3": "ON"}, {"state_l3": "OFF"} or {"state_l3": "TOGGLE"}). and it works all the same. 

### What did you expect to happen?

switch each switch individually.

### How to reproduce it (minimal and precise)

1. setup device
2. With mqtt, publish following msg(state l1~l3, ON to OFF)

zigbee2mqtt/FRIENDLY_NAME/set with payload {"state_l3": "ON"}

2-1. in zigbee2mqtt, try to toggle each switch

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20210320

### Adapter

custom zigbee dongle with cc2652p

### Debug log

root@zigbee2mqtt:/opt/zigbee2mqtt/data/log/2022-09-04.04-21-04# cat log.txt | grep Lelki
info  2022-09-04 04:23:57: MQTT publish: topic 'zigbee2mqtt/Lelki_powerstrip1/availability', payload 'online'
info  2022-09-04 04:23:57: MQTT publish: topic 'zigbee2mqtt/Lelki_powerstrip1', payload '{"current":0,"energy":0,"linkquality":60,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":227}'
info  2022-09-04 04:23:57: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/rename', payload '{"data":{"from":"0xa4c138ad62f4745c","homeassistant_rename":true,"to":"Lelki_powerstrip1"},"status":"ok","transaction":"6syz2-3"}'
info  2022-09-04 04:23:59: MQTT publish: topic 'homeassistant/switch/0xa4c138ad62f4745c/switch_l1/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Lelki_powerstrip1/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Lelki_powerstrip1/l1/set","device":{"identifiers":["zigbee2mqtt_0xa4c138ad62f4745c"],"manufacturer":"LELLKI","model":"Power cord 4 sockets EU (with power monitoring) (WP30-EU)","name":"Lelki_powerstrip1"},"json_attributes_topic":"zigbee2mqtt/Lelki_powerstrip1","name":"Lelki_powerstrip1_l1","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/Lelki_powerstrip1","unique_id":"0xa4c138ad62f4745c_switch_l1_zigbee2mqtt","value_template":"{{ value_json.state_l1 }}"}'
info  2022-09-04 04:23:59: MQTT publish: topic 'homeassistant/switch/0xa4c138ad62f4745c/switch_l2/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Lelki_powerstrip1/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Lelki_powerstrip1/l2/set","device":{"identifiers":["zigbee2mqtt_0xa4c138ad62f4745c"],"manufacturer":"LELLKI","model":"Power cord 4 sockets EU (with power monitoring) (WP30-EU)","name":"Lelki_powerstrip1"},"json_attributes_topic":"zigbee2mqtt/Lelki_powerstrip1","name":"Lelki_powerstrip1_l2","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/Lelki_powerstrip1","unique_id":"0xa4c138ad62f4745c_switch_l2_zigbee2mqtt","value_template":"{{ value_json.state_l2 }}"}'
info  2022-09-04 04:23:59: MQTT publish: topic 'homeassistant/switch/0xa4c138ad62f4745c/switch_l3/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Lelki_powerstrip1/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Lelki_powerstrip1/l3/set","device":{"identifiers":["zigbee2mqtt_0xa4c138ad62f4745c"],"manufacturer":"LELLKI","model":"Power cord 4 sockets EU (with power monitoring) (WP30-EU)","name":"Lelki_powerstrip1"},"json_attributes_topic":"zigbee2mqtt/Lelki_powerstrip1","name":"Lelki_powerstrip1_l3","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/Lelki_powerstrip1","unique_id":"0xa4c138ad62f4745c_switch_l3_zigbee2mqtt","value_template":"{{ value_json.state_l3 }}"}'
info  2022-09-04 04:23:59: MQTT publish: topic 'homeassistant/sensor/0xa4c138ad62f4745c/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Lelki_powerstrip1/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138ad62f4745c"],"manufacturer":"LELLKI","model":"Power cord 4 sockets EU (with power monitoring) (WP30-EU)","name":"Lelki_powerstrip1"},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Lelki_powerstrip1","name":"Lelki_powerstrip1_power","state_class":"measurement","state_topic":"zigbee2mqtt/Lelki_powerstrip1","unique_id":"0xa4c138ad62f4745c_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
info  2022-09-04 04:23:59: MQTT publish: topic 'homeassistant/sensor/0xa4c138ad62f4745c/current/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Lelki_powerstrip1/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138ad62f4745c"],"manufacturer":"LELLKI","model":"Power cord 4 sockets EU (with power monitoring) (WP30-EU)","name":"Lelki_powerstrip1"},"device_class":"current","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Lelki_powerstrip1","name":"Lelki_powerstrip1_current","state_class":"measurement","state_topic":"zigbee2mqtt/Lelki_powerstrip1","unique_id":"0xa4c138ad62f4745c_current_zigbee2mqtt","unit_of_measurement":"A","value_template":"{{ value_json.current }}"}'
info  2022-09-04 04:23:59: MQTT publish: topic 'homeassistant/sensor/0xa4c138ad62f4745c/voltage/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Lelki_powerstrip1/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138ad62f4745c"],"manufacturer":"LELLKI","model":"Power cord 4 sockets EU (with power monitoring) (WP30-EU)","name":"Lelki_powerstrip1"},"device_class":"voltage","enabled_by_default":false,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/Lelki_powerstrip1","name":"Lelki_powerstrip1_voltage","state_class":"measurement","state_topic":"zigbee2mqtt/Lelki_powerstrip1","unique_id":"0xa4c138ad62f4745c_voltage_zigbee2mqtt","unit_of_measurement":"V","value_template":"{{ value_json.voltage }}"}'
info  2022-09-04 04:23:59: MQTT publish: topic 'homeassistant/sensor/0xa4c138ad62f4745c/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Lelki_powerstrip1/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138ad62f4745c"],"manufacturer":"LELLKI","model":"Power cord 4 sockets EU (with power monitoring) (WP30-EU)","name":"Lelki_powerstrip1"},"device_class":"energy","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/Lelki_powerstrip1","name":"Lelki_powerstrip1_energy","state_class":"total_increasing","state_topic":"zigbee2mqtt/Lelki_powerstrip1","unique_id":"0xa4c138ad62f4745c_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
info  2022-09-04 04:23:59: MQTT publish: topic 'homeassistant/sensor/0xa4c138ad62f4745c/power_outage_memory/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Lelki_powerstrip1/availability"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c138ad62f4745c"],"manufacturer":"LELLKI","model":"Power cord 4 sockets EU (with power monitoring) (WP30-EU)","name":"Lelki_powerstrip1"},"enabled_by_default":false,"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/Lelki_powerstrip1","name":"Lelki_powerstrip1_power_outage_memory","state_topic":"zigbee2mqtt/Lelki_powerstrip1","unique_id":"0xa4c138ad62f4745c_power_outage_memory_zigbee2mqtt","value_template":"{{ value_json.power_outage_memory }}"}'
info  2022-09-04 04:23:59: MQTT publish: topic 'homeassistant/select/0xa4c138ad62f4745c/power_outage_memory/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/Lelki_powerstrip1/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/Lelki_powerstrip1/set/power_outage_memory","device":{"identifiers":["zigbee2mqtt_0xa4c138ad62f4745c"],"manufacturer":"LELLKI","model":"Power cord 4 sockets EU (with power monitoring) (WP30-EU)","name":"Lelki_powerstrip1"},"entity_category":"config","icon":"mdi:power-settings","json_attributes_topic":"zigbee2mqtt/Lelki_powerstrip1","name":"Lelki_powerstrip1_power_outage_memory","options":["on","off","restore"],"sta [... cut ...] te_l3":"OFF","voltage":226}'
debug 2022-09-04 04:44:39: Received Zigbee message from 'Lelki_powerstrip1', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":226}' from endpoint 1 with groupID 0
info  2022-09-04 04:44:39: MQTT publish: topic 'zigbee2mqtt/Lelki_powerstrip1', payload '{"current":0,"energy":0,"linkquality":60,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"ON","state_l3":"OFF","voltage":226}'

Koenkk, 2022-09-06:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-06 by Koenkk

MaxMSchneider, 2024-01-05:
This seems still to be an issue. Switched to the dev branch and still have this behavior.