Koenkk/zigbee2mqtt #13844: [Feature request]: Smart locks - expose the list of pincodes in exposes.pincode

adamkoch, 2022-09-04:
### Is your feature request related to a problem? Please describe

I would like smart lock devices to be able to fetch/retrieve existing pin codes set on the device. Right now, it seems this is a read-only attribute. I have a [Schlage lock](https://www.zigbee2mqtt.io/devices/BE468.html) but I think most (if not all?) other locks are the same.

The reason for this, is so that something like Home Assistant can read the the lock codes, present then in a UI and also manage the lock codes (eg. setting a lock code temporarily, setting lock codes on a schedule etc).

In particular, [Keymaster](https://github.com/FutureTense/keymaster/) can do this for zwave locks. Open discussion on zigbee support for Keymaster [here](https://github.com/FutureTense/keymaster/issues/163).

I previously switched from Hubitat which could read the codes from same lock over zigbee so it doesn't seem to be a limitation of the lock and/or zigbee.

### Describe the solution you'd like

Right now zigbee2mqtt can **set** the lock codes, the proposal is a method to **get** the lock does as well. This may need to be behind some sort of setting flag as some may not want this to be exposed for security reasons.

I tried looking at the code and I think this involves updating [`zigbee-herdsman-converters`](https://github.com/Koenkk/zigbee-herdsman-converters/)

I think maybe here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/lib/exposes.js#L639

I filed an [issue](https://github.com/Koenkk/zigbee-herdsman-converters/issues/4566) in that repo as well as I wasn't sure where to file.

### Describe alternatives you've considered

It could be possible to implement some of this features without visibility into the lock codes themselves. Blindly adding/removing code positions. However, this isn't ideal as there is no way to be sure the lock codes were changed/updated successfully.

### Additional context

No other context.

mghersi65, 2022-10-04:
Hi Adam, looking for the same solution for a client any news from your side?

adamkoch, 2022-10-04:
No updates unfortunately.

@Koenkk - any thoughts on if this is possible or what it would take to add? Or any pointers on what code to look at for contributions?

Koenkk, 2022-10-04:
@adamkoch what happens if you publish to `zigbee2mqtt/DEVICE_FRIENDLY_NAME/get` payload `{"pin_code":""}`

adamkoch, 2022-10-05:
Oh bingo! That does seem to work. I see the codes returned as part of `users`:
```
{
    "action": null,
    "action_source_name": null,
    "action_user": null,
    "auto_relock_time": 0,
    "battery": 96,
    "last_seen": "2022-10-05T14:46:28+11:00",
    "linkquality": 255,
    "lock_state": "locked",
    "state": "LOCK",
    "users":
    {
        "0":
        {
            "pin_code": "0000",
            "status": "enabled"
        },
        "1":
        {
            "pin_code": "0000",
            "status": "enabled"
        },
        "10":
        {
            "status": "available"
        },
        "11":
        {
            "status": "available"
        },
        "12":
        {
            "status": "available"
        },
        ...
    }
}
```

adamkoch, 2022-10-05:
Will close this as no updates needed on zigbee2mqtt side. Thank you @Koenkk!

closed 2022-10-05 by adamkoch

joshuaspence, 2023-01-30:
How can I expose `users` in Home Assistant?