Koenkk/zigbee2mqtt #13847: Bad state on TS130F after updating to 1.27.2

Smashfer, 2022-09-04:
### What happened?

Since i have made the update to 1.27.2 (previously i was on 1.27), my shutter switch TS130F show me the wrong state. Now, when my shutter is open, the state is "closed" and when my shutter is closed, the state is "open".
If i want a correct state i have to put motor_reversal on "On" but it's make no sense since my wired connexion are ok.
My counters and cards in home assistant are thus wrong...
An example with a shutter open :
{
    "backlight_mode": "LOW",
    "calibration": "OFF",
    "calibration_time": 13.6,
    "linkquality": 15,
    "motor_reversal": "OFF",
    "moving": "STOP",
    "position": 100,
    "state": "CLOSE"
}

### What did you expect to happen?

I would like to have the correct state as before.

### How to reproduce it (minimal and precise)

Just make an update to 1.27 to 1.27.2

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20190608

### Adapter

zStack12 (generic CC2530 usb sticker)

Smashfer, 2022-09-04:
I made a mistake, sorry.
I have dowgrade to 1.27.0 and the state is the same...
The difference between 1.27 and 1.27.2 is that the specific setting "invert_cover" don't change the state on Home assistant with the last version.

convicte, 2022-09-04:
There are at minimum 5 already open issues for the problem -  e.g. #13805 

Opening another dilutes the efforts and comments on these.

All reference the device type so are a search away.

Cheers!

Koenkk, 2022-09-06:
Fixed by https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-06 by Koenkk