Koenkk/zigbee2mqtt #13856: Unable to update ZNCZ02LM Xiaomi CN Socket

sorryusernameisalreadytaken, 2022-09-04:
### What happened?

I try to update the ZNCZ02LM socket but it fails always.

Current fw-date is: 02-28-2017

### Zigbee2MQTT version

1.27.2-dev commit: 24d83b0 (docker)

### Adapter firmware version

20220219

### Adapter

Slaesh's CC2652RB Stick

### Debug log

```
info  2022-09-04 16:34:28: MQTT publish: topic 'zigbee2mqtt/Socket_Kue_AquaMix', payload '{"consumption":4.66,"device_temperature":35,"energy":4.66,"link>
debug 2022-09-04 16:34:28: Received Zigbee message from 'Socket_Kue_AquaMix', type 'readResponse', cluster 'genBasic', data '{"dateCode":"02-28-2017"}' f>
debug 2022-09-04 16:34:28: lumi.plug: unknown key dateCode with value 02-28-2017
debug 2022-09-04 16:34:28: lumi.plug: unknown key swBuildId with value undefined
debug 2022-09-04 16:34:28: lumi.plug: Processed data into payload {}
debug 2022-09-04 16:34:28: Updating to latest '0x00158d00019c9e3d' (lumi.plug)
debug 2022-09-04 16:34:28: Using endpoint '1'
debug 2022-09-04 16:34:28: Received Zigbee message from 'Socket_Kue_AquaMix', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:28: Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":257,"fileVersion":22}'
debug 2022-09-04 16:34:28: Received Zigbee message from 'Socket_Kue_AquaMix', type 'attributeReport', cluster 'genOnOff', data '{"61440":58121984,"onOff">
info  2022-09-04 16:34:28: MQTT publish: topic 'zigbee2mqtt/Socket_Kue_AquaMix', payload '{"consumption":4.66,"device_temperature":35,"energy":4.66,"link>
debug 2022-09-04 16:34:28: ZigbeeOTA: downloaded main index
debug 2022-09-04 16:34:28: ZigbeeOTA: Loading override index https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json
debug 2022-09-04 16:34:28: getNewImage for '0x00158d00019c9e3d', meta {"fileVersion":92,"fileSize":185918,"url":"https://github.com/Koenkk/zigbee-OTA/raw>
debug 2022-09-04 16:34:28: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/lumi.plug_20211224_v9>
debug 2022-09-04 16:34:29: OTA update checksum validation succeeded for '0x00158d00019c9e3d'
debug 2022-09-04 16:34:29: getNewImage for '0x00158d00019c9e3d', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHe>
debug 2022-09-04 16:34:29: Got new image for '0x00158d00019c9e3d'
debug 2022-09-04 16:34:29: Starting upgrade
debug 2022-09-04 16:34:29: Received Zigbee message from 'Socket_Kue_AquaMix', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:29: OTA update at 0%, remaining Infinity seconds
info  2022-09-04 16:34:29: Update of 'Socket_Kue_AquaMix' at 0.00%
info  2022-09-04 16:34:29: MQTT publish: topic 'zigbee2mqtt/Socket_Kue_AquaMix', payload '{"consumption":4.66,"device_temperature":35,"energy":4.66,"link>
debug 2022-09-04 16:34:32: Received Zigbee message from 'Socket_Kue_AquaMix', type 'attributeReport', cluster 'genOnOff', data '{"61440":58121984,"onOff">
info  2022-09-04 16:34:32: MQTT publish: topic 'zigbee2mqtt/Socket_Kue_AquaMix', payload '{"consumption":4.66,"device_temperature":35,"energy":4.66,"link>
debug 2022-09-04 16:34:34: Received Zigbee message from 'Socket_Kue_AquaMix', type 'attributeReport', cluster 'genOnOff', data '{"61440":58121984,"onOff">
info  2022-09-04 16:34:34: MQTT publish: topic 'zigbee2mqtt/Socket_Kue_AquaMix', payload '{"consumption":4.66,"device_temperature":35,"energy":4.66,"link>
debug 2022-09-04 16:34:34: Received Zigbee message from '0x001788010448a00a', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:34: Device '0x001788010448a00a' requested OTA
debug 2022-09-04 16:34:34: Received Zigbee message from '0x00178801036c0ce6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:34: Device '0x00178801036c0ce6' requested OTA
debug 2022-09-04 16:34:34: Received Zigbee message from '0x001788010206a264', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:34: Device '0x001788010206a264' requested OTA
debug 2022-09-04 16:34:42: Received Zigbee message from 'Socket_Kue_AquaMix', type 'attributeReport', cluster 'genOnOff', data '{"61440":58121984,"onOff">
info  2022-09-04 16:34:42: MQTT publish: topic 'zigbee2mqtt/Socket_Kue_AquaMix', payload '{"consumption":4.66,"device_temperature":35,"energy":4.66,"link>
debug 2022-09-04 16:34:44: Received Zigbee message from '0x001788010448a00a', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:44: Device '0x001788010448a00a' requested OTA
debug 2022-09-04 16:34:44: Received Zigbee message from '0x00178801036c0ce6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:44: Device '0x00178801036c0ce6' requested OTA
debug 2022-09-04 16:34:44: Received Zigbee message from '0x001788010206a264', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:44: Device '0x001788010206a264' requested OTA
debug 2022-09-04 16:34:47: Received Zigbee message from 'Socket_Sz_PC', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,44>
info  2022-09-04 16:34:47: MQTT publish: topic 'zigbee2mqtt/Socket_Sz_PC', payload '{"current":0.27,"energy":44.18,"indicator_mode":"off/on","linkquality>
debug 2022-09-04 16:34:47: Received Zigbee message from 'Socket_Wz_Subwoofer', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered>
info  2022-09-04 16:34:47: MQTT publish: topic 'zigbee2mqtt/Socket_Wz_Subwoofer', payload '{"current":0,"energy":97.6,"indicator_mode":"off/on","linkqual>
debug 2022-09-04 16:34:54: Received Zigbee message from '0x001788010448a00a', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:54: Device '0x001788010448a00a' requested OTA
debug 2022-09-04 16:34:54: Received Zigbee message from '0x00178801036c0ce6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl>
debug 2022-09-04 16:34:54: Device '0x00178801036c0ce6' requested OTA
[... 159 lines cut ...]
info  2022-09-04 16:52:28: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Socket_Kue_AquaMix"},"error":"Update of 'Socket_Kue_AquaMix' failed (Update failed with reason: 'aborted by device')","status":"error","transaction":"c64ej-11"}'
error 2022-09-04 16:52:28: Update of 'Socket_Kue_AquaMix' failed (Update failed with reason: 'aborted by device')
debug 2022-09-04 16:52:28: Error: Update failed with reason: 'aborted by device'
    at /app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:410:24
debug 2022-09-04 17:06:55: Received Zigbee message from '0x0017880103a54f45', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
debug 2022-09-04 17:06:55: Device '0x0017880103a54f45' requested OTA
debug 2022-09-04 17:06:55: ZigbeeOTA: downloaded main index
debug 2022-09-04 17:06:55: ZigbeeOTA: Loading override index https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/index.json
debug 2022-09-04 17:06:55: Is new image available for '0x0017880103a54f45', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":265,"fileVersion":1107324829}', latest meta '{"fileVersion":1107324829,"fileSize":240760,"url":"http://fds.dc1.philips.com/firmware/ZGB_100B_0109/1107324829/Switch-ATmega_6.1.1.28573_0012.sbl-ota","sha512":"d20057d6c4e61d4ccdd3947b8a8c9bf9e126a26c080b9ce3e6839a0615b912c6f93246a60c55a44f2c8b8ec49f5041f46ac38b8a3c0a1c6ab44f062cc62089cf"}'
debug 2022-09-04 17:06:55: Update available for '0x0017880103a54f45': NO
debug 2022-09-04 17:07:05: Received Zigbee message from '0x0017880103a54f45', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
debug 2022-09-04 17:07:05: Device '0x0017880103a54f45' requested OTA
debug 2022-09-04 17:07:15: Received Zigbee message from '0x0017880103a54f45', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1107324829,"imageType":265,"manufacturerCode":4107}' from endpoint 2 with groupID 0
debug 2022-09-04 17:07:15: Device '0x0017880103a54f45' requested OTA
[... cut ...]
debug 2022-09-04 16:52:28: Update of 'Socket_Kue_AquaMix' failed (Error: Update failed with reason: 'aborted by device')
info  2022-09-04 16:52:28: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Socket_Kue_AquaMix"},"erro>
error 2022-09-04 16:52:28: Update of 'Socket_Kue_AquaMix' failed (Update failed with reason: 'aborted by device')
debug 2022-09-04 16:52:28: Error: Update failed with reason: 'aborted by device'
    at /app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:410:24
```

closed 2022-10-12 by github-actions[bot]

Happyboy0980, 2023-01-24:
same isseu with my 2016 firmware...

hongliangl, 2023-01-30:
> same isseu with my 2016 firmware...

@Happyboy0980 https://github.com/Koenkk/zigbee2mqtt/issues/6906#issuecomment-1375780694

GeorgeCastanza, 2023-08-27:
Same.  Device gets to 100% transferring firmware, but then does not update.

realwax, 2023-09-11:
This issue exist since years. Are there any news how to get the update working? A suggested restart of the device does not resolve the update issue. 

<html>
<body>

zigbee.0 | 2023-09-11 10:59:08.859 | error | Update of '0x00158d00028b3478' failed (Update failed with reason: 'aborted by device')
-- | -- | -- | --
zigbee.0 | 2023-09-11 10:58:50.686 | info | Update of '0x00158d00028b3478' at 98.36%, +- 0 minutes remaining
zigbee.0 | 2023-09-11 10:58:20.488 | info | Update of '0x00158d00028b3478' at 95.55%, +- 1 minutes remaining
zigbee.0 | 2023-09-11 10:57:50.302 | info | Update of '0x00158d00028b3478' at 92.74%, +- 1 minutes remaining
zigbee.0 | 2023-09-11 10:57:20.109 | info | Update of '0x00158d00028b3478' at 89.92%, +- 2 minutes remaining
zigbee.0 | 2023-09-11 10:56:49.899 | info | Update of '0x00158d00028b3478' at 87.11%, +- 2 minutes remaining
zigbee.0 | 2023-09-11 10:56:19.680 | info | Update of '0x00158d00028b3478' at 84.29%, +- 3 minutes remaining
zigbee.0 | 2023-09-11 10:55:49.457 | info | Update of '0x00158d00028b3478' at 81.48%, +- 3 minutes remaining
zigbee.0 | 2023-09-11 10:55:19.253 | info | Update of '0x00158d00028b3478' at 78.67%, +- 4 minutes remaining
zigbee.0 | 2023-09-11 10:54:49.042 | info | Update of '0x00158d00028b3478' at 75.85%, +- 4 minutes remaining
zigbee.0 | 2023-09-11 10:54:18.800 | info | Update of '0x00158d00028b3478' at 73.04%, +- 5 minutes remaining
zigbee.0 | 2023-09-11 10:53:48.601 | info | Update of '0x00158d00028b3478' at 70.22%, +- 5 minutes remaining
zigbee.0 | 2023-09-11 10:53:18.413 | info | Update of '0x00158d00028b3478' at 67.41%, +- 6 minutes remaining
zigbee.0 | 2023-09-11 10:52:48.173 | info | Update of '0x00158d00028b3478' at 64.6%, +- 6 minutes remaining
zigbee.0 | 2023-09-11 10:52:17.947 | info | Update of '0x00158d00028b3478' at 61.78%, +- 7 minutes remaining
zigbee.0 | 2023-09-11 10:51:47.754 | info | Update of '0x00158d00028b3478' at 58.97%, +- 7 minutes remaining
zigbee.0 | 2023-09-11 10:51:17.518 | info | Update of '0x00158d00028b3478' at 56.15%, +- 8 minutes remaining
zigbee.0 | 2023-09-11 10:50:47.284 | info | Update of '0x00158d00028b3478' at 53.34%, +- 8 minutes remaining
zigbee.0 | 2023-09-11 10:50:17.078 | info | Update of '0x00158d00028b3478' at 50.52%, +- 9 minutes remaining
zigbee.0 | 2023-09-11 10:49:46.832 | info | Update of '0x00158d00028b3478' at 47.71%, +- 9 minutes remaining
zigbee.0 | 2023-09-11 10:49:16.628 | info | Update of '0x00158d00028b3478' at 44.9%, +- 10 minutes remaining
zigbee.0 | 2023-09-11 10:48:46.421 | info | Update of '0x00158d00028b3478' at 42.08%, +- 10 minutes remaining
zigbee.0 | 2023-09-11 10:48:16.200 | info | Update of '0x00158d00028b3478' at 39.27%, +- 11 minutes remaining
zigbee.0 | 2023-09-11 10:47:46.196 | info | Update of '0x00158d00028b3478' at 36.48%, +- 11 minutes remaining
zigbee.0 | 2023-09-11 10:47:15.988 | info | Update of '0x00158d00028b3478' at 33.67%, +- 12 minutes remaining
zigbee.0 | 2023-09-11 10:46:45.737 | info | Update of '0x00158d00028b3478' at 30.85%, +- 12 minutes remaining
zigbee.0 | 2023-09-11 10:46:15.499 | info | Update of '0x00158d00028b3478' at 28.04%, +- 13 minutes remaining
zigbee.0 | 2023-09-11 10:45:45.221 | info | Update of '0x00158d00028b3478' at 25.22%, +- 13 minutes remaining
zigbee.0 | 2023-09-11 10:45:14.973 | info | Update of '0x00158d00028b3478' at 22.41%, +- 14 minutes remaining
zigbee.0 | 2023-09-11 10:44:44.710 | info | Update of '0x00158d00028b3478' at 19.59%, +- 14 minutes remaining
zigbee.0 | 2023-09-11 10:44:14.442 | info | Update of '0x00158d00028b3478' at 16.78%, +- 15 minutes remaining
zigbee.0 | 2023-09-11 10:43:44.178 | info | Update of '0x00158d00028b3478' at 13.97%, +- 15 minutes remaining
zigbee.0 | 2023-09-11 10:43:13.904 | info | Update of '0x00158d00028b3478' at 11.15%, +- 16 minutes remaining
zigbee.0 | 2023-09-11 10:42:43.676 | info | Update of '0x00158d00028b3478' at 8.34%, +- 17 minutes remaining
zigbee.0 | 2023-09-11 10:42:13.669 | info | Update of '0x00158d00028b3478' at 5.55%, +- 17 minutes remaining
zigbee.0 | 2023-09-11 10:41:43.656 | info | Update of '0x00158d00028b3478' at 2.76%, +- 18 minutes remaining
zigbee.0 | 2023-09-11 10:41:13.568 | info | Update of '0x00158d00028b3478' at 0%
zigbee.0 | 2023-09-11 10:41:12.394 | info | Start firmware update for 0x00158d00028b3478
zigbee.0 | 2023-09-11 10:41:10.840 | info | current Firmware for 0x00158d00028b3478 is 22 new is 92

</body>
</html>

isa220, 2024-05-09:
you can find solution by next link (Xiaomi Gateway 3 needed):

https://github.com/kvazis/training/tree/master/lessons/articles/gateway%203%20xiaomi%20update