Koenkk/zigbee2mqtt #13858: [New device support]:

leonidnf, 2022-09-04:
### Link

https://www.aliexpress.com/item/1005003624122067.html?spm=a2g0o.order_list.0.0.65d91802W69lhE

### Database entry

MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138bda9b71b68","ieee_address":"0xa4c138bda9b71b68","status":"successful","supported":false},"type":"device_interview"}' Warning 2022-09-04 21:38:00Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_b28wrpvx'

### Comments

Hi Zigbee2mqtt support team,
I'm excited about your work!
Can you please add the following socket from BSEED?

Warning 2022-09-04 21:37:35Device '0xa4c138bda9b71b68' left the network
Info 2022-09-04 21:37:35MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138bda9b71b68","ieee_address":"0xa4c138bda9b71b68"},"type":"device_leave"}'
Info 2022-09-04 21:37:55Device '0xa4c138bda9b71b68' joined
Info 2022-09-04 21:37:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138bda9b71b68","ieee_address":"0xa4c138bda9b71b68"},"type":"device_joined"}'
Info 2022-09-04 21:37:55Starting interview of '0xa4c138bda9b71b68'
Info 2022-09-04 21:37:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138bda9b71b68","ieee_address":"0xa4c138bda9b71b68","status":"started"},"type":"device_interview"}'
Info 2022-09-04 21:37:55MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138bda9b71b68","ieee_address":"0xa4c138bda9b71b68"},"type":"device_announce"}'
Info 2022-09-04 21:37:56Successfully interviewed '0xa4c138bda9b71b68', device has successfully been paired
Warning 2022-09-04 21:37:56Device '0xa4c138bda9b71b68' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_b28wrpvx' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2022-09-04 21:37:56MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c138bda9b71b68","ieee_address":"0xa4c138bda9b71b68","status":"successful","supported":false},"type":"device_interview"}'
Warning 2022-09-04 21:38:00Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_b28wrpvx'
Warning 2022-09-04 21:38:00Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning 2022-09-04 21:38:00Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_b28wrpvx'

Koenkk, 2022-09-06:
`_TZ3000_b28wrpvx` is supported in the dev branch and will be in the next release on 1 October. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-06 by Koenkk

leonidnf, 2022-09-15:
@Koenkk Thank you very much!