Koenkk/zigbee2mqtt #13861: [Feature request]: Add "read attributes command" (i.e. refresh) via API

schasj, 2022-09-05:
### Is your feature request related to a problem? Please describe

The "problem" is that KMPCIL_RES005 devices (and perhaps others) don't always track illuminance to zero, sometimes "hanging" at as much as 1000. I have motion lighting automations that only turn lights on if the room is dark in node-red and I don't have a way to force a refresh right after the lights are turned off thus blocking the lights from coming back on when they should.

### Describe the solution you'd like

I want to send a refresh command to a specific device via Andrei's node-red-contrib-zigbee2mqtt Out node (obviously requires him to implement). Thus I'll be able to rapidly refresh for only a few seconds when the light is turned off.

### Describe alternatives you've considered

The only (unacceptable) alternative I can think of is to set both min and max reporting time to 1s which would produce a LOT of useless mesh traffic.

### Additional context

5 minutes after the lights turned off

[image]

sjorge, 2022-09-05:
There is already support for this, you can just do a get for the attribute and if the device is willing, it will be read.

```
mosquitto_pub -t zigbee2mqtt/light/office/desk/bulb/get -m '{"state": ""}'
```

```
debug 2022-09-05 13:19:51: Received MQTT message on 'zigbee2mqtt/light/office/desk/bulb/get' with data '{"state": ""}'
debug 2022-09-05 13:19:51: Publishing get 'get' 'state' to 'light/office/desk/bulb'
debug 2022-09-05 13:19:51: Received Zigbee message from 'light/office/desk/bulb', type 'readResponse', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-09-05 13:19:52: MQTT publish: topic 'zigbee2mqtt/light/office/desk/bulb', payload '{"brightness":254,"color":{"hue":40,"saturation":50,"x":0.3813,"y":0.3773},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":251,"linkquality":127,"power_on_behavior":"off","state":"OFF","update":{"state":"idle"}}'
```

Do note for battery powered device a get might fail on a timeout.

sjorge, 2022-09-05:
It's probably just missing a convertor for it then, I see it doesn't have a tz.illuminance one which is probably why it's not working.

Let me test with one of mine, different sensor but also missing the convertor.

sjorge, 2022-09-05:
OK easy, just missing a convertor

```
debug 2022-09-05 19:26:11: Publishing get 'get' 'illuminance' to 'lux/office'
debug 2022-09-05 19:26:16: Received Zigbee message from 'lux/office', type 'readResponse', cluster 'msIlluminanceMeasurement', data '{"measuredValue":27068}' from endpoint 1 with groupID 0
info  2022-09-05 19:26:16: MQTT publish: topic 'zigbee2mqtt/lux/office', payload '{"battery":100,"illuminance":27068,"illuminance_lux":509,"linkquality":51,"voltage":3200}'
```

Let me do a PR

schasj, 2022-09-05:
@sjorge, that's great! And thanks for doing the PR.

Just one thing though. I know that many of us use Andrei's node-red-contrib-zigbee2mqtt and it would be nice to be able to send "refresh" with his Out node. I don't think it needs to be for a specific measurement, just have all of them updated. Personally, I'd be happy with that but I have no idea if others would want to specify a measurement.

sjorge, 2022-09-05:
I'm not familier with that node, but I assume if generic get support gets added it will work where there are convertors available.

schasj, 2022-09-05:
Right now there is no "refresh" option in Andrei's node but I've given him a "heads up" that support for it may be coming due to this FR.

schasj, 2022-09-23:
@Koenkk, I now have a capture of Hubitat sending a "refresh" to the exact device I need it for. I hope this gives you everything that is needed to provide the same function. Please understand that the refresh command is available on all devices with HE so it's not just the device I care about.

[HErefresh.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/9636584/HErefresh.pcapng.zip)

Koenkk, 2022-09-23:
I need to have your network key to decrypt the sniff.

schasj, 2022-09-23:
Right. It's one of these

01:03:05:07:09:0B:0D:0F:00:02:04:06:08:0A:0C:0D
d7ae7266fdbbc37483f1ba4064c45bcc
d0f1c1cc6ffd28605be8469df043a37b

Koenkk, 2022-09-25:
So it seems to be just a standard read request:

[image]

I'm a bit lost where this issue is about, z2m supports read commands if a proper converter is defined for it. This is also documented for every property on the device page. 

e.g. https://www.zigbee2mqtt.io/devices/ZBMINI-L.html#switch

> To read the current state of this switch publish a message to topic zigbee2mqtt/FRIENDLY_NAME/get with payload {"state": ""}

schasj, 2022-09-26:
@Koenkk, "/get" should work now that @sjorge has added the needed elements above but it's not as convenient as what HE does (issue a read request for each attribute) and I don't think Andrei will be able to add a "refresh" to his node. But I see that it's just a difference in philosophy.

I've made numerous attempts to get the converter right using the one you provided some weeks ago for the Iris 3326-L, pieces of what @sjorge added in his PR, and what I read in [new devices](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#instructions). All the attempts I've made resulted in an error upon startup:

[image]

I also don't understand the relationship between the external converter and the built-in device code...Does the converter replace or augment the built-in?

Anyway, here is my latest attempt

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
   zigbeeModel: ['RES005'],
   model: 'KMPCIL_RES005',
   vendor: 'KMPCIL',
   description: 'Environment sensor',
   fromZigbee: [
      fz.battery, fz.temperature, fz.humidity, fz.pressure, fz.illuminance, fz.kmpcil_res005_occupancy,
      fz.kmpcil_res005_on_off
   ],
   toZigbee: [tz.kmpcil_res005_on_off, tz.illuminance],
   exposes: [
      e.battery(), e.temperature(), e.humidity(), e.pressure(), e.illuminance().withAccess(ea.STATE_GET),
      e.illuminance_lux().withAccess(ea.STATE_GET), e.occupancy(), e.switch()
   ],
};

module.exports = definition;
`

Koenkk, 2022-09-28:
> I've made numerous attempts to get the converter right using the one you provided some weeks ago for the Iris 3326-L, pieces of what @sjorge added in his PR, and what I read in [new [...]

Try removing the entries of `toZigbee`/`fromZigbee` one-by-one to figure out which one causes the problem

> I also don't understand the relationship between the external converter and the built-in device code...Does the converter replace or augment the built-in?

It is a replacement.

schasj, 2022-09-28:
> Try removing the entries of `toZigbee`/`fromZigbee` one-by-one to figure out which one causes the problem

It turns out to be tz.illuminance which, if I understand correctly, is what makes the thing I need possible. So now what?

I suppose I just have to wait for the next release to have a Docker image made.

Koenkk, 2022-09-28:
That converter is only available in the dev branch, so try upgrading to that (or wait for the next release on 1 October) (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

schasj, 2022-10-01:
It's not the whole-device refresh that I hoped for but v1.28 does allow the illuminance "get" that was the specific item I currently need.

closed 2022-10-01 by schasj