Koenkk/zigbee2mqtt #13863: [New device support]:

annunity, 2022-09-05:
### Link

https://www.dlink.com.au/home-solutions/DCH-B112-Smart-Door-Window-Sensor

### Database entry

0xbc0f9afffe15b5d0

### Comments

I am having trouble adding support for D-Link DCH-B112 for Z2M.

I am new to .js formatting so not sure how to fix this.

The device looks most similar to https://www.zigbee2mqtt.io/devices/LDSENK08.html so I have tried cloning https://raw.githubusercontent.com/Koenkk/zigbee-herdsman-converters/master/devices/adeo.js

Sorry I couldn't find the database entry in home assistant. 
0xbc0f9afffe15b5d0

        "0xbc0f9afffe15b5d0": {
            "date_code": "20190509-54",
            "definition": null,
            "endpoints": {
                "1": {
                    "bindings": [
                        {
                            "cluster": "genPollCtrl",
                            "target": {
                                "endpoint": 1,
                                "ieee_address": "0x84b4dbfffef18087",
                                "type": "endpoint"
                            }
                        }
                    ],
                    "clusters": {
                        "input": [
                            "genBasic",
                            "genPowerCfg",
                            "genIdentify",
                            "genPollCtrl",
                            "ssIasZone",
                            "haDiagnostic"
                        ],
                        "output": [
                            "genOta"
                        ]
                    },
                    "configured_reportings": [],
                    "scenes": []
                }
            },
            "friendly_name": "frontdoor_doorsensor",
            "ieee_address": "0xbc0f9afffe15b5d0",
            "interview_completed": true,
            "interviewing": false,
            "manufacturer": "D-Link",
            "model_id": "DCH-B112",
            "network_address": 51998,
            "power_source": "Battery",
            "software_build_id": "1.00.02",
            "supported": false,
            "type": "EndDevice"

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    DCH-B112: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                contact: !((zoneStatus & 1) > 0),
                vibration: (zoneStatus & 1<<1) > 0,
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
            };
        },
    },
};

module.exports = [
    {
        zigbeeModel: ['DCH-B112'],
        model: 'DCH-B112',
        vendor: 'D-Link',
        description: 'D-Link wireless smart door window sensor with vibration',
        fromZigbee: [fzLocal.DCH-B112, fz.battery],
        toZigbee: [],
        exposes: [e.battery_low(), e.contact(), e.vibration(), e.tamper(), e.battery()],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
        },
    },
];
```

annunity, 2022-10-06:
Any help? Cheers

Koenkk, 2022-10-06:
- With the external converter you posted, what works?
- If nothing works, can you provide the debug log when triggering the open/close event?

annunity, 2022-10-06:
Hi,

Thanks for replying. 
Nothing works unfortunately. The info I get from the debug screen probably is not very helpful.

Warning 2022-10-07 10:05:11Device '0xbc0f9afffe15b5d0' with Zigbee model 'DCH-B112' and manufacturer name 'D-Link' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning 2022-10-07 10:05:31Device '0xbc0f9afffe15b5d0' with Zigbee model 'DCH-B112' and manufacturer name 'D-Link' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Warning 2022-10-07 10:05:35Received message from unsupported device with Zigbee model 'DCH-B112' and manufacturer name 'D-Link'
Warning 2022-10-07 10:05:44Received message from unsupported device with Zigbee model 'DCH-B112' and manufacturer name 'D-Link'

Koenkk, 2022-10-07:
This means you did not load the external converter.

- save the external converter as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m

closed 2022-10-08 by annunity