Koenkk/zigbee2mqtt #13871: TuYa TS0042 TuYa TS0041 indicator lights randomly(?) blinking

evlo, 2022-09-05:
### What happened?

I have couple of TuYa TS0042 TuYa TS0041 that actually looks like this https://allegro.pl/oferta/pilot-6-scen-i-akcji-zigbee-3-0-dla-aplikacji-tuya-12093799425bit not like picture here https://www.zigbee2mqtt.io/devices/TS0042.html#tuya-ts0042 but that is jsut different faceplate, inside can be same. Anyway after i switched from tuya to zigbee2mqtt these switches randomly flash green indicator leds (it does not happen with wired variant of these switches). It is quite sporadic flash, i did not see anything related in the log around that time, but i only noticed it late at night and did not check very thoroughly.

### What did you expect to happen?

Not to flash indicator light unless pairing or actuated.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

zigstar ethernet

### Debug log

i will post when i notice it again and not almost sleeping

james-fry, 2022-09-05:
I have the 4 button scene controller. It blinks all 4 LEDs periodically. As far as I know there is no way to stop it. Originally I thought it was for low battery indication, but changed to new batteries and it still happens. Tt seems its a "feature" of this family of devices.

Mine is:
Zigbee Model
TS0044
Zigbee Manufacturer
_TZ3000_vp6clf9d
Description
Wireless switch with 4 buttons

evlo, 2022-09-05:
Thought it does not blink when used with tiya app so something must be different

james-fry, 2022-09-05:
OK fair point. I only ever used my 4 button scene switch with Z2M. Seems like there is a missing configuration that the device expects that would prevent the flashing.

evlo, 2022-09-05:
I don't know how to pause log or open log in some history way as i run z2m as HAsupervised addon, but blink happens when device announces itselfs, that is not surprising, but why does it feel need to announce itself and not with tuya hub (I would have most likely noticed the blinks with tuya hub, as I had used those switeches with tuya hub too).
I do not know how to debug the zigbee config etc.
Does the configuration happen when pairing? Maybe it is possible to sniff out the tuya hub communication with the device somehow?

evlo, 2022-09-09:
[image]

evlo, 2022-09-09:
Maybe this can be related?
[image]

I do not know how to copy from log in z2m ha addon, it scrolls just way to fast, also in the files like zigbee2mqtt/log/2022-09-08.12-24-53/log2.txt i don't understand what the timestamp used is and i can't find the latest log etc.

I did manage to capture occurrence when clicking the switch button did nothing, it is this switch https://www.zigbee2mqtt.io/devices/TS0041.html#tuya-ts0041 , i did blink the led, but no action was done in HA, then on second click HA registered the click action and turned on the switch that is connected to the lights etc.
[image]

Do you see anything wrong there.

Also the battery status is now missing {I don't mind that, just observation)

I wonder if I maybe misunderstood something basic with zigbee and this sleeping of the devices or whatever it is, is expected behaviour?!

evlo, 2022-09-10:
I tried to migrate to ZHA and it does the same led flashing stuff :/

evlo, 2022-09-13:
https://github.com/Koenkk/zigbee2mqtt/issues/8072 this issue is probably where to continue as both (+battery] issues are discussed there and that is where most of the posts are.

james-fry, 2022-09-13:
> #8072 this issue is probably where to continue as both (+battery] issues are discussed there and that is where most of the posts are., but seems to be capures os z2m traffic, not on tuya gw

It might be related, but certainly I dont see the battery drain that was mentioned. My 4 button scene switch is still on its original battery after 1-2 years. I do however see the other issue you have with [missed button presses](https://github.com/zigpy/zha-device-handlers/issues/1746)...

robertalexa, 2022-10-01:
the blinking is happening when the device reconnects to the network. issue https://github.com/Koenkk/zigbee2mqtt/issues/8072 includes sniffs from the tuya gateway as well and the behaviour does not occur on the tuya gateway. i have lacked time to continue that investigation

Awebian, 2022-11-27:
I am having the same issue.  Has anyone found a solution?  Woke up to find my Livingroom switch with all 4 LEDs flashing...

Awebian, 2022-11-27:
OK, swapped battery and it stopped flashing.  Funny that the battery level was not reported as even close to 0%?  Possibly the firmware on the switch is going into low battery mode too early??

robertalexa, 2022-11-27:
Have you actually re-paired the affected devices since the fix was released?

Awebian, 2022-11-27:
> 

There was a specific fix for this flashing issue?  I must have missed it above.   I am using:

Zigbee2MQTT version, [1.28.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.2) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)

Coordinator type, zStack3x0
Coordinator revision, 20220219
Frontend version, 0.6.114

robertalexa, 2022-11-27:
> the blinking is happening when the device reconnects to the network. issue #8072 includes sniffs from the tuya gateway as well and the behaviour does not occur on the tuya gateway. i have lacked time to continue that investigation
> 
> LE: i believe this issue might be a duplicate of 8072, so keep an eye on that thread

Follow this thread

robertalexa, 2022-12-08:
Probably issue during pairing? Try to re-pair the device, make sure to click the button once after allow join, to ensure the device is awake and ready to talk to your network.

Should look somthing like this:
[image]

Action is only populated while you press the button for a split second only.

evlo, 2023-01-16:
I wonder if flashing with https://github.com/MattWestb/EFR32-FW/blob/main/tuya_ZS3L/README.md can make the device behave better, but I do not understand how to use it.