Koenkk/zigbee2mqtt #13875: Unable to add TRADFRI motion sensor to a group

Sonyoyo, 2022-09-05:
### What happened?

I tried (from the frontend) to add a TRADFRI motion sensor to a newly created group.
The group is correctly created and I can add light to it (for testing purpose).

However, when I try with the TRADFRI motion sensor, nothing happen for Iwhile and I then have an error message:

```
Debug 2022-09-05 22:21:49Error: Command 0x0c4314fffe547557/1 genGroups.add({"groupid":2,"groupname":""}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 5897 - 1 - 6 - 4 - 0 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7)
Info 2022-09-05 22:21:49MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x0c4314fffe547557/1","group":"Motion cucina"},"error":"Failed to add from group (Command 0x0c4314fffe547557/1 genGroups.add({\"groupid\":2,\"groupname\":\"\"}, {\"sendWhen\":\"active\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 5897 - 1 - 6 - 4 - 0 after 10000ms))","status":"error","transaction":"1jpul-1"}'
Error 2022-09-05 22:21:49Failed to add from group (Command 0x0c4314fffe547557/1 genGroups.add({"groupid":2,"groupname":""}, {"sendWhen":"active","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 5897 - 1 - 6 - 4 - 0 after 10000ms))
```

The message generated by the frontend to add the device to the group looks fine...

```
Debug 2022-09-05 22:27:48Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/add' with data '{"device":"0x0c4314fffe547557/1","group":"Motion cucina","transaction":"1jpul-12"}'
Info 2022-09-05 22:27:48Adding 'Sensore movimento cucina' to 'Motion cucina'
```

The question is: can I add this kind of `binary sensors` to a zigbee2mqtt group?

Thanks in advance for your help.

### What did you expect to happen?

I expect the sensor to be added to the group

### How to reproduce it (minimal and precise)

Create a group
Trying to add a TRADFRI motion sensor to it with the frontend (select the device, then click on `add to group`).

### Zigbee2MQTT version

1.27.2 commit: 7dc48fb

### Adapter firmware version

20210708

### Adapter

Sonoff zigbee 3.0 dongle plus

MattWestb, 2022-09-06:
IKEAs motion sensors is one **Light Controller** device and they is not added to one group as light they is being bounded to groups for sending the commands to it.

Sonyoyo, 2022-09-06:
So, if I understand correctly, I cannot make a group of "binary sensors" directly in zigbee2mqtt? I can do that in Home Assistant and I assumed I could do the same to group sensors such as motion sensors, window/door sensors and so on. If that is not the case, I will do that in Home Assistant for my application.

closed 2022-10-15 by github-actions[bot]