Koenkk/zigbee2mqtt #13876: Device became unsupported (again) after update 1.27.2 ECO-DIM.07 Zigbee Pro (EcoDim Smart Dimmer Switch 200W LED)

aykin, 2022-09-05:
### What happened?

### Description

I had recently updated Zigbee2Mqtt to 1.27.2 and noticed that the Zigbee dimmer that had worked normally was not being recognized as a supported device.

`Received message from unsupported device with Zigbee model 'EcoDim-Zigbee 3.0' and manufacturer name 'EcoDim BV'`

I have looked at the supported devices page and did not find it anymore. As mentioned in a previous issues #7325 and #7282 the device used to be found on the page. The problem seems to have resurfaced after a recent update. 

Note that the modelName that was mentioned in the previous issues "Dimmer-Switch-ZB3.0" is different from what I see which is "EcoDim-Zigbee 3.0" . I am pretty sure this is the device : [store link](https://www.ecobright.nl/led-dimmer/smart-led-dimmer-zigbee-zwave-0-200w-voor-alle-afdekramen/)

### What I have tried

I have tried adding a custom device definition and using the guide but the existing ecodim definitions use a `fingerprint` parameter to identify rather than `modelName`. I couldn't find documentation for how this fingerprinting works.

I have managed to recognize the device using 
`zigbeeModel: ['EcoDim-Zigbee 3.0']` . 

Problem: 
Unfortunately I did not manage to get too much further because any combination of fz, tz or expose parameters resulted in irrelevant parameters ending up in mqtt (brightness_left, brighness_right, state_left, etc.) Note that this device has only one dimmer. It should not have left/right endpoints. I think the existing definitions somehow confuse it with another dimmer with 2 knobs (which is already supported recently). Note that I am not adding anything for this multi endpoint feature. Here is the config:

`{`
`        zigbeeModel: ['EcoDim-Zigbee 3.0'],`
`        model: 'Eco-Dim.07',`
`        vendor: 'EcoDim',`
`        description: 'Zigbee Pro',`
`        extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),`
`    }`

This configuration results in these payloads:

`Zigbee2MQTT:info  2022-09-05 23:05:36: MQTT publish: topic 'zigbee2mqtt/kitchen-light', payload '{"brightness":193,"brightness_left":254,"brightness_right":62,"linkquality":255,"state":"ON","state_left":"OFF","state_right":"ON"}'`

For some reason **brightness_right** is always 62 and never changes. brightness_left seem to follow the actual dimmer setting

### Additional information

database.db parameter for the device:
`{"id":13,"type":"Router","ieeeAddr":"0x0c4314fffe1f9f40","nwkAddr":18777,"manufId":4714,"manufName":"EcoDim BV","powerSource":"Mains (single phase)","modelId":"EcoDim-Zigbee 3.0","epList":[1,67,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff059c06","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00212effff059c06","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"67":{"epId":67,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20210228","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1662194586394,"defaultSendRequestWhen":"immediate"}`

[Link to the manual](https://www.ecobright.nl/wms/fm/userfiles/catalogus/attachments/ECO-DIM.07-handleiding-V3-Zigbee-3-talen-2021.pdf)

### What did you expect to happen?

I expected to be able to use the device after the update

When adding custom device configuration, I did not expect to see `brightness_left` and `brighness_right` values 

### How to reproduce it (minimal and precise)

Pull zigbee2mqtt:1.27.2 on docker. Pair with a Smart Dimmer Switch 200W LED Eco-Dim.07 Zigbee Pro and manipulate the device afterwards

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

0x26780700

### Adapter

Conbee II

Koenkk, 2022-09-06:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const color = require('zigbee-herdsman-converters/lib/color');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
        ]},
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 67, inputClusters: [], outputClusters: []},
            {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
        ]},
    ],
    zigbeeModel: ['Dimmer-Switch-ZB3.0'],
    model: 'Eco-Dim.07/Eco-Dim.10',
    vendor: 'EcoDim',
    description: 'Zigbee & Z-wave dimmer',
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

aykin, 2022-09-06:
The light is recognized but I still see the "brightness_right" and "state_right" etc. I have no idea where these are coming from

`Zigbee2MQTT:info  2022-09-06 11:18:18: kitchen-light (0x0c4314fffe1f9f40): Eco-Dim.07/Eco-Dim.10 - EcoDim Zigbee & Z-wave dimmer (Router)`

Here is what happens when turn it off through MQTT:

`Zigbee2MQTT:info  2022-09-06 11:18:19: Configuring 'kitchen-light'`
`Zigbee2MQTT:debug 2022-09-06 11:18:19: Received MQTT message on 'zigbee2mqtt/kitchen-light/set' with data '{"state":"OFF","transition":3}'`
`Zigbee2MQTT:debug 2022-09-06 11:18:19: Publishing 'set' 'transition' to 'kitchen-light'`
`Zigbee2MQTT:debug 2022-09-06 11:18:19: Publishing 'set' 'state' to 'kitchen-light'`
`Zigbee2MQTT:info  2022-09-06 11:18:19: MQTT publish: topic 'zigbee2mqtt/kitchen-light', payload '{"brightness":1,"brightness_left":254,"brightness_right":62,"linkquality":252,"state":"OFF","state_left":"OFF","state_right":"ON"}'`

Now, it has 3 brightnesses and 3 states for some reason. Is this because of this config? Or Does mqtt somehow hold onto some previous parameters from my previous attempts?

Koenkk, 2022-09-07:
You can delete those manually by stopping z2m, removing these from `data/state.json` and starting z2m again.

aykin, 2022-09-07:
I can confirm that the irrelevant params are gone. Thanks! I didn't know about this state.json usage. The dimmer seems to work correctly now.

How do you want to close the issue? Should I create a PR by adding this new definition? Or modify one of the existing definitions? It is your code, so maybe it is not right for me to even to create a PR ?

Koenkk, 2022-09-08:
Integrated the changes, you can keep using the external converter until the next release (every 1st of the month).

closed 2022-09-08 by Koenkk

zwartevogel, 2022-11-17:
@Koenkk In my case this device only worked without the deviceId. 

My external convertor:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const color = require('zigbee-herdsman-converters/lib/color');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 242, profileID: 41440, inputClusters: [], outputClusters: [33]},
        ]},
        {type: 'Router', manufacturerName: 'EcoDim BV', modelID: 'EcoDim-Zigbee 3.0', endpoints: [
            {ID: 1, profileID: 260, deviceID: 257, inputClusters: [0, 3, 4, 5, 6, 8, 2821, 4096], outputClusters: [25]},
            {ID: 67, inputClusters: [], outputClusters: []},
            {ID: 242, profileID: 41440, deviceID: 97, inputClusters: [], outputClusters: [33]},
        ]},
    ],
    zigbeeModel: ['Dimmer-Switch-ZB3.0'],
    model: 'Eco-Dim.07/Eco-Dim.10',
    vendor: 'EcoDim',
    description: 'Zigbee & Z-wave dimmer',
    extend: extend.light_onoff_brightness({noConfigure: true, disableEffect: true}),
    configure: async (device, coordinatorEndpoint, logger) => {
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.onOff(endpoint);
        await reporting.brightness(endpoint);
    },
};

module.exports = definition;
`

I only used the first fingerprint so I left the deviceId there but I think it can be removed there also.

Koenkk, 2022-11-20:
@zwartevogel can you provide the data/database.db entry of your device?

zwartevogel, 2022-11-21:
@Koenkk this is what I found there:
``{"id":22,"type":"Router","ieeeAddr":"0x0c4314fffe253bb2","nwkAddr":59913,"manufId":4714,"manufName":"EcoDim BV","powerSource":"Mains (single phase)","modelId":"EcoDim-Zigbee 3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":177}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe7401e5","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffe7401e5","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":8,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20210228","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-526939472},"lastSeen":1669025188248,"defaultSendRequestWhen":"immediate"}
``

Koenkk, 2022-11-21:
Are you using an EZSP adapter? If yes, the deviceid issue just has been fixed in the latest dev. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

- Update to the latest dev
- Force remove the device through the z2m frontend
- Restart z2m
- Re-pair the device
- Should work now.

zwartevogel, 2022-11-21:
Yes, using ezsp. Thanks for your response! I’ll look into it.