Koenkk/zigbee2mqtt #13878: Smartwings WM25L-Z Not Reporting State after 1.27.1

warllo54, 2022-09-06:
### What happened?

After 1.27-1, I think PR 4507 causes the Smartwings WM25L-Z to not report state correctly.

When the blinds are closed, the attribute `Read result of 'closuresWindowCovering': {"currentPositionLiftPercentage":0}` 

According to the device specific settings  0 should equal Closed however this is the state information:

`
{
    "battery": 77,
    "last_seen": "2022-09-06T00:49:26.530Z",
    "linkquality": 39,
    "position": 100,
    "state": "OPEN"
}
`

### What did you expect to happen?

I would expect that when the `Read result of 'closuresWindowCovering': {"currentPositionLiftPercentage":0}` is equal to 0 the state of the cover should be  **_Closed_**

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

zStack3x0

### Adapter

zStack3x0 Sonoff 3.0 c2600

Koenkk, 2022-09-06:
Could you check if the issue is fixed with the following external converter:

```js
const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const reporting = require('../lib/reporting');
const e = exposes.presets;

module.exports = [
    {
        zigbeeModel: ['WM25/L-Z'],
        model: 'WM25L-Z',
        vendor: 'Smartwings',
        description: 'Roller shade',
        fromZigbee: [fz.cover_position_tilt, fz.battery],
        toZigbee: [tz.cover_state, tz.cover_position_tilt],
        meta: {battery: {dontDividePercentage: true}, coverInverted: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.currentPositionLiftPercentage(endpoint);
        },
        exposes: [e.cover_position(), e.battery()],
    },
];

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

warllo54, 2022-09-06:
When applying the external converter zigbee2mqtt fails to load. Did I not understand the directions correctly?

Here are the logs:

```
[07:44:22] INFO: Handing over control to Zigbee2MQTT Core ...
node:internal/modules/cjs/loader:936
  throw err;
  ^
Error: Cannot find module '../lib/exposes'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at /app/dist/util/externally-loaded.js:1:17
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
```

Koenkk, 2022-09-07:
great, pushed the fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-07 by Koenkk

digiblur, 2022-09-08:
@warllo54 are  you using True or False on the inverted flag on the settings for these.  I've got 4 of these and two of these are driving me nuts with Zigbee2MQTT even without this change.  They report their state correctly for 0-100 and such but the Open Close commands are backwards.  Oddly only two of them do it.  The open command is close and close command is open yet the state is correct.  

With this change all of my 4 blinds now show open when closed and the commands are still backwards on two of them.  Super weird.  

@Koenkk any way to do a settings specific config on these to just be able to invert the open/close commands and not the state?

warllo54, 2022-09-08:
@digiblur

I’m using false for the inverted setting.

Which side of the blind is your motor on? Are all 4 the same?

I had to reverse the motor direction per the directions, otherwise for the control up was close and down was open. My remote is backwards but the zigbee stuff all works great.

digiblur, 2022-09-08:
Motor is on the right on all 4. However 2 of them were manufactured before the other two. I thought about changing the direction but we do use the remote at times on group and that would mess it up.

Weird that my states are correct and yours are not.

digiblur, 2022-09-08:
@warllo54 do you get these error messages from them as well.  I usually notice them on the Z2M startup.  I've ignored them as the first two worked perfectly.  Not sure if this part of the inverted commands issue.  

Failed to configure 'ZigLivingRoomBlinds', attempt 4 (Error: ConfigureReporting 0x2c1165fffee19293/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 17901 - 1 - 220 - 1 - 7 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:559:17) at processTimers (node:internal/timers:502:7))

warllo54, 2022-09-08:
I do get these errors however they don’t seem to affect functionality.

digiblur, 2022-09-08:
I reached out to the manufacturer as well to see if there is some way to reverse the commands if we can't get it done in Z2M. I might throw one or two on a dev ZHA network just to see if it is backwards there too.

warllo54, 2022-09-08:
I started with zha and it was backwards for me as well there.

digiblur, 2022-09-08:
> I started with zha and it was backwards for me as well there. Sent from Proton Mail for iOS
> […](#)

ahhh...thanks for saving me that time!  I'll have to dig through some other shades to see if there is an example of inverting the commands per device.

nathankopp, 2022-10-09:
This is not fixed for me.

Zigbee2mqtt 1.28.0-1
SmartWings WM25L-Z
Motor on right side

The position (0..100) and state (OPEN and CLOSE) in Zigbee2mqtt show reversed from reality.  Changing the "invert_cover" setting reverses the position but does not fix the OPEN/CLOSE state.  Also, the actual position is different from the inline help text.

Inline help text says "false: open=100,close=0, true: open=0,close=100 (default false)."   HOWEVER if I set it to false, then I see open=0,close=100.  And if I set it to true, I see open=100,close=0.

I tried the "external_converters" suggestion above, but it did not make any difference for my situation.

nathankopp, 2022-10-09:
I had to use the following external converter to get my Smartwings shades to work correctly (motor on right side).

Note: coverInverted: false
Added a new backwards_cover_state to reverse the commands sent for open/close

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;

const utils = require('zigbee-herdsman-converters/lib/utils');

backwards_cover_state = {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {'open': 'downClose', 'close': 'upOpen', 'stop': 'stop', 'on': 'downClose', 'off': 'upOpen'};
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.command('closuresWindowCovering', lookup[value], {}, utils.getOptions(meta.mapped, entity));
        },
    };

module.exports = [
    {
        zigbeeModel: ['WM25/L-Z'],
        model: 'WM25L-Z',
        vendor: 'Smartwings',
        description: 'Roller shade',
        fromZigbee: [fz.cover_position_tilt, fz.battery],
        toZigbee: [backwards_cover_state, tz.cover_position_tilt],
        meta: {battery: {dontDividePercentage: true}, coverInverted: false},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.currentPositionLiftPercentage(endpoint);
        },
        exposes: [e.cover_position(), e.battery()],
    },
];`

digiblur, 2022-10-09:
@nathankopp any way to do the reversed commands on the settings specific tab?  Looks like this would reverse the commands for all blinds of this model which wouldn't work in all cases.

nathankopp, 2022-10-09:
Probably something like this would work (untested).  When `meta.coverInverted` is true, it will use the "left motor" (normal) lookup.  When `meta.coverInverted` is false, it will use the "right motor" (inverted) lookup.  I realize this seems backwards, but it is what would work for my units, since for me, `meta.coverInverted` is false but the commands still need to be backwards..  An alternative would be to introduce another attribute in the "meta" object specifically to switch the command lookup independently of the `coverInverted` flag.

```
const lookupSmartwingsLeftMotor = {'open': 'upOpen', 'close': 'downClose', 'stop': 'stop', 'on': 'upOpen', 'off': 'downClose'};
const lookupSmartwingsRightMotor = {'open': 'downClose', 'close': 'upOpen', 'stop': 'stop', 'on': 'downClose', 'off': 'upOpen'};
backwards_cover_state = {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = meta.coverInverted ? lookupSmartwingsLeftMotor : lookupSmartwingsRightMotor;
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.command('closuresWindowCovering', lookup[value], {}, utils.getOptions(meta.mapped, entity));
        },
    };
```

digiblur, 2022-10-09:
I have seen blinds that have the state correct on both but yet the commands are backwards on both.

nickell, 2022-11-01:
Thank you @nathankopp for posting your solution. I have copied your external converter to my installation and it fixed my issues as well (motor right side).

stu247, 2022-11-03:
I received two Zigbee indoor roller blinds from Smartwings.  These blinds have the motor on the left. I am running zigbee2mqtt 1.28.0 in a container and use Sonoff Zigbee 3.0 USB Dongle Plus. I had the same issue as @nathankopp and his external converter code worked for me.  I did not need to make any changes because my motor is on the left.  Thanks @nathankopp!

guillochon, 2022-12-13:
Can we reopen this issue as it is not fixed? I'm having the same problem as everyone else: shade says it is closed when it is open and vice-versa.

Koenkk, 2022-12-13:
@guillochon 
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/2b5f2a27233c9dc7d91ade369cff81a8

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

guillochon, 2022-12-13:
Hi @Koenkk, yep that fixed it! When this is eventually fixed in the latest release, will have to disable this converter?

digiblur, 2022-12-13:
I don't think there is a one fix for all given some of their motors commands are backwards.    I've been using this with good results on the recent motors they have been sending:  
```js
backwards_cover_state = {
        key: ['state'],
        convertSet: async (entity, key, value, meta) => {
            const lookup = {'open': 'downClose', 'close': 'upOpen', 'stop': 'stop', 'on': 'downClose', 'off': 'upOpen'};
            value = value.toLowerCase();
            utils.validateValue(value, Object.keys(lookup));
            await entity.command('closuresWindowCovering', lookup[value], {}, utils.getOptions(meta.mapped, entity));
        },
    };

module.exports = [
    {
        zigbeeModel: ['WM25/L-Z'],
        model: 'WM25L-Z',
        vendor: 'Smartwings',
        description: 'Roller shade3',
        fromZigbee: [fz.cover_position_tilt, fz.battery],
        toZigbee: [backwards_cover_state, tz.cover_position_tilt],
        meta: {battery: {dontDividePercentage: true}, coverInverted: false},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.currentPositionLiftPercentage(endpoint);
        },
        exposes: [e.cover_position(), e.battery()],
    },
];
```

stu247, 2022-12-13:
With the W25L-Z.js file that @Koenkk posted above on 13 Dec 2022, the read and set of position work correctly, but open and close commands do not.  When the shade is at 50%, clicking open will close the shades (to 0%) and clicking close will open the shades (to 100%).  This is true for both motor on the left and motor on the right.  When the shade is at 0%, clicking open does nothing (because it is trying to close and it is already closed?).  When the shade is 100%, clicking close does nothing (because it is trying to open and it is already open?).

The code with backwards_cover_state that @digiblur just posted works for me for both motor on the left shades and motor on the right shades.

digiblur, 2022-12-13:
I snagged that code out of a thread here. Works great as long as all of yours are backwards. I had two that were not backwards but have since swapped these so I can use the backwards command code. It would be great if we had a per entity option for this.

guillochon, 2022-12-14:
Could someone explain why my shade suddenly flipped to reporting the wrong state? Was there an update to z2m? My shade was working perfectly fine for months prior to the last week or two.

highincolorado, 2022-12-31:
I bought 3 Smartwings shades, 1 indoor and 2 outdoor. The indoor is mounted and functions but as described above gives state as open when closed and closed when open. I have not installed the 2 outdoor shades yet because winter. The remote open and close buttons work as expected. Reading the above is not helping me because I don't know enough yet. Be happy to send any data I can to help resolve the issue. The indoor unit is right hand motor the outdoor are 1 of each as I had to buy 2 to span the gap between posts. If we have a warmer day I will mount the outdoor units and see what's what with them.

drizuid, 2023-01-10:
mine, delivered today & ordered last week, both suffer from the inversion issue. Setting invert_cover appears to have no effect
```
{
    "battery": 39,
    "last_seen": "2023-01-10T15:29:45-05:00",
    "linkquality": 80,
    "position": 0,
    "state": "CLOSE"
}
```
[image]

I've not made any changes from this thread yet, but i saw a request for the data/database.db and thought I would try to help out, so below are my 2.

```
{"id":24,"type":"EndDevice","ieeeAddr":"0x5c0272fffe06f0dd","nwkAddr":53865,"manufId":4098,"manufName":"Smartwings","powerSource":"Unknown","modelId":"WM25/L-Z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,1,3,4,5,258],"outClusterList":[3,25],"clusters":{"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":90}},"genBasic":{"attributes":{"modelId":"WM25/L-Z","manufacturerName":"Smartwings","powerSource":0,"zclVersion":8}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a5120c","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a5120c","endpointID":1}],"configuredReportings":[],"meta":{}}},"zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1673380536390,"defaultSendRequestWhen":"immediate"}

{"id":25,"type":"EndDevice","ieeeAddr":"0x60a423fffe2fbf53","nwkAddr":51581,"manufId":4098,"manufName":"Smartwings","powerSource":"Unknown","modelId":"WM25/L-Z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,1,3,4,5,258],"outClusterList":[3,25],"clusters":{"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":39}},"genBasic":{"attributes":{"modelId":"WM25/L-Z","manufacturerName":"Smartwings","powerSource":0,"zclVersion":8}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a5120c","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b0022a5120c","endpointID":1}],"configuredReportings":[],"meta":{}}},"zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1673381222906,"defaultSendRequestWhen":"immediate"}
```

BradleyFord, 2023-03-13:
We need to be very careful with this change, as I feel it could make blinds that are working correctly for some people to then be inverted.
So far people are only talking about the motor side being on left or right; but that is not the only variable.

The variables are:
- Motor on left or right
- Fabric rolling to front (away from window) vs Fabric rolling to back (towards window)

turbotiger74, 2023-03-13:
I have one of their remotes from smartwings, and one of the options on the remote is to reverse the motor. I'm not sure if that changes the actual motor itself, or it's just the remote that changes.

digiblur, 2023-03-17:
The only way to really solve it is to be able to configure this per device in the settings for each.

bobokun, 2024-04-04:
Anyone having issues with the external converter not working on the latest version?

digiblur, 2024-04-04:
I know they changed it up in 1.36.1 so I haven't upgraded myself.  I tried the procedure here to reverse things but pressing the P button on the remote doesn't jog mine like it says and then it gets into some other mode.  https://github.com/Koenkk/zigbee2mqtt/issues/16514#issuecomment-1500374128

If anyone has the updated converter syntax for these, let me know as well.  I've got a few to fix too.

digiblur, 2024-04-04:
> Anyone having issues with the external converter not working on the latest version? I'm seeing these errors
> 
> ```
> Zigbee2MQTT:error 2024-04-03 21:24:41: Failed to load external converter file 'ext_converter.js' (Cannot find module 'zigbee-herdsman-converters/lib/extend'
> Require stack:
> - /app/dist/util/utils.js
> - /app/dist/util/settings.js
> - /app/index.js)
> Zigbee2MQTT:error 2024-04-03 21:24:41: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
> Zigbee2MQTT:error 2024-04-03 21:24:41: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
> Zigbee2MQTT:info  2024-04-03 21:24:41: Logging to console and directory: '/config/zigbee2mqtt/log/2024-04-03.21-24-41' filename: log.txt
> Zigbee2MQTT:info  2024-04-03 21:24:41: Starting Zigbee2MQTT version 1.36.1 (commit #unknown)
> Zigbee2MQTT:info  2024-04-03 21:24:41: Starting zigbee-herdsman (0.40.3)
> ```

Try this fix...looks like it works - https://github.com/Koenkk/zigbee2mqtt/issues/22026#issuecomment-2030622945

bobokun, 2024-04-04:
Thanks! That looks like it fixed the issue :)

digiblur, 2024-04-04:
> Thanks! That looks like it fixed the issue :)

I thought it was going to be crazy based on some other things I was reading but then I saw the one about the commenting out the one line and well... end of story.  Noice!!!