Koenkk/zigbee2mqtt #13882: from 1.22.0 to 1.27.2 I have problem with sockets (tuya) and switchers (livolo)

xfishernet, 2022-09-06:
### What happened?

Ho All! from 1.22.0 to 1.27.2 I have problem with sockets (tuya) and switchers (livolo), but in 1.21.0 all ok. please show bug in my video: https://youtube.com/shorts/OJtrYOxrjJY

### What did you expect to happen?

I expected without debounce 

### How to reproduce it (minimal and precise)

ziggbee2mqtt + homebridge (z2m plug) + homekit 1.22.0 or higher

### Zigbee2MQTT version

1.22.0 or higer

### Adapter firmware version

 zStack3x0 and other

### Adapter

 zStack3x0 and other

### Debug log

no logs

xfishernet, 2022-09-06:
in the current dev branch also does not work (although they wrote here what was fixed

Sigalov, 2022-09-07:
> in the current dev branch also does not work (although they wrote here what was fixed

I've checked this:
test -w [PORT] && echo success || echo failure
and got **success**, from the beginning.

Also added 'pi' to the next groups:
sudo usermod -a -G uucp $USER
sudo usermod -a -G tty $USER
sudo usermod -a -G dialout $USER

and reboot

xfishernet, 2022-09-07:
Привет! А какой ты PORT указывал? У меня failure :-( ветка была текущая dev?

closed 2022-10-17 by github-actions[bot]