Koenkk/zigbee2mqtt #13886: Zigbee2MQTT fails to start - Error Resource temporarily unavailable Cannot lock port

Sigalov, 2022-09-06:
### What happened?

Followed the linux guide to install it on Raspberry pi 4.
I've read all "failed to start".
After "npm start":
pi@homebridge:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.27.2 start
> node index.js

Zigbee2MQTT:info  2022-09-06 14:10:22: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2022-09-06.14-10-22' filename: log.txt
Zigbee2MQTT:info  2022-09-06 14:10:22: Starting Zigbee2MQTT version 1.27.2 (commit #7dc48fb)
Zigbee2MQTT:info  2022-09-06 14:10:22: Starting zigbee-herdsman (0.14.53)
Zigbee2MQTT:error 2022-09-06 14:10:23: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2022-09-06 14:10:23: Failed to start zigbee
Zigbee2MQTT:error 2022-09-06 14:10:23: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
Zigbee2MQTT:error 2022-09-06 14:10:23: Exiting...
Zigbee2MQTT:error 2022-09-06 14:10:23: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/@serialport/stream/lib/index.js:242:12

I've executed: ls -l /proc/[0-9]/fd/ |grep /dev/ttyUSB0
and got:

pi@homebridge:/opt/zigbee2mqtt $ ls -l /proc/[0-9]/fd/ |grep /dev/ttyUSB0
ls: cannot open directory '/proc/1/fd/': Permission denied
ls: cannot open directory '/proc/2/fd/': Permission denied
ls: cannot open directory '/proc/3/fd/': Permission denied
ls: cannot open directory '/proc/4/fd/': Permission denied
ls: cannot open directory '/proc/5/fd/': Permission denied
ls: cannot open directory '/proc/9/fd/': Permission denied

@ko

### What did you expect to happen?

Zigbee2MQTT to start

### How to reproduce it (minimal and precise)

Installed using linux guide
npm start

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219.hex

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

TzachiGuetta, 2022-09-06:
@Koenkk

oberlon, 2022-09-06:
Check permissions on your user pi. Can pi access your usb dev?

TzachiGuetta, 2022-09-07:
@oberlon , tried:
  This can be tested by executing: test -w [PORT] && echo success || echo failure ( e.g. test -w /dev/ttyACM0 && echo
success || echo failure).
it outputs success

Sigalov, 2022-09-07:
> Check permissions on your user pi. Can pi access your usb dev?

I've checked this:
test -w [PORT] && echo success || echo failure
and got success, from the beginning.

Also added 'pi' to the next groups:
sudo usermod -a -G uucp $USER
sudo usermod -a -G tty $USER
sudo usermod -a -G dialout $USER

and reboot

Koenkk, 2022-09-07:
`Zigbee2MQTT:error 2022-09-06 14:10:23: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'`

Another process is already locking the port, see https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-resource-temporarily-unavailable-cannot-lock-port

I will close this since this cannot be fixed from z2m.

closed 2022-09-07 by Koenkk