Koenkk/zigbee2mqtt #13888: Aqara Magic Cube is not returning/exposing values for all the required attributes.

sujithg9, 2022-09-06:
### What happened?

I was trying to pair both my Aqara Magic Cubes to Zigbee2Mqtt in HomeAssistant.  I see that both join the network successfully but then non of the attributes that are to be exposed like side, action_from_side, action_side, and so on are N/A. 

### What did you expect to happen?

All the required attributes should be available. 

### Zigbee2MQTT version

1.27.2-1 

### Adapter firmware version

6.10.3.0 build 297

### Adapter

EFR32MG21

### Debug log

Zigbee2MQTT:info  2022-09-06 12:43:51: MQTT publish: topic 'zigbee2mqtt/First Floor Control Cube', payload '{"action":null,"action_angle":null,"action_from_side":null,"action_side":null,"action_to_side":null,"battery":70,"current":0,"device_temperature":20,"linkquality":216,"power":4,"power_outage_count":168,"side":null,"voltage":2955}'

danieledwardgeorgehitchcock, 2022-09-06:
That is correct behaviour I believe. As these are actions, they should briefly flash up or be displayed in the Z2M debug logs. You'd then use that as a trigger in an automation

sujithg9, 2022-09-06:
@danieledwardgeorgehitchcock But what about the side, doesn't that be reported .. but it's showing Null, so not sure if that is the expected behavior. Because I remember it to change previously when ever the cube flipped or changed its side.

sujithg9, 2022-09-06:
@danieledwardgeorgehitchcock I think you are right, my bad .. I was having some issue with the automation which was was not pointing to the right entity in HA. Thanks a ton for reminding me

closed 2022-09-06 by sujithg9

Heisenberg2980, 2023-09-09:
@sujithg9 I am having the same issue where my cube is allways reporting "action_from_side":null, "action_side":null, "action_to_side":null. It is reporting the value "side" but unfortunately it is not an accurate value to be used in automations as every time the cube is changed to a difference side in a way that doesn´t trigger an action (i.e. rotating the cube in diagonal) the side doesn´t change even after performing other actions, is your cube reporting those other values now?