Koenkk/zigbee2mqtt #13889: State of Insta model 57005000 switching actuator not updated correctly if pyhiscally switched

kalbfuss, 2022-09-06:
### What happened?

I am using an [Insta](https://www.zigbee2mqtt.io/supported-devices/#v=Insta) model [57008000](https://www.zigbee2mqtt.io/devices/57008000.html#insta-57008000) blinds actor with lift/tilt calibration and inputs for wall switches. The device is recognized by z2m and integrated into the network. I am able to control the device out of z2m. However, the position of the blind is not reported correctly if changed physically via the connected switch.
If I set the position in z2m, the blind is moved to the corresponding position and the position is correctly shown in z2m.
If I change the position via the connected switch, the blind moves until stopped again. The position in z2m is not updated, but remains unchanged. 
The position is correctly shown in the Nexentro app, which communicates with the actor via bluetooth. Means, the actor is aware of the position of the blind.
See also the potentially related bug report [https://github.com/Koenkk/zigbee2mqtt/issues/13877#top](State of Insta model 57005000 switching actuator not updated correctly if pyhiscally switched)

### What did you expect to happen?

If I change the position via the connected switch, the position is updated in z2m.

### How to reproduce it (minimal and precise)

1. Note the position of the blind in z2m.
2. Change the position of the blind via the connected wall switch.
3. Verify the state in z2m. The state in z2m remained unchanged.

### Zigbee2MQTT version

1.27.0

### Adapter firmware version

0x26690700

### Adapter

RaspBee2

### Debug log

Position set in z2m:
```
debug 2022-09-06 19:45:20Received MQTT message on 'zigbee2mqtt1/Garten Markise/set' with data '{"position":47}'
debug 2022-09-06 19:45:20Publishing 'set' 'position' to 'Garten Markise'
info 2022-09-06 19:45:20MQTT publish: topic 'zigbee2mqtt1/Garten Markise', payload '{"linkquality":199,"position":47}'
debug 2022-09-06 19:45:45Received MQTT message on 'zigbee2mqtt1/Garten Markise/set' with data '{"position":0}'
debug 2022-09-06 19:45:45Publishing 'set' 'position' to 'Garten Markise'
info 2022-09-06 19:45:45MQTT publish: topic 'zigbee2mqtt1/Garten Markise', payload '{"linkquality":215,"position":0}'
```
Position changed via connected wall swtich
```
debug 2022-09-06 19:49:36Received Zigbee message from 'Garten Markise', type 'commandUpOpen', cluster 'closuresWindowCovering', data '{}' from endpoint 7 with groupID 8586
info 2022-09-06 19:49:36MQTT publish: topic 'zigbee2mqtt1/Garten Markise', payload '{"action":"open","action_group":8586,"linkquality":255,"position":0}'
info 2022-09-06 19:49:36MQTT publish: topic 'zigbee2mqtt1/Garten Markise', payload '{"action":"","linkquality":255,"position":0}'
info 2022-09-06 19:49:36MQTT publish: topic 'zigbee2mqtt1/Garten Markise/action', payload 'open'
debug 2022-09-06 19:49:56Received Zigbee message from 'Garten Markise', type 'commandStop', cluster 'closuresWindowCovering', data '{}' from endpoint 7 with groupID 8586
info 2022-09-06 19:49:56MQTT publish: topic 'zigbee2mqtt1/Garten Markise', payload '{"action":"stop","action_group":8586,"linkquality":255,"position":0}'
info 2022-09-06 19:49:56MQTT publish: topic 'zigbee2mqtt1/Garten Markise', payload '{"action":"","linkquality":255,"position":0}'
info 2022-09-06 19:49:56MQTT publish: topic 'zigbee2mqtt1/Garten Markise/action', payload 'stop'
debug 2022-09-06 19:49:58Received Zigbee message from 'Garten Markise', type 'commandDownClose', cluster 'closuresWindowCovering', data '{}' from endpoint 7 with groupID 8586
info 2022-09-06 19:49:58MQTT publish: topic 'zigbee2mqtt1/Garten Markise', payload '{"action":"close","action_group":8586,"linkquality":239,"position":0}'
info 2022-09-06 19:49:58MQTT publish: topic 'zigbee2mqtt1/Garten Markise', payload '{"action":"","linkquality":239,"position":0}'
info 2022-09-06 19:49:58MQTT publish: topic 'zigbee2mqtt1/Garten Markise/action', payload 'close'
```

kalbfuss, 2022-09-06:
Reading through the debug log I am getting the impression that the device does not report the position correctly. Or is it possible that z2m interprets the payload incorrectly?

kalbfuss, 2022-09-07:
I have contacted the Insta/Nextron support. The problem is known and indeed caused by the actor. They are working on a firmware update, but the release date is not yet known. I am going to follow up and keep you posted.

closed 2022-10-16 by github-actions[bot]

kalbfuss, 2022-10-29:
Since a couple of days the new firmware version 0.0.52.5 is available. With this firmware version, the blinds actor correctly reports the position of the blinds. For the new firmware to become effective, the actor needs to be removed from the network and rejoined afterwards.

runningman84, 2023-10-25:
I have just two actors both suffer from the same issue. At least one of them got a factory reset but the problem still persists with firmware v.00.00.52.12. Is there any fix available?