Koenkk/zigbee2mqtt #13894: [New device support]: Zigbee 3.0 USB Dongle Plus Dongle E (with Router Firmware)

sujithg9, 2022-09-06:
### Link

https://sonoff.tech/product/diy-smart-switch/sonoff-zigbee-3-0-usb-dongle-plus-e

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x84b4dbfffef8283f","nwkAddr":41483,"manufId":4742,"manufName":"SONOFF","powerSource":"Unknown","modelId":"DONGLE-E_R","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"inClusterList":[0,3,4,5,6,8],"outClusterList":[],"clusters":{"genLevelCtrl":{"attributes":{"currentLevel":51}},"genBasic":{"attributes":{"modelId":"DONGLE-E_R","manufacturerName":"SONOFF","powerSource":0,"zclVersion":8,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20220515","swBuildId":"1.0.0"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20220515","swBuildId":"1.0.0","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1662495458359,"defaultSendRequestWhen":"immediate"}

### Comments

I am kinda stuck with what has to be done, I have tried the below converter but seems like its throwing some kinda error with exposes lib. 

### External converter

```shell
I have tried some thing like below for

const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const reporting = require('../lib/reporting');
const extend = require('../lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['UT-02'],
        model: 'EFR32MG21-USB',
        vendor: 'Silicon Labs',
        description: 'EFR32MG21 Silabs USB Router',
        fromZigbee: [],
        toZigbee: [],
        exposes: [],
        configure: async (device, coordinatorEndpoint, logger) => {
            device.powerSource = 'Mains (single phase)';
            device.save();
        },		
    }
];
```
```

sujithg9, 2022-09-07:
For now, I got it to work. Not sure if that's all I need to do. 

[image]
[image]

Koenkk, 2022-09-07:
Change fromZigbee to `fromZigbee: [fz.linkquality_from_basic],`, after that can you make a merge request?

sujithg9, 2022-09-07:
@Koenkk 
Yeah sure I can do that .. but can you suggest to me how to handle the below
```
Debug 2022-09-07 13:00:40No converter available for 'ZBDongle-E' with cluster 'genLevelCtrl' and type 'attributeReport' and data '{"currentLevel":51}'
```

sujithg9, 2022-09-08:
@Koenkk Below is the cluster structure I see for these adapters .. can we enable touchlink somehow as we have oot for the texas instruments adapter ??

[image]

I might be dumb with all these questions as I am new to the world of Zigbee.

Koenkk, 2022-09-08:
> Below is the cluster structure I see for these adapters .. can we enable touchlink somehow as we have oot for the texas instruments adapter ??

It is not a feature we can enable over Zigbee itself, so if possible, this has to be done directly from the device by pressing a button on it (like the IKEA and Hue remotes)

Regarding the exposes, use `exposes: [exposes.numeric('light_indicator_level').withDescription('Brightness of the indicator light')]` and add the top of the file add `const exposes = require('zigbee-herdsman-converters/lib/exposes');`

sujithg9, 2022-09-08:
@Koenkk I have opened a PR for this change under converters repo 
ref: https://github.com/Koenkk/zigbee-herdsman-converters/pull/4641

Koenkk, 2022-09-09:
Great, can you also make a pr for the docs with the link to the fw? Adding it to the description makes it too long.

sujithg9, 2022-09-09:
@Koenkk yeah I understand, let me open up a PR for the docs to be updated with the firmware info in there. Thanks

Hedda, 2022-09-16:
> @Koenkk yeah I understand, let me open up a PR for the docs to be updated with the firmware info in there. Thanks

@sujithg9 Nice! See your converter PR for "ZBDongle-E" with Zigbee Router device firmware was added to [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters) 7-days ago, so likely Zigbee2MQTT ”Edge” (dev branch) will support it now but not in the master branch (mainline) or any stable releases as yet.

https://github.com/Koenkk/zigbee2mqtt/issues/13894

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4641

https://github.com/Koenkk/zigbee-herdsman-converters/tags

https://github.com/Koenkk/zigbee2mqtt/commits/dev

https://github.com/Koenkk/zigbee2mqtt/commits/master

https://github.com/Koenkk/zigbee2mqtt/releases

miksen, 2022-09-26:
> yeah I understand, let me open up a PR for the docs to be updated with the firmware info in there. 

Hi @sujithg9 did you get a chance to update docs? I looked [here](https://github.com/Koenkk/zigbee2mqtt.io/commits/master/docs/guide/adapters/README.md), and don't see a change, but I could be looking in the wrong place...

sujithg9, 2022-09-27:
@miksen sry I dint get a chance to update the docs. Have been pretty busy at work this month ☹️

Will get to it may be end of this week.

Hedda, 2022-09-27:
> Hi @sujithg9 did you get a chance to update docs? I looked [here](https://github.com/Koenkk/zigbee2mqtt.io/commits/master/docs/guide/adapters/README.md), and don't see a change, but I could be looking in the wrong place...

Please note that the https://www.zigbee2mqtt.io/guide/adapters / https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/guide/adapters/README.md document should probably not be updated when using ZBDongle-E with Router Firmware as it is working as a router device and not as a Zigbee Coordinator.

See https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_4-add-device-picture-to-zigbee2mqtt-io-documentation

So instead, you might want to consider adding a new device document instead for ZBDongle-E with Router Firmware -> https://www.zigbee2mqtt.io/supported-devices/ which is under -> https://github.com/Koenkk/zigbee2mqtt.io/tree/master/docs/devices

Hedda, 2022-09-27:
> It is not a feature we can enable over Zigbee itself, so if possible, this has to be done directly from the device by pressing a button on it (like the IKEA and Hue remotes)

@sujithg9 FYI, not sure if it will reach them but I have posted this request to ITead for "touchlink" (touchlinking) support in new/future Zigbee Router firmware images for Dongle-E -> https://github.com/itead/Sonoff_Zigbee_Dongle_Firmware/issues/7

Hedda, 2022-09-27:
By the way, FYI, there's another cheap and relatively popular EFR32MG21 based USB adapter that also has Zigbee Router firmware.

ZB-GW04 USB stick (a.k.a. "easyiot Zigbee dongle") based on "SM-011 V1.0" (ZYZBP008) radio module from CoolKit-Technologies:

https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/Zigbee3.0_Dongle-NoSigned/RouterForDongle

They sell it in two different variants but the only difference is if it comes pre-flashed with Zigbee Router firmware or EZSP Zigbee Coordinator firmware, and interestingly enough they sell it for a lot less when they ship the variant pre-flashed with Zigbee Router firmware:

https://easyiot.aliexpress.com/store/5839056
https://www.aliexpress.com/item/1005003683621008.html
https://www.aliexpress.com/item/1005002791666029.html
https://www.aliexpress.com/item/1005003736123654.html
https://www.aliexpress.com/item/1005003692212448.html
https://www.aliexpress.com/item/1005003578599189.html
https://www.aliexpress.com/item/1005003814709714.html
https://www.aliexpress.com/item/1005004562353633.html
https://www.aliexpress.com/item/1005003501898658.html
https://www.aliexpress.com/item/1005004410926709.html
https://www.aliexpress.com/item/1005002469947910.html

[image]

[image]

[image]

Note! That ZB-GW04/ZYZBP008 USB adapter might work well with Zigbee Router firmware as a Zigbee repeater for range extending but personally, I would not recommend that dongle as a Zigbee Coordinator because it has previously been proven that the antenna design it uses is not optimal which makes it extremely sensitive to all types of EMF/EMI/RMI signal interference.

PS: That "SM-011 V1.0" (ZYZBP008) radio module from CoolKit-Technologies is the same used in the original ITead Sonoff ZBBridge Zigbee gateway as well as unbranded ZB-GW03 eWeLink Ethernet Zigbee Gateway (also sold rebranded under EACHEN and SmartWise brands + others), and also a Modkam DIY USB dongle.

https://www.coolkit.cn/product/sm-011/

https://github.com/CoolKit-Technologies/DevDocs/blob/master/Zigbee/SM-011%E5%BA%94%E7%94%A8%E6%8C%87%E5%AF%BC%E4%B9%A6.md

https://community.home-assistant.io/t/sonoff-zbbridge-sonoff-zigbee-bridge-from-itead/187346

https://community.home-assistant.io/t/zb-gw03-ewelink-ethernet-zigbee-gateway-now-hacked-with-tasmota-zbbridge-so-can-be-used-via-mqtt-or-as-a-remote-zigbee-adapter-with-home-assistant-zha/341223

https://modkam.ru/?p=1889

closed 2022-11-04 by github-actions[bot]