Koenkk/zigbee2mqtt #13896: [New device support]: Ynoa Smart Plug Zigbee3.0 LA-PLUG-10Amp

HertogArjan, 2022-09-06:
### Link

https://www.ledlampenkopen.nu/ynoa-slimme-stekker-zigbee-3-0-met-energiemeter.html

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0x847127fffe1a200d","nwkAddr":61769,"manufId":4456,"manufName":"LDS","powerSource":"Mains (single phase)","modelId":"ZBT-ONOFFPlug-D0009","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,8,1794,2820,2821,4096,64642],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"ZBT-ONOFFPlug-D0009","manufacturerName":"LDS","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200616","swBuildId":"1.00.08"}},"genOnOff":{"attributes":{"onOff":1}},"haElectricalMeasurement":{"attributes":{"activePower":2,"rmsCurrent":0,"rmsVoltage":0}},"genLevelCtrl":{"attributes":{"currentLevel":255}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002390dd9b","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b002390dd9b","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b002390dd9b","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002390dd9b","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200616","swBuildId":"1.00.08","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1021942205},"lastSeen":1662498669164,"defaultSendRequestWhen":"immediate"}

### Comments

I bought a bunch of these Ynoa Smart Plugs and wish to add support for them. I can't seem to find any website of the device manufacturer, so I linked the shop where I bought them. 

Currently what works well is switching (it reports the new state when I switch the device on/off with the button and I can switch it on/off remotely) and power measurement. Current and voltage seem to be reported as 0 after I plug in the device and then are never updated again. Same with 'currentSummDelivered', it starts at zero and is never updated. I tried to configure 'acFrequency' but it caused an 'UNKNOWN_ATTRIBUTE' error.

Is there anything else I can try or does this mean power measurement is as good as it is going to get?

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
//const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
//const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZBT-ONOFFPlug-D0009'],
    model: 'LA-PLUG-10Amp',
    vendor: 'LDS',
    description: 'Ynoa Smart Plug ZigBee3.0',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering],
    toZigbee: [tz.on_off],
    exposes: [e.switch(),e.energy(),e.power(),e.voltage(),e.current()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.onOff(endpoint);
        await reporting.activePower(endpoint);
        await reporting.rmsCurrent(endpoint);
        await reporting.rmsVoltage(endpoint);
        await reporting.currentSummDelivered(endpoint);
        //await reporting.acFrequency(endpoint);
    },
};

module.exports = definition;
```

Koenkk, 2022-09-07:
> Current and voltage seem to be reported as 0 after I plug in the device and then are never updated again. Same with 'currentSummDelivered', it starts at zero and is never updated. I tried to configure 'acFrequency' but it caused an 'UNKNOWN_ATTRIBUTE' error.

Sounds like the device doesn't support this, so this seems to be as good as it gets.

HertogArjan, 2022-09-07:
Should I add this definition to lds.js or ynoa.js? The firmware reports vendor LDS but the box says Ynoa.

Koenkk, 2022-09-08:
Ynoa (since that is one the box and the site you linked also mentions Ynoa)

HertogArjan, 2022-09-11:
https://github.com/Koenkk/zigbee2mqtt.io/pull/1566
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4650

closed 2022-09-12 by Koenkk