Koenkk/zigbee2mqtt #13903: Xiaomi Aqara Plug (SP-EUC01/lumi.plug.maeu01) Toggles without reason

gszigethy, 2022-09-07:
### What happened?

I've executed OTA firmware update on the Xiaomi Aqara plug (SP-EUC01). Old fw version: 09-22-2020. New version came from zigb222mqtt OTA location, version 20211206.

```
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: lumi.plug.maeu01: unknown key dateCode with value 09-22-2020
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: lumi.plug.maeu01: unknown key swBuildId with value undefined
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: lumi.plug.maeu01: Processed data into payload {}
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: Updating to latest '0x54ef441000157f54' (lumi.plug.maeu01)
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: Using endpoint '1'
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: Received Zigbee message from 'plug_002', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":32,"imageType":24,"manufacturerCode":4447}' from endpoint 1 with groupID 0
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: Got OTA request '{"fieldControl":0,"manufacturerCode":4447,"imageType":24,"fileVersion":32}'
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: ZigbeeOTA: downloaded main index
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: getNewImage for '0x54ef441000157f54', meta {"fileVersion":41,"fileSize":273278,"url":"https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20211209165104_OTA_lumi.plug.maeu01_0.0.0_0041_20211206_0C22EC.ota","sha512":"3d5567a29eb7d4bacaaf823c47b66a700fb1e7b935b1d547673bcacb2ff4efacb7780e8caf72dc80f0ac942c1ec1f6964bf67cfacba359d38132b7bcfe80133b"}
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:30: ZigbeeOTA: downloading firmware image from https://github.com/Koenkk/zigbee-OTA/raw/master/images/Xiaomi/20211209165104_OTA_lumi.plug.maeu01_0.0.0_0041_20211206_0C22EC.ota
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:31: OTA update checksum validation succeeded for '0x54ef441000157f54'
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:31: getNewImage for '0x54ef441000157f54', image header {"otaUpgradeFileIdentifier":{"type":"Buffer","data":[30,241,238,11]},"otaHeaderVersion":256,"otaHeaderLength":56,"otaHeaderFieldControl":0,"manufacturerCode":4447,"imageType":24,"fileVersion":41,"zigbeeStackVersion":2,"otaHeaderString":"ROUTERX-----JN5180--ENCRYPTED000","totalImageSize":273278}
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:31: Got new image for '0x54ef441000157f54'
2022-09-01.13-13-03/log.txt:debug 2022-09-07 08:54:31: Starting upgrade
```
The upgrade completed without problems.

After that update, whenever I press the button on any IKEA E1812 or E1743, the plug toggles its state which is really undesired. It happens only when the button is connected to that plug (routed via it). Repairing the button using another router helps for a moment but as the network reconfigures itself, the buttons are again routed via the plug, so random buttons start to turn that router plug on and off. Since that plug is not solely a router but provides power to my home office, this is very annoying and made the device unusable. It didn't brick it, but the functionality became useless.

This is in the logs when I press an unrelated button routed through the plug, and switching it:

```
debug 2022-09-07 16:09:48: Received Zigbee message from 'switch_006', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2022-09-07 16:09:48: MQTT publish: topic 'zigbee2mqtt/switch_006', payload '{"action":"on","battery":87,"linkquality":47,"update":{"state":"idle"},"update_available":null}'
info  2022-09-07 16:09:48: MQTT publish: topic 'zigbee2mqtt/switch_006', payload '{"action":"","battery":87,"linkquality":47,"update":{"state":"idle"},"update_available":null}'
info  2022-09-07 16:09:48: MQTT publish: topic 'zigbee2mqtt/switch_006/action', payload 'on'
debug 2022-09-07 16:09:48: Received Zigbee message from '0x54ef441000157f54', type 'attributeReport', cluster 'genOnOff', data '{"245":128283218,"onOff":1}' from endpoint 1 with groupID 0
info  2022-09-07 16:09:48: MQTT publish: topic 'zigbee2mqtt/0x54ef441000157f54', payload '{"consumer_connected":false,"consumption":0,"current":0,"device_temperature":27,"energy":0,"linkquality":47,"power":0,"power_outage_count":0,"power_outage_memory":true,"state":"ON","update":{"state":null},"update_available":null,"voltage":229}'
```

I accidentally realized that these ghost switches work when zigbee2mqtt is down, ie. the plug works in a binding manner with all switching devices that are routed through it. In the meantime sensors routed through it work fine. It seems like when the firmware got upgraded to the one on the OTA server, it makes the plug and all routed switches like binded devices. It may be that the firmware is corrupt, but I have found no way to roll back. I've tried to unbind the switches that are routed through the plug via zigbee2mqtt GUI. The unbind gives a success message but the faulty behaviour remains.

Apparently happened to ohers but not too much traffic in that thread: https://community.home-assistant.io/t/aqara-plug-sp-euc01-lumi-plug-maeu01-toggles-without-reason/451671/2

### What did you expect to happen?

After the firmware upgrade the device behaviour is identical and can be switched by binded devices or via zigbee2mqtt.

### How to reproduce it (minimal and precise)

1. Get a Xiaomi Aquara plug with original firmware (mine was 09-22-2020).
2. OTA upgrade via zigbee2mqtt.
3. Make a few switches route through it in a zigbee network. I used the following:
    IKEA | E1812 | 20210727 | 2.3.080
    IKEA | E1743 | 20210722 | 2.3.079
5. Toggle the switch.
6. Observe that it will toggle the status of the plug.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

Slaesh CC2652RB

0ip, 2022-09-08:
I can confirm and reproduce that behavior. My plug has the same firmware. With my E1743, I can reliably turn on or off the plug.

Koenkk, 2022-09-08:
Seems the plug started listening to group 0, can you try to make a group 0 in z2m, add the plug to it and then remove it again?

gszigethy, 2022-09-08:
Creating group:

```
debug 2022-09-08 09:50:33: Received MQTT message on 'zigbee2mqtt/bridge/request/group/add' with data '{"friendly_name":"plugtest","id":"0","transaction":"5x1m7-2"}'
info  2022-09-08 09:50:34: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/add', payload '{"data":{"friendly_name":"plugtest","id":0},"status":"ok","transaction":"5x1m7-2"}'
debug 2022-09-08 09:50:34: Received Zigbee message from 'rain_001', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3165,"10":24590,"100":0,"3":26,"4":13224,"5":12,"6":[0,65537]}}' from endpoint 1 with groupID 0
```

Adding the plug to the group (plug is 0x54ef441000157f54):

```
debug 2022-09-08 09:51:31: Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/add' with data '{"device":"0x54ef441000157f54/1","group":"plugtest","transaction":"65o9i-26"}'
info  2022-09-08 09:51:31: Adding '0x54ef441000157f54' to 'plugtest'
info  2022-09-08 09:51:31: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x54ef441000157f54/1","group":"plugtest"},"status":"ok","transaction":"65o9i-26"}'
info  2022-09-08 09:51:31: MQTT publish: topic 'homeassistant/switch/1221051039810110150109113116116_0/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/plugtest/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/plugtest/set","device":{"identifiers":["zigbee2mqtt_1221051039810110150109113116116_0"],"name":"plugtest","sw_version":"Zigbee2MQTT 1.27.2"},"name":"plugtest","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/plugtest","unique_id":"0_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
debug 2022-09-08 09:51:32: Received MQTT message on 'homeassistant/switch/1221051039810110150109113116116_0/switch/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/plugtest/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/plugtest/set","device":{"identifiers":["zigbee2mqtt_1221051039810110150109113116116_0"],"name":"plugtest","sw_version":"Zigbee2MQTT 1.27.2"},"name":"plugtest","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/plugtest","unique_id":"0_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
```

Testing the switch off and on (switch is 0x8cf681fffe2d74ae, plus ig 0x54ef441000157f54):

```
debug 2022-09-08 09:51:58: Received Zigbee message from '0x8cf681fffe2d74ae', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2022-09-08 09:51:58: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe2d74ae', payload '{"action":"off","battery":87,"linkquality":51,"update":{"state":"idle"},"update_available":null}'
info  2022-09-08 09:51:58: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe2d74ae', payload '{"action":"","battery":87,"linkquality":51,"update":{"state":"idle"},"update_available":null}'
info  2022-09-08 09:51:58: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe2d74ae/action', payload 'off'
debug 2022-09-08 09:51:58: Received Zigbee message from '0x54ef441000157f54', type 'attributeReport', cluster 'genOnOff', data '{"245":125989379,"onOff":0}' from endpoint 1 with groupID 0
info  2022-09-08 09:51:58: MQTT publish: topic 'zigbee2mqtt/plugtest', payload '{"state":"OFF"}'
info  2022-09-08 09:51:58: MQTT publish: topic 'zigbee2mqtt/0x54ef441000157f54', payload '{"consumer_connected":false,"consumption":0,"current":0,"device_temperature":29,"energy":0,"linkquality":51,"power":0,"power_outage_count":0,"power_outage_memory":true,"state":"OFF","update":{"state":null},"update_available":null,"voltage":233}'
debug 2022-09-08 09:51:58: Received Zigbee message from '0x54ef441000157f54', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 21 with groupID 0
info  2022-09-08 09:51:58: MQTT publish: topic 'zigbee2mqtt/0x54ef441000157f54', payload '{"consumer_connected":false,"consumption":0,"current":0,"device_temperature":29,"energy":0,"linkquality":51,"power":0,"power_outage_count":0,"power_outage_memory":true,"state":"OFF","update":{"state":null},"update_available":null,"voltage":233}'

debug 2022-09-08 09:52:01: Received Zigbee message from '0x8cf681fffe2d74ae', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2022-09-08 09:52:01: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe2d74ae', payload '{"action":"on","battery":87,"linkquality":51,"update":{"state":"idle"},"update_available":null}'
info  2022-09-08 09:52:01: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe2d74ae', payload '{"action":"","battery":87,"linkquality":51,"update":{"state":"idle"},"update_available":null}'
info  2022-09-08 09:52:01: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe2d74ae/action', payload 'on'
debug 2022-09-08 09:52:01: Received Zigbee message from '0x54ef441000157f54', type 'attributeReport', cluster 'genOnOff', data '{"245":125989380,"onOff":1}' from endpoint 1 with groupID 0
info  2022-09-08 09:52:01: MQTT publish: topic 'zigbee2mqtt/plugtest', payload '{"state":"ON"}'
info  2022-09-08 09:52:01: MQTT publish: topic 'zigbee2mqtt/0x54ef441000157f54', payload '{"consumer_connected":false,"consumption":0,"current":0,"device_temperature":29,"energy":0,"linkquality":51,"power":0,"power_outage_count":0,"power_outage_memory":true,"state":"ON","update":{"state":null},"update_available":null,"voltage":233}'
[... cut ...]
debug 2022-09-08 09:56:43: Received Zigbee message from '0x54ef441000157f54', type 'attributeReport', cluster 'genOnOff', data '{"245":125989382,"onOff":1}' from endpoint 1 with groupID 0
info  2022-09-08 09:56:43: MQTT publish: topic 'zigbee2mqtt/0x54ef441000157f54', payload '{"consumer_connected":false,"consumption":0,"current":0,"device_temperature":29,"energy":0,"linkquality":47,"power":0,"power_outage_count":0,"power_outage_memory":true,"state":"ON","update":{"state":null},"update_available":null,"voltage":233}'
```

The behaviour of the plug is still the same, it switches based on switch commands only routed through it.

Koenkk, 2022-09-08:
That's strange, look that the fw is buggy since it shouldn't respond to that command. Can you try to bind the ikea remote to group 0, unbind it and then bind it to another group (random, e.g. 128). You should bind the `onOff` cluster.

max-koptev, 2022-09-08:
I have the same problem with LLKZMK11LM relay on firmware 9-22-2021. IKEA remote E1524/E1810 toggles the relay switch at Endpoint: l1 when it routed through it. Seems like the problem is in Aqara devices...

gszigethy, 2022-09-08:
What I also noticed since the firmware upgrade is the SP-EUC01 plug creates these kind of errors in the log:

Zigbee2MQTT:debug 2022-09-08 14:09:14: lumi.plug.maeu01: unknown key 220 with value 㡬*`r���

It is written in the log as is, with these special characters.

gszigethy, 2022-09-08:
> That's strange, look that the fw is buggy since it shouldn't respond to that command. Can you try to bind the ikea remote to group 0, unbind it and then bind it to another group (random, e.g. 128). You should bind the `onOff` cluster.

When I want to bind the Ikea switch to group 0, the button is not active:

[image]

I can unbind it on the GUI, then I press a button on the Ikea switch to wake it up, then the GUI says it was successful with the green popup and I see it in the logs.

```
Zigbee2MQTT:debug 2022-09-08 14:38:43: Received MQTT message on 'zigbee2mqtt/bridge/request/device/unbind' with data '{"clusters":["genOnOff"],"from":"0x8cf681fffe2d74ae/1","to":"plugtest","transaction":"m7n59-5"}'
Zigbee2MQTT:debug 2022-09-08 14:38:43: unbinding cluster 'genOnOff' from '0x8cf681fffe2d74ae' to 'plugtest'
Zigbee2MQTT:debug 2022-09-08 14:38:48: Received Zigbee message from '0x8cf681fffe2d74ae', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-09-08 14:38:48: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe2d74ae', payload '{"action":"off","battery":87,"linkquality":32,"update":{"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2022-09-08 14:38:48: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe2d74ae', payload '{"action":"","battery":87,"linkquality":32,"update":{"state":"idle"},"update_available":null}'
Zigbee2MQTT:info  2022-09-08 14:38:48: MQTT publish: topic 'zigbee2mqtt/0x8cf681fffe2d74ae/action', payload 'off'
Zigbee2MQTT:debug 2022-09-08 14:38:48: Received Zigbee message from '0x54ef441000157f54', type 'attributeReport', cluster 'genOnOff', data '{"245":125989476,"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-09-08 14:38:48: MQTT publish: topic 'zigbee2mqtt/0x54ef441000157f54', payload '{"consumer_connected":false,"consumption":0,"current":0,"device_temperature":27,"energy":0,"linkquality":32,"power":0,"power_outage_count":0,"power_outage_memory":true,"state":"OFF","update":{"state":null},"update_available":null,"voltage":233}'
Zigbee2MQTT:info  2022-09-08 14:38:49: **Successfully unbound cluster 'genOnOff' from '0x8cf681fffe2d74ae' to 'plugtest'**
Zigbee2MQTT:info  2022-09-08 14:38:49: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/unbind', payload '{"data":{"clusters":["genOnOff"],"failed":[],"from":"0x8cf681fffe2d74ae/1","to":"plugtest"},"status":"ok","transaction":"m7n59-5"}'
```

After this the plug still responds to the switch commands if the switch is routed via the plug.

I stronlgy suspect this is the results of the firmware, is there a copy of the old version, can the device be rolled back?

Koenkk, 2022-09-09:
@otnow given the bugs in this fw, can we put back the old ones?

Otnow, 2022-09-09:
> @Otnow given the bugs in this fw, can we put back the old ones?

It is a pity that this firmware version (`0.0.0_0041`) causes such problems.

Created a [PR](https://github.com/Koenkk/zigbee-OTA/pull/159) to return the previous version (`0.0.0_0032`): [OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946.ota](https://github.com/Koenkk/zigbee-OTA/raw/91de0ee565eb5d2bb4d1d92fcef4009c4226f846/images/Xiaomi/OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946.ota)

TheJulianJES, 2022-09-10:
Only saw this issue because of the updated OTA files:

This issue is fixed in ZHA by just making the plug leave group 0: https://github.com/zigpy/zha-device-handlers/pull/1706/files
Z2M could probably implement a similar fix.

(The issue is that the plug is in group 0 when paired initially and IKEA remotes also need to be in group 0. The plug can be removed from that group though.)

johannes-z, 2022-09-12:
I have the same issue.

I was unable to make the plug leave group 0, unbind the OnOff cluster, etc. Rolling back the firmware also doesn't work as the device prevents it (tried via Local OTA index).

Is there **any** way to at least temporarily fix this until a new firmware is released?

TheJulianJES, 2022-09-12:
WARNING: Do this at your own risk. You could fully **brick** your device.

<details>
  <summary>Instructions on how to downgrade to v32 by bypassing the file version check</summary>
​

The Aqara MAEU01 plug only checks that the header file version of the provided OTA image is bigger than the one currently installed.
To bypass this check, here's a modified image of v32 (which shouldn't have the bug) with header file version 42:
[OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v42.ota.zip](https://github.com/Koenkk/zigbee2mqtt/files/9552415/OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v42.ota.zip)
(Make sure to extract the image from the zip)

Using zigpy/zigpy-cli, this image was created by doing the following:
  
  ```python
import pathlib
from zigpy.ota.image import parse_ota_image

data = pathlib.Path('OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946.ota').read_bytes()
image, remaining = parse_ota_image(data)
assert not remaining

image.header.file_version = 42

pathlib.Path('OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v42.ota').write_bytes(image.serialize())

  ```

You should be able to use that image to downgrade your plug to v32 from v41. If you're lucky, the device doesn't brick itself.
(I didn't have any issues downgrading using that image with ZHA and my plug but no guarantees it'll work for you).

I did have to re-activate pairing mode on my MAEU01 plug for some reason though. (The device didn't actually re-pair itself to the network though.)

You can confirm the downgrade was successful by checking that ``app_version`` on the ``Basic`` cluster returns ``32`` (instead of ``41``).
(This won't return ``42``, as we didn't actually modify the OTA image content but just the header. You should be able to upgrade to v41 again if you want.)

If you also try this, please let me know how it went.

</details>

@Otnow Do you know if there are any firmware versions for that plug between v32 and v41?

Otnow, 2022-09-13:
> Instructions on how to downgrade to v32 by bypassing the file version check

Excellent. I hope this will help others to roll back from the problematic firmware version.

---

> Do you know if there are any firmware versions for that plug between v32 and v41?

It seems that there is no judging by the information from the [Aqara Developer Platform](https://developer.aqara.com/console/):

```json
{
  "result": [
    {
      "releaseTime": "1642585695000",
      "necessary": 0,
      "fileSize": 273278,
      "updateLog": "1.Fix the offline problem of device",
      "state": 2,
      "firmwareVersion": "0.0.0_0041"
    },
    {
      "releaseTime": "1603976430000",
      "necessary": 0,
      "fileSize": 278094,
      "updateLog": "1.Fixes known issues",
      "state": 2,
      "firmwareVersion": "0.0.0_0032"
    }
  ]
}
```

0ip, 2022-09-13:
@TheJulianJES It worked for me, thanks for the tutorial and the image!

johannes-z, 2022-09-14:
@TheJulianJES Thanks, worked for me too! 

I used Z2M. In case anybody wonders how to do it: 
* Download the modified firmware, unzip, upload into `zigbee2mqtt` folder.
* Add `index.json` into your `zigbee2mqtt/` folder (next to z2m's `configuration.yaml`).
* Update the `index.json` file contents:
```
[
  {
    "modelId": "lumi.plug.maeu01",
    "url": "OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v42.ota",
    "path": "./OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v42.ota",
    "force": true
  }
]
```
* In Zigbee2MQTT go to `Settings > OTA updates` and add `index.json` to the `OTA index override file name` field.
* Restart Z2M Addon
* Check for OTA updates and install

0ip, 2022-09-15:
I suppose until we have a solution, perhaps a warning about this problem should appear on the documentation page of this plug along with a link to the discussion here. What do you think?

greiseknoue, 2022-09-27:
@TheJulianJES Hi! Does this also work for the lumi.plug.mmeu01 or do i need another firmware? (Model: ZNCZ04LM Firmware: 01-23-2022)

TheJulianJES, 2022-09-27:
@greiseknoue Does the mmeu01 also have the problem with toggling randomly on the 2022 firmware? Or what exactly are you trying to achieve?

greiseknoue, 2022-09-27:
@TheJulianJES Yes i have the same issue where the E2001/E2002 and E1812 toggles the plug. I have three mmeu01 plugs where the one i updated has this issue. The firmware build on the non-updated plugs are on 09-06-2019.

in one case, i randomly got  E2001/E2002 to work only to the coordinator (I noticed it didn't route through the plug in the "map"-section").

TheJulianJES, 2022-09-27:
NOTE: This is only for the Xiaomi ``mmeu01`` (square EU plug). **NOT** THE ROUND EU PLUG.
For the round Aqara EU plug, see the instructions here: https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072

NOTE: I haven't tried this firmware at all and it could **brick** your device. It's the v22 firmware file modded to have a v25 header (so the plug will accept "upgrading" to it from the apparently buggy v24 release).
In about 10 hours, I might be able to to try the downgrade for an ``mmeu01`` plug with that firmware.

For the ``mmeu01`` plug, you can try the following firmware:
[OTA_lumi.plug.mmeu01_V22_20190906_D32362.20191008105750_modded_to_v25.ota.zip](https://github.com/Koenkk/zigbee2mqtt/files/9654652/OTA_lumi.plug.mmeu01_V22_20190906_D32362.20191008105750_modded_to_v25.ota.zip)
(make sure to extract the image from the ZIP)

The instructions for installing that firmware should be basically identical to the ones in https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072.

greiseknoue, 2022-09-27:
Thank you so much for your help. I think i got it working, but it still says "Firmware build date 01-23-2022" in the ui of the device. I'm guessing i should look somewhere else to find the right firmware?

TheJulianJES, 2022-09-27:
Mhm, ``date_code`` should be ``09-06-2019`` and ``app_version=22`` (both are attributes on the Basic cluster).
Maybe Z2M caches the info for some time. Try to re-read it or re-pair the plug maybe?
(I'll try to see later if the firmware downgrade for the mmeu01 plug works for me and see if I have the IKEA random toggling issue for that plug as well before/after)

greiseknoue, 2022-09-27:
After removing the plug from the wall and plugging it in, it works and it says 09-06-2019! Thank you once again! I will report back if the issue still persist.

lo97, 2022-10-08:
How to disable ota if the device is already on version 32?

TheJulianJES, 2022-10-08:
@lo97 Are you using ZHA or Z2M?

For ZHA, just remove the modified OTA file from the ota directory.
For Z2M, just remove the OTA index override.

vetinari, 2022-10-12:
> > Do you know if there are any firmware versions for that plug between v32 and v41?
> 
> It seems that there is no judging by the information from the [Aqara Developer Platform](https://developer.aqara.com/console/):
There may be one, mine has `Firmware build date
12-06-2021`

uxdesignerhector, 2022-10-14:
> @TheJulianJES Thanks, worked for me too!
>
> I used Z2M. In case anybody wonders how to do it:
>
>     * Download the modified firmware, unzip, upload into `zigbee2mqtt` folder.
>
>     * Add `index.json` [...]

I'm trying to find that option in the step (go to `Settings > OTA updates`...) but I don't see it, could you show me the way? I'm using the addon, thanks
[image]

johannes-z, 2022-10-14:
@uxdesignerhector Try adding the config in your `configuration.yaml` for Z2M:

```yaml
ota:
  disable_automatic_update_check: true
  ikea_ota_use_test_url: false
  zigbee_ota_override_index_location: index.json
```

uxdesignerhector, 2022-10-15:
> @uxdesignerhector Try adding the config in your `configuration.yaml` for Z2M:
> 
> ```yaml
> ...
> permit_join: false
> devices:
>   ...
> ota:
>   disable_automatic_update_check: true
>   ikea_ota_use_test_url: false
>   zigbee_ota_override_index_location: index.json
> ```

Solved! I was on 1.18.1 old deprecated repo, no wonder I was no receiving any update :-) can confirm that this smart plug 

- (SP-EUC01/lumi.plug.maeu01)
- Firmware build date : 12-06-2021

which a bought one week ago works flawlessly with current updated repo version 1.28.0 no need to downgrade!

Thought there will be some kind of warning when a repo is deprecated in home-assistant, maybe there are others stuck in 1.18.1 like me without noticing it. Thank you for helping me

rotilho, 2022-10-23:
I created a ticket in their website. If someone want to put some pressure in a real fix from Xiaomi, please submit a ticket as well.

https://static-resource.aqara.com/html/feedback.html

mSapik, 2022-10-27:
> NOTE: This is only for the Xiaomi `mmeu01` (square EU plug). **NOT** THE ROUND EU PLUG. For the round Aqara EU plug, see the instructions here: [#13903 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072)
> 
> NOTE: I haven't tried this firmware at all and it could **brick** your device. It's the v22 firmware file modded to have a v25 header (so the plug will accept "upgrading" to it from the apparently buggy v24 release). In about 10 hours, I might be able to to try the downgrade for an `mmeu01` plug with that firmware.
> 
> For the `mmeu01` plug, you can try the following firmware: [OTA_lumi.plug.mmeu01_V22_20190906_D32362.20191008105750_modded_to_v25.ota.zip](https://github.com/Koenkk/zigbee2mqtt/files/9654652/OTA_lumi.plug.mmeu01_V22_20190906_D32362.20191008105750_modded_to_v25.ota.zip) (make sure to extract the image from the ZIP)
> 
> The instructions for installing that firmware should be basically identical to the ones in [#13903 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072).

Thank you so much! Return to life my plug.

closed 2023-01-14 by github-actions[bot]

przemekgithub, 2023-01-23:
Hi, 
I have problem with downgrading:
Zigbee2Mqtt 1.29.2
One of my plugs has firmware date 12-06-2021, and it returns app version 41
2023-01-23 14:25:01Read result of 'genBasic': {"appVersion":41}

OTA logs:
Info 2023-01-23 14:23:20Checking if update available for 'AAA'
Info 2023-01-23 14:23:20No update available for 'AAA'
Info 2023-01-23 14:23:20MQTT publish: topic 'zigbee2mqtt_lan/AAA', payload '{"consumer_connected":true,"consumption":2.99,"current":0,"detection_period":15,"device_temperature":9,"energy":2.99,"linkquality":123,"power":0,"power_outage_count":5,"power_outage_memory":true,"state":"OFF","update":{"state":"idle"},"voltage":225}'
Info 2023-01-23 14:23:20MQTT publish: topic 'zigbee2mqtt_lan/bridge/response/device/ota_update/check', payload '{"data":{"id":"AAA","updateAvailable":false},"status":"ok","transaction":"kngsq-2"}'

So the image attached to this topic with v42 is not recognized as something newer...

P.S. i wrote about this bug to aqara but got answer:
"We do not support third-party software or hardware for the time being. The Aqara devices can be paired with the Aqara hubs. 

When Matter is introduced in the future, it will allow more devices from different brands to communicate."

paulz-gh, 2023-01-30:
Hi, 
Looks like V43 firmware was released today according to https://www.aqara.com/eu/version/smart-plug-eu
Any chance of putting this in the OTA list so I can try if it fixes this problem ?

TheJulianJES, 2023-01-30:
@Otnow If you have some time, can you send the link for the MAEU01 update here, so I can test if the issue is fixed (before putting it in the zigbee-ota repo)?

(And it would be amazing if you could also check if there's an update for the MMEU01 which had the same issue on newer firmware versions)

Otnow, 2023-01-30:
> @Otnow If you have some time, can you send the link for the MAEU01 update here, so I can test if the issue is fixed (before putting it on the zigbee-ota repo)?
> 
> (And it would be amazing if you could also check if there's an update for the MMEU01 which had the same issue on newer firmware versions)

I don't see any updates for these devices in the Aqara Home app yet, but I'm keeping an eye on it.

As far as I understand, Aqara distributes updates gradually and may become available for someone earlier, and for someone later.

I'll let you know as soon as I get it, or maybe someone else will get it sooner.

konstantinozh, 2023-02-12:
Has anyone tried the V43 firmware to see if they've fixed the issue? I'm about to order one of these, but might look for alternatives if it keeps being a problem.

Otnow, 2023-02-28:
@TheJulianJES, @paulz-gh, @konstantinozh, firmware version `0.0.0_0043` for `lumi.plug.maeu01` became available for download:

```json
"releaseDate": 1677553012000,
"part": "zigbee",
"downloadUrl": "https://cdn.aqara.com/cdn/opencloud-product/mainland/product-firmware/prd/lumi.plug.maeu01/20230129165424_OTA_lumi.plug.maeu01_0.0.0_0043_20230113_7E5AB5.ota",
"upgradeTakesTime": 0,
"updateLog": "Function optimization.",
"updateTime": 1677553012000,
"isNecessary": 0,
"fileSize": 277486,
"createTime": 1675060876000,
"firmwareMD5": "e2af34c841d425dbe80cb6e4517e5ab5",
"state": 3,
"isForce": false,
"firmwareVersion": "0.0.0_0043",
"imageType": "exe",
"firmwareCRC32": "72C4FC24"
```

swoopdk, 2023-03-01:
Tried to update to the firmware @Otnow posted without any success - still seeing the issue :( 

My procedure was as follows - download from the url above and insert this into my my_index.json file and update through the zigbee2mqtt ota interface. 

``` 
[
  {
    "modelId": "lumi.plug.maeu01",
    "url": "20230129165424_OTA_lumi.plug.maeu01_0.0.0_0043_20230113_7E5AB5.ota",
    "path": "./20230129165424_OTA_lumi.plug.maeu01_0.0.0_0043_20230113_7E5AB5.ota",
    "force": false
  }
]
``` 

Not sure if that was the wrong way to go about it though. I would love some help as i have two of these without any success now and can't use them since a random button push would toggle them :(

Otnow, 2023-03-01:
@swoopdk, it's a pity, but you can get rid of the random switching problem by [downgrading the firmware](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072).

swoopdk, 2023-03-01:
> @swoopdk, it's a pity, but you can get rid of the random switching problem by [downgrading the firmware](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072).

Perhaps i spoke too soon - i'll test for a weeks time. Not sure how to trigger the problem except try my ikea switches nearby the two plugs. Any ideas ? 

The thing is i decided i might as well try to completely remove the plug(s) and re-pair them. Had to head to my controller to get them re-paired but afterwards they are no longer turning on/off when i press the switch that was togging before. 

However should i need to downgrade from the 43 version i'll need to create a 32 version with the 43 header (or does it need to be newer?) pr the instructions here https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072 ?

TheJulianJES, 2023-03-01:
There's some code in my instructions to modify the OTA header (change the 42 to 44). You want to install zigpy and so on to run that.
If you can't get it to work, I can also create that file in a bit.
Do note that you could brick your device (although this hasn't happened yet), but you're running a newer version now.

Otnow, 2023-03-01:
@swoopdk, try this modified firmware (_version in header is 99_) to downgrade to version `32` (_as @TheJulianJES said there is a risk of bricking the device_):

[OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v99.ota.zip](https://github.com/Koenkk/zigbee2mqtt/files/10865431/OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v99.ota.zip)

swoopdk, 2023-03-03:
I'll try the latest one from you @Otnow this weekend when i carve out some time. However i have a question first (btw i'm apparently not friends with zigpy as i couldn't get it working do the procedure myself) - if we use a header version 99 any future updates are out of the question at least without flashing with a header above 99 right ? 
I have a tiny voice that is annoyed to have to kinda go this route though currently i have two plugs that are useless as i have several ikea triggers that would turn them off randomly hehe 

Is there any way of flashing the 32 firmware without using a modified header? 

Asking since i already now have the 43 header firmware on there so even creating a 44 firmware excludes the option of using a version 44 actual firmware in the future that might fix this (and possibly other bugs). So if it's at all possible i would love to go to the actual 32 firmware. 

And in regards to bricking i'm okay with the risk  :) It's a learning project in these two plugs mostly :)

rotilho, 2023-03-03:
Did anyone send a message to Xiaomi support? I sent it to support@aqara.com a while ago but I couldn't pass through the support.

I'll try again, but maybe if more people send emails, they may finally understand that this is a problem.

Note: I have several of those plugs, so I'm quite annoyed that such an issue exists for so long.

Otnow, 2023-03-03:
> I'll try the latest one from you @Otnow this weekend when i carve out some time. However i have a question first (btw i'm apparently not friends with zigpy as i couldn't get it working do the [...]

Unfortunately, without modifying the version number in the OTA file header, it will not be possible to downgrade the firmware version, but the firmware itself remains under the same version number (`32`, _not 99_) and it will be possible to upgrade from it to any other newer version (_>32_).

I deliberately changed the version number in the OTA file header to a large number (_99_) so that this file can be used to roll back to version `32` if future updates do not fix the current problem again.

konstantinozh, 2023-03-03:
> Did anyone send a message to Xiaomi support? I sent it to [support@aqara.com](mailto:support@aqara.com) a while ago but I couldn't pass through the support.
> 
> I'll try again, but maybe if more people send emails, they may finally understand that this is a problem.
> 
> Note: I have several of those plugs, so I'm quite annoyed that such issue exists for so long.

I just made a ticket with them using your link https://static-resource.aqara.com/html/feedback.html . Seeing the response @przemekgithub got where they simply said they don't support third-party software or devices I should probably have phrased it in a different way.

If anyone has a suggestion for a message template that can phrase this problem in a way that might motivate them to fix them that might help. Having a finished text to go from might make more people contact them instead of having to come up with the message body themselves.

Since the bug seems to make the device not respect group bindings (it toggles even if the message is sent on a different group), does that conform with Zigbee 3.0? If not, that might be a valid argument for them to prioritize this since they advertise zigbee 3.0 support. Keep in mind here that I am not very familiar with the zigbee protocol so I do not know if my assumption is correct.

rotilho, 2023-03-04:
Just sent another one:

```
Dear Aqara Support Team,

I hope this message finds you well. I am reaching out to you regarding an issue I have been experiencing with my Aqara Plug (SP-EUC01) for the past six months. Since the update 41 was released, the plug seems to have stopped following the Zigbee 3.0 standard, which has caused significant inconvenience.

I have been eagerly waiting for a new update to address this issue, but to my disappointment, the latest firmware 43 released last month did not resolve the problem. As a loyal customer who owns not only 6 of the SP-EUC01 plugs but also many other Aqara devices, I must say that I am deeply disappointed.

Before I consider any legal action or filing a claim with the CSA, I kindly request that your technical team takes a closer look into this matter and provides a solution for the SP-EUC01. I would also like to remind you that Aqara is advertised as Zigbee 3.0 Certified, and the current implementation seems to contradict this claim.

For the technical team: The SP-EUC01 is disregarding groups it's bound to, causing it to react to any on/off command, even if it's not for its group when the message is routed by the plug. I hope this information is useful in resolving the issue.

Thank you for your attention to this matter, and I look forward to hearing from you soon.

Sincerely
```

swoopdk, 2023-03-04:
> 

Thanks for the reply - i'm trying to use the header 99 version right now to confirm it's working here for me. My 'fear' was mostly that for each new version (if there ever is a new version that seems to fix this bug) I/we would have to get a firmware with a header version 99+ to even get that installed. But for now i'm dropping this header 99 firmware onto both my plugs and i'll test later today if it's working as intended (i suspect it will) :) 

I can only hope that they fix this issue going forward and i'll follow the example of @rotilho and send them a message on support hoping they'll confirm to standards soon. 

Btw slightly off topic does anybody have suggestions for other smartplugs that can measure power consumption that doesn't have this issue ? e.g. non aqara plugs.

swoopdk, 2023-03-04:
Well first plug downgraded with success and reports appVersion 32 now with firmware build date of 09-22-2020

I have not been able to provoke the issue like before so i'll assume that version 32 like everybody says works fine and i'll cross my fingers for updates in the future that works "normally" for this plug :) 

[image]
[image]

Thanks for helping a newbie out here :)

Roarrk, 2023-03-13:
> Just sent another one:
>
> ```
> Dear Aqara Support Team,
>
> I hope this message finds you well. I am reaching out to you regarding an issue I have been experiencing with my Aqara Plug (SP-EUC01) for the [...]

I would also point out that this non-compliance can have significant economic and safety implications. A power plug semi-randomly turning on and off is not really something you want in a house.

rotilho, 2023-03-15:
Good idea @Roarrk 

I managed to pass through the support.

```
Thank you for getting back to us.

Our team is working on investigating this issue, we will get back to you once there is any news.
Sorry for any inconvenience this may have caused. 
 
Sincerely,
Ada
Aqara Technical Support
```

Roarrk, 2023-03-24:
@TheJulianJES Shouldn't this issue stay open until there is a new firmware that solves it? I understand the other issues were closed.

maurice-p, 2023-05-11:
A new firmware update (V43) is listed on the Aqara product page (2023-03-10)
https://www.aqara.com/en/version/smart-plug-eu

The information says it is about function optimization, unfortunately no specifics. Does it solve the problem? 

I don't have an original Aqara hub, so I guess I can't test it myself.

guy561, 2023-09-02:
> NOTE: This is only for the Xiaomi `mmeu01` (square EU plug). **NOT** THE ROUND EU PLUG. For the round Aqara EU plug, see the instructions here: [#13903 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072)
> 
> NOTE: ~I haven't tried this firmware at all~ (works fine for me) and it could **brick** your device. It's the v22 firmware file modded to have a v25 header (so the plug will accept "upgrading" to it from the apparently buggy v24 release). In about 10 hours, I might be able to to try the downgrade for an `mmeu01` plug with that firmware.
> 
> For the `mmeu01` plug, you can try the following firmware: [OTA_lumi.plug.mmeu01_V22_20190906_D32362.20191008105750_modded_to_v25.ota.zip](https://github.com/Koenkk/zigbee2mqtt/files/9654652/OTA_lumi.plug.mmeu01_V22_20190906_D32362.20191008105750_modded_to_v25.ota.zip) (make sure to extract the image from the ZIP)
> 
> The instructions for installing that firmware should be basically identical to the ones in [#13903 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13903#issuecomment-1244720072).

My Lumi plug mmeu01 works again after downgrading the firmware with "modded 25 ota version" A big thank-you. If a more recent update is available I could install it easily or because of the modified version I would be blocked?

TheJulianJES, 2023-09-02:
You could easily install it, but newer versions all seem to have the issue.

guy561, 2023-09-03:
Great new, thanks

zoltanf, 2023-11-11:
For anyone who had the latest update (at the moment of writing it is 43) the only way to update is to set the manual file to v44, via the python script above.

wassereimer86, 2023-11-20:
What can I do if I checked for OTA updates (modified to v42) and found them. For some devices, I need the modified to v99 (they are on v43). How can I recheck for updates? I can only start it, but it fails because of the v42 that is remembered.

zoltanf, 2023-11-22:
@wassereimer86 they are coming with v43 out of the box. You need to a modified file to flash these. I have been using this one bellow. You also need a 'index.json' in your home assistant zigbee2mqtt folder, and somehow you must copy the OTA file bellow over to that folder as well (I used scp over ssh). 

index.json
`[
    {
      "modelId": "lumi.plug.maeu01",
      "url": "OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v44.ota",
      "path": "./OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v44.ota",
      "force": true
    }
]`

[OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v44.ota.zip](https://github.com/Koenkk/zigbee2mqtt/files/13435116/OTA_lumi.plug.maeu01_V32_20200922_C11F8A.20200924102946_modded_to_v44.ota.zip)

wassereimer86, 2023-11-22:
@zoltanf Thank you! That worked. I already got the OTA notification that there is an update to v43 available. Is there a way to get rid of the notification? I think the automatic search was right between the update and me disabling the OTA. I also removed the uploaded files. Or is there a modified Version below v32 that I can upload, then try updating and the device will block the update, perhaps?

[44 more replies]