Koenkk/zigbee2mqtt #13918: Devices stop working

lordraiden, 2022-09-08:
### What happened?

All this setup was working before, I went on hollidays and now is not working at all.
I'm running zigbee2mqtt on docker on Unraid
root@localhost:# /usr/local/emhttp/plugins/dynamix.docker.manager/scripts/docker run -d --name='Zigbee2MQTT' --net='br0' `--ip='10.10.10.102' --cpuset-cpus='8,9,10,11,20,21,22,23' -e TZ="Europe/Paris" -e HOST_OS="Unraid" -e HOST_HOSTNAME="Unraid" -e HOST_CONTAINERNAME="Zigbee2MQTT" -e 'TCP_PORT_9442'='9442' -l net.unraid.docker.managed=dockerman -l net.unraid.docker.icon='https://raw.githubusercontent.com/digiblur/unraid-docker-templates/master/images/zigbee2mqtt.png' -v '/mnt/user/Docker/Zigbee2MQTT':'/app/data':'rw' --device='/dev/ttyUSB0' --dns=10.10.50.5 'koenkk/zigbee2mqtt:latest'`

None of my devices were working, I read somewhere that switching to channel 25 could fix it, I did it and now of works until I restart the container.
I can't pair my ikea lighs they don't even appear in the pairing proccess.
I have ben able to pair a MOES swicth it works but if I restart the container I still have it paired but If I change the status or anything it doesn't work.

[image]

info 2022-09-08 13:22:53MQTT publish: topic 'zigbee2mqtt/bridge/response/networkmap', payload '{"data":{"routes":false,"type":"raw","value":{"links":[{"depth":1,"linkquality":170,"lqi":170,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x847127fffe09a99d","networkAddress":6893},"sourceIeeeAddr":"0x847127fffe09a99d","sourceNwkAddr":6893,"target":{"ieeeAddr":"0x00124b0021cc46cd","networkAddress":0},"targetIeeeAddr":"0x00124b0021cc46cd"},{"depth":1,"linkquality":170,"lqi":170,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d0002b52c57","networkAddress":28053},"sourceIeeeAddr":"0x00158d0002b52c57","sourceNwkAddr":28053,"target":{"ieeeAddr":"0x00124b0021cc46cd","networkAddress":0},"targetIeeeAddr":"0x00124b0021cc46cd"},{"depth":1,"linkquality":170,"lqi":170,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x00158d0002b5c994","networkAddress":49736},"sourceIeeeAddr":"0x00158d0002b5c994","sourceNwkAddr":49736,"target":{"ieeeAddr":"0x00124b0021cc46cd","networkAddress":0},"targetIeeeAddr":"0x00124b0021cc46cd"},{"depth":1,"linkquality":170,"lqi":170,"relationship":1,"routes":[],"source":{"ieeeAddr":"0x60a423fffeef59f5","networkAddress":12710},"sourceIeeeAddr":"0x60a423fffeef59f5","sourceNwkAddr":12710,"target":{"ieeeAddr":"0x00124b0021cc46cd","networkAddress":0},"targetIeeeAddr":"0x00124b0021cc46cd"}],"nodes":[{"definition":null,"failed":[],"friendlyName":"Coordinator","ieeeAddr":"0x00124b0021cc46cd","lastSeen":null,"networkAddress":0,"type":"Coordinator"},{"definition":{"description":"Wall touch light switch (1 gang)","model":"ZTS-EU_1gang","supports":"switch (state), indicate_light, power_on_behavior, linkquality","vendor":"Moes"},"friendlyName":"Timbre entrada","ieeeAddr":"0x847127fffe09a99d","lastSeen":1662635824533,"manufacturerName":"_TZE200_amp6tsvy","modelID":"TS0601","networkAddress":6893,"type":"EndDevice"}]}},"status":"ok","transaction":"gg35t-3"}'
error 2022-09-08 13:23:18Publish 'set' 'state' to 'Timbre entrada' failed: 'Error: Command 0x847127fffe09a99d/1 manuSpecificTuya.dataRequest({"seq":0,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'

### What did you expect to happen?

I have been running this for years and I have always known how to fix it when broken, but right now I am totally lost on what to do.

### Zigbee2MQTT version

1.27.1

### Adapter firmware version

20211217

### Adapter

Slaesh's zStack3x0

### Debug log

debug 2022-09-08 13:30:56Received MQTT message on 'zigbee2mqtt/Timbre entrada/set' with data '{"state":"ON"}'
debug 2022-09-08 13:30:56Publishing 'set' 'state' to 'Timbre entrada'
debug 2022-09-08 13:30:56Received MQTT message on 'zigbee2mqtt/Timbre entrada/set' with data '{"state":"ON"}'
debug 2022-09-08 13:30:56Publishing 'set' 'state' to 'Timbre entrada'
debug 2022-09-08 13:31:28Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"error"}},"transaction":"ubijb-1"}'
info 2022-09-08 13:31:30Succesfully changed options
info 2022-09-08 13:31:30MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"ubijb-2"}'
debug 2022-09-08 13:31:33Received MQTT message on 'zigbee2mqtt/bridge/request/options' with data '{"options":{"advanced":{"log_level":"error"}},"transaction":"ubijb-3"}'
info 2022-09-08 13:31:34Succesfully changed options
info 2022-09-08 13:31:34MQTT publish: topic 'zigbee2mqtt/bridge/response/options', payload '{"data":{"restart_required":false},"status":"ok","transaction":"ubijb-4"}'
debug 2022-09-08 13:31:49Received MQTT message on 'zigbee2mqtt/Timbre entrada/set' with data '{"state":"ON"}'
debug 2022-09-08 13:31:49Publishing 'set' 'state' to 'Timbre entrada'
error 2022-09-08 13:32:16Publish 'set' 'state' to 'Timbre entrada' failed: 'Error: Command 0x847127fffe09a99d/1 manuSpecificTuya.dataRequest({"seq":0,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
debug 2022-09-08 13:32:16Error: Command 0x847127fffe09a99d/1 manuSpecificTuya.dataRequest({"seq":0,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
debug 2022-09-08 13:32:20Received MQTT message on 'zigbee2mqtt/Timbre entrada/set' with data 'ON'
debug 2022-09-08 13:32:20Publishing 'set' 'state' to 'Timbre entrada'
error 2022-09-08 13:32:59Publish 'set' 'state' to 'Timbre entrada' failed: 'Error: Command 0x847127fffe09a99d/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
debug 2022-09-08 13:32:59Error: Command 0x847127fffe09a99d/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)
error 2022-09-08 13:33:42Publish 'set' 'state' to 'Timbre entrada' failed: 'Error: Command 0x847127fffe09a99d/1 manuSpecificTuya.dataRequest({"seq":2,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
debug 2022-09-08 13:33:42Error: Command 0x847127fffe09a99d/1 manuSpecificTuya.dataRequest({"seq":2,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23) at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32)

lordraiden, 2022-09-08:
more logs during the pairing process

2022-09-08 14:16:53Publish 'set' 'state' to '0x847127fffe09a99d' failed: 'Error: Command 0x847127fffe09a99d/1 manuSpecificTuya.dataRequest({"seq":1,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
2022-09-08 14:17:15Publish 'set' 'state' to '0x847127fffe09a99d' failed: 'Error: Command 0x847127fffe09a99d/1 manuSpecificTuya.dataRequest({"seq":2,"dpValues":[{"dp":1,"datatype":1,"data":[1]}]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'

lordraiden, 2022-09-08:
I have resintalled from scratch Zigbe2mqtt docker I configure it mostly with defaults not restoring previous config and I get the same behaviour, it works but if I reboot the docker it stops working

lordraiden, 2022-09-08:
now I have integrated my aqara switch
[image]

but is not expossing the same values as before
[image]

rosarioscavo, 2022-09-11:
> Could there be any cirscuntance under which my coordinator would have lost range? devices works but they need to be too close. Before it was working fine, it has been the same setup for years, same routers and end devices, batteries checked.... I have tried to change the power transmit but didnt fix anything.

Have you changed the coordinator?

liquid-motion, 2022-09-23:
I have exactly the same problem, after returning from a holiday. Not one Zigbee device is working anymore...

lordraiden, 2022-09-23:
> Have you changed the coordinator?

No

Zver5, 2022-10-14:
there is also such a problem that helps to change ports in mqqt broker. Rebooting doesn't help

lost-carrier, 2022-10-31:
Same here. Freshly set up Zigbee2MQTT v1.28.0 ... a few hours later I only get "'MAC transaction expired' (240)" for all devices (a couple of Aqara and Ikea Traedfi Switches)

closed 2022-12-08 by github-actions[bot]

lluisd, 2023-03-04:
@lordraiden did you solved it? Are you running it in a synology nas?