Koenkk/zigbee2mqtt #13920: [New device support]: Zemismart Tubular Motor ZM25RX-08/30

nara1111, 2022-09-08:
### Link

https://www.zemismart.com/products/zm25rx-08-30

### Database entry

"id":17,"type":"EndDevice","ieeeAddr":"0x84ba20fffe0375e7","nwkAddr":41910,"manufId":4098,"manufName":"_TZE200_7eue9vhc","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"Y\f�*\u0019�\f�*\u0019I\r�*\u0019�\r�*\u00199\u000e�*\u0019�\u000e�*\u0019)\u000f�*\u0019\u0010\u0010�*\u0019","65506":31,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE200_7eue9vhc","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c0a934","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1662640001007,"defaultSendRequestWhen":"immediate"}

### Comments

This is Zemismart Tubular Motor

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_7eue9vhc'
        },
    ],
    model: 'ZM25RX-08/30',
    vendor: 'Zemismart',
    description: 'Roller Blind Tubular Motor',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

Koenkk, 2022-09-09:
Does it work with this external converter?

nara1111, 2022-09-09:
@Koenkk Yes. The external converter can adjust open/close status.

Koenkk, 2022-09-10:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-10 by Koenkk

DariBer, 2022-09-11:
Does it report battery status? Mine doesn't, I just have a question mark

nara1111, 2022-09-11:
@DariBer No. I used wired motor's converter. It might not report battery status.

DariBer, 2022-09-11:
@nara1111 OK, would be nice to have the battery status, I tried myself to find out how to get that info, but without any luck

nara1111, 2022-09-12:
@DariBer Maybe, with Tuya Zigbee Gateway, be able to decode tuya signal, I think. However, unfortunately, no I have no the gateway.

DariBer, 2022-09-12:
@nara1111 I might have a Tuya gateway somewhere in a drawer. Will see if I can find it. But unfortunately I don't know how to get the data to decode. but maybe you or someone can help me

mattkerrison, 2022-11-20:
Any chance anyone got battery status working?

mattkerrison, 2022-12-05:
I've got this motor and connected to zigbee2mqtt fine but can't set the limits of the blind via the blind itself or zigbee2mqtt, the instructions only talk about the remote or app that needs the zemismart hub, anyone work out a button combo on the motor to set the limits?