Koenkk/zigbee2mqtt #13921: Support for Tuya USB Zigbee Repeater - Device wrongly detected

KoenChiau, 2022-09-08:
### What happened?

The device is detected as https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector but should be https://www.zigbee2mqtt.io/devices/TS0207_repeater.html. 

This problem was only partially fixed later: https://github.com/Koenkk/zigbee2mqtt/issues/4329

### What did you expect to happen?

Device should be recognized as TuYa TS0207_repeater.

### How to reproduce it (minimal and precise)

Pair a TuYa TS0207_repeater and it's displayed as follow:

- Device type: Router
- Zigbee Model: TS0207
- Zigbee Manufacturer:    _TZ3000_gszjt2xx
- Description: Water leak detector
- Support status: Supported
- IEEE Address:     0x04cd15fffe0a2496
- Network address: 0x3035
- Manufacturer:     [TuYa](https://www.zigbee2mqtt.io/supported-devices/#v=TuYa)
- Model:     [TS0207_water_leak_detector](https://www.zigbee2mqtt.io/devices/TS0207_water_leak_detector.html#tuya-ts0207_water_leak_detector)
- Power:
- Interview completed: True

### Zigbee2MQTT version

1.27.2-dev commit: 2b1d1ff

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

Koenkk, 2022-09-09:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-09 by Koenkk

KoenChiau, 2022-09-09:
Many thanks, Koen !

Reply to: Koenkk/zigbee2mqtt ***@***.***>
Date: Friday, 9 September 2022 at 09:24
To: Koenkk/zigbee2mqtt ***@***.***>
Cc: Koen Chiau ***@***.***>, Author ***@***.***>