Koenkk/zigbee2mqtt #13922: Is my configuration set in HomeAssistant ignored?

iz3man, 2022-09-08:
### What happened?

Setting parameters in yaml edit mode in z2m add-on configuration tab seems to be ignored.

This is my config, and I set channel to 15, but in z2m the dongle's channel still remains on 11.
pan_id is set to GENERATE, but it still remains GENERATE after restart, or reboot.

So basically everything below "advanced" seems ignored?!

`data_path: /config/z2m
socat:
  enabled: false
  master: pty,raw,echo=0,link=/tmp/ttyZ2M,mode=777
  slave: tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5
  options: "-d -d"
  log: false
mqtt: {}
serial:
  port: >-
    /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_fe373a119f19ec119b3a37cc47486eb0-if00-port0
advanced:
  pan_id: GENERATE
  channel: 15
  last_seen: ISO_8601_local
device_options:
  homeassistant:
    last_seen:
      enabled_by_default: true
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500`

### What did you expect to happen?

I was expecting the settings to be read and used

### How to reproduce it (minimal and precise)

Go to yaml edit, change to what I posted above, save and restart.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

Sonoff 3.0 P

Koenkk, 2022-09-09:
Only MQTT and serial section can be configured via this page, rest is done via the UI

iz3man, 2022-09-09:
Ah! Makes sense. So if I want to change Zigbee channel 11 -> 15 I remove EVERYTHING under the port setting, and do my changes in the UI?

[image]

And the "device_options" with "last seen" etc goes where? In the main HomeAssistant configuration.yaml?

Thanks!!

Koenkk, 2022-09-10:
> So if I want to change Zigbee channel 11 -> 15 I remove EVERYTHING under the port setting, and do my changes in the UI?

Only change the channel and save

> And the "device_options" with "last seen" etc goes where? In the main HomeAssistant configuration.yaml?

All in the z2m configuration.yaml (can be configured from the ui)

closed 2022-10-18 by github-actions[bot]

Stooovie, 2023-09-05:
yes, same here. Channel set in yaml is ignored, and stays as before in Z2M UI.