Koenkk/zigbee2mqtt #13923: Z2M fails to start unless "/share/zigbee2mqtt" is in quotes

Stooovie, 2022-09-08:
### What happened?

Zigbee2MQTT fails to start unless I enclose "/share/zigbee2mqtt" in quotes. That has to be done on every HA host boot as the quotes get removed. 

Issue started last week

### What did you expect to happen?

Z2M to start

### How to reproduce it (minimal and precise)

Reboot HA host

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

No idea

### Adapter

Slaeh.sh CC2652RB

### Debug log

19:56:52] INFO: Handing over control to Zigbee2MQTT Core ...
Error: ENOENT: no such file or directory, open 'null/configuration.yaml'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.read (/app/lib/util/yaml.ts:7:29)
    at read (/app/lib/util/settings.ts:311:20)
    at getInternalSettings (/app/lib/util/settings.ts:418:21)
    at Object.reRead (/app/lib/util/settings.ts:696:5)
    at start (/app/index.js:93:14)
[19:56:55] INFO: Handing over control

Koenkk, 2022-09-09:
> `Error: ENOENT: no such file or directory, open 'null/configuration.yaml'`

See https://github.com/Koenkk/zigbee2mqtt/issues/12801, this is a HA bug and should be fixed in supervisor 2022.08.5

closed 2022-09-09 by Koenkk

Calimerorulez, 2022-09-09:
Sadly to say that it is back again, I'm experiencing it in 2022.9.1 again.

Koenkk, 2022-09-10:
@Calimerorulez then please report it in the HA issue tracker (it is not a z2m issue)

Stooovie, 2022-09-12:
HA devs answered that this is Zigbee2MQTT issue, not a HA issue. Meanwhile, the quote trick doesn't work anymore and I'm left with half the household not working. Any tips?

Koenkk, 2022-09-13:
> HA devs answered that this is Zigbee2MQTT issue, not a HA issue.

Can you link me this?

Stooovie, 2022-09-13:
> Can you link me this?

Sure, [here](https://github.com/home-assistant/core/issues/78263#issuecomment-1243556521)

Koenkk, 2022-09-14:
That issue talks about the quotes, not about the `Error: ENOENT: no such file or directory, open 'null/configuration.yaml'` issue. I suggest to create a new issue and link to https://github.com/Koenkk/zigbee2mqtt/issues/12801. It also seems another user has the same issue https://github.com/Koenkk/zigbee2mqtt/issues/12801#issuecomment-1246332839. Once that is fixed I can look into the double quote issue.