Koenkk/zigbee2mqtt #13925: Aurora RGBW LED strip Controller

zakery292, 2022-09-08:
### Link

https://auroralighting.com/gb/trade/productdetail/au-a1zbscrgbcx

### Database entry

{"id":21,"type":"Router","ieeeAddr":"0x000d6ffffed79dc3","nwkAddr":22703,"manufId":4636,"manufName":"Aurora","powerSource":"Mains (single phase)","modelId":"RGBCXStrip50AU","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"RGBCXStrip50AU","manufacturerName":"Aurora","powerSource":1,"zclVersion":2,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":1,"colorTempPhysicalMin":166,"colorTempPhysicalMax":400,"colorMode":2,"currentX":24939,"currentY":24701,"currentSaturation":0,"colorTemperature":166,"options":0,"coupleColorTempToLevelMin":0,"enhancedColorMode":2}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1662663764615,"defaultSendRequestWhen":"immediate"}

### Comments

Have created a converter, got the device working with limited function;
Working:
on/off 
Brightness
Not working:
Colour 
Temperature
Not sure on code (trying to learn as we speak) any ideas more than happy to try them. Dev console will show current X/Y 
 2022-09-08 20:18:58Read result of 'lightingColorCtrl': {"currentX":24939,"currentY":24701}
 
Thank you 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['RGBCXStrip50AU'],
    model: 'RGBCXStrip50AU',
    vendor: 'Aurora',
    description: 'RGBW LED Strip Controller',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    extend: extend.light_onoff_brightness_colortemp_color({"colorTempPhysicalMin":166,"colorTempPhysicalMax":400,"color_mode":2,"colorCapabilities":1}),
};

module.exports = definition;
```

### Supported color modes

Temp, RGBW

### Color temperature range

{"colorTempPhysicalMin":166,"colorTempPhysicalMax":400}

Koenkk, 2022-09-09:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-09 by Koenkk

zakery292, 2022-09-09:
Your amazing! Thank you!